node-opcua/node-opcua #1089: Bug Found: Unable to use ExtensionObjects with properties "key" and "value"

paulking00, 2021-11-25:
Bug found in the file 
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-variant/source/variant.ts

in the function
isEnumerationItem

**Current behavior**

I am using node-opcua as a client to access an OPCUA server not controlled by me.

The server has defined an ExtensionObject (Structure) called "KeyIntPair" with the fields
"key" and
"value"

I need to call a method to write a value of the type KeyIntPair

**Describe the bug**

The method call fails with the Error
*Error: expecting DataType.Int32 for enumeration values ; got DataType.22 instead*
which is generated in this file
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-variant/source/variant.ts

**Step by step instruction to reproduce**

`    

    const keyIntPairParams = {
        key: "test",
        value: 2
    };

    const dataTypeNodeId = resolveNodeId("ns=2;i=3002");
    const KeyIntPairConstructor = await getExtensionObjectConstructor(session, dataTypeNodeId);
    const keyIntPairData = new KeyIntPairConstructor(keyIntPairParams);

    // Populate Method call with ExtensionObject as InputArgument
    const methodToCall = {
        objectId: "ns=2;i=5005",
        methodId: "ns=2;i=7001",
        inputArguments: [{
            dataType: DataType.ExtensionObject,
            value: keyIntPairData
        }]
    };

    session.call(methodToCall)
`

**Expected behavior**

The Method call should work correctly.

!!
**After editing the function isEnumerationItem, the method call does indeed work as expected**
!!

**Cause of the Bug**

`
function isEnumerationItem(value)  {
    return value instanceof Object && value.hasOwnProperty("value") && value.hasOwnProperty("key") && value.hasOwnProperty("enum") ;
}
`

The function isEnumerationItem incorrectly identifies any ExtensionObject with the properties "value" and "key" as an Enumeration, resulting in the above Error.

**Screenshots**

**Gist**

 
gist:

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: __Ubuntu____
- OS version: _20.04_____

    - ( ) Windows : version : ______
    - (X) Linux : version : _Ubuntu 20.04_____
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
2.44.0
  
  

- **Node:**
  `node --version` =v15.13.0

erossignon, 2021-12-01:
@paulking00 Can you please check in node-opcua@2.62.0 that the issue is now resolved?

paulking00, 2021-12-02:
@erossignon Can confirm that the issue is resolved in node-opcua@2.62.0.
Thank you!

erossignon, 2021-12-02:
Thank you, Paul

closed 2021-12-02 by erossignon