node-opcua/node-opcua #1090: Unable to connect to my server with security modes "sign" or "signAndEncrypt"

PatersonKopp, 2021-11-26:
Hello, currently I have the problem that my OPC UA server rejects all clients with the security mode "sign" and "signAndEncrypt" with the error message BadSecurityChecksFailed (0x80130000). 

Since there is absolutely nothing documented about this, can anyone help me?

How do I proceed here? What could be possible causes that the server rejects the requests?
Does the server certificate have to fulfil any requirements or trust the individual client certificates?

Tested with the UaExpert and the prosys-opc-ua-client

erossignon, 2021-12-14:
Hi Patrick,
Have you checked that the client certificate appears in the trusted folder of your PKI  and not in the rejected Folder ?

please reach out to us at contact@sterfive.com for a dedicated investigation on this case or subscribe to our membership support program at https://support.sterfive.com  for professional support as this forum is used for community help between community members.

PatersonKopp, 2021-12-14:
Hi @erossignon, thanks for your reply! 
The certificates really do appear in the "rejected" folder. Do you have any idea why? Or do you have an idea how I can debug why this happens?

Or is it meant to copy the certs manually, into the trusted folder?

erossignon, 2021-12-14:
Hi Patrick,
By default, unknown certificates goes to the rejected folder to prevent unknown client to connect to your server. This is by design for security reasons. There are several technique you could use to accept the client certificate on the server:

   - either use the automaticallyAcceptUnknownCertificate flag to true on the certificate manager (not recommended )
   - either manually move the certificate from the rejected folder to the trusted/cert folder of your PKI.
   - either implement the CertificatePush management feature on your server so that an administrator could remotely control the trusted/rejected certificate from a OPCUAClient. (preferred approach)

We can assist in setting up the CertificatePush management on your node-opcua based server, within our professional support program. Please contact us for further information.

PatersonKopp, 2021-12-14:
Hi @erossignon ,

thanks, that helped me a lot!
Would be very nice to have that a little bit better documented ;) 

I'll close the issue.

closed 2021-12-14 by PatersonKopp