node-opcua/node-opcua #1093: Input argument diagnostics always returned and requestHeader not available in MethodFunctor arguments

hoancea, 2021-12-02:
**Current behavior**

When binding a function to a UAMethod, the `requestHeader` cannot be found inside any of the arguments of the function. The `requestHeader` contains a flag that specifies if the input argument diagnostics should be returned or not, according to the link below.
https://reference.opcfoundation.org/v104/Core/docs/Amendment3/5.7.3/

**Describe the bug**

At the moment, the diagnostics are always returned, regardless of what is in the request header. 

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. use `bindMethod` of `UAMethod`
2. debug inside of the bound method
3. `requestHeader` not available anywhere

**Expected behavior**

Either provide us with the `requestHeader` inside the `SessionContext` in order to add the diagnostics ourselves OR strip away the returned diagnostic information if the flag (`returnDiagnostics`) on the `requestHeader` is set to true.

**Screenshots**

**Gist**

 
gist:

**Context**

- (X) my request is related to node-opcua acting as an OPCUA SERVER
- (X) I have installed node-opcua as a package ( using npm install )

- **node-opcua version:** : 2.59.0
  
  

- **Node:**
  `node --version` = 12.22.1

erossignon, 2021-12-02:
The specs says
[image]

nodet-opcua always returns an empty list arrayForInputArgumentDiagnosticInfo. 

[image]

hoancea, 2021-12-02:
I could assist and create a PR, however I would need to know which solution you would prefer, letting the consumer of the library check the `returnDiagnostics` flag or taking care of it in the final response generation.

erossignon, 2021-12-02:
Thank !
The second approach would be nice
>taking care of it in the final response generation

so that the diagnostic get filtered out before the response is sent if the client claimed did not request it in the `return diagnostics` field of the request header.
This will also not break the existing API. 

The other solution would require that `bindVariable` method signature changes