node-opcua/node-opcua #1096: Extension object instance defined in node set file is not loaded correctly

OSjoerdWie, 2021-12-03:
**Current behavior**

Data are not assigned correctly to an instance of an extension object defined in a node set file.

**Describe the bug**

The xml2json parser of the extension object has 'sub'-parsers for the respective fields. The internal fieldnames to access these are all lower camel case, whereas fields are not necessarily encoded the same. In particular, when `_on_startElement(..)` is called for a field in `node-opcua-xml2json@xml2json.ts:185`; the parser is `undefined` and will result in field data (text) not being assigned to the field's parser but to the parser of the parent.

In addition, after a field is parsed properly (i.e. given the name defined name is lower camel case in the type definition already), any data present (text) is referred to as `this.value` in `node-opcua-xml2json@extension_object_parser.ts:240`, yet is never assigned to `this.value`. The latter is always `undefined`

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. define a nodeset with a definition of an extension object, the extension object can have string fields `Field1` and `field2`
2. in the nodeset define an instance of the extension object, with `Field1` having value `value1` and `field2` having value `value2`
3. instantiate an OPCUA server with the corresponding nodeset file
4. put a breakpoint on `node-opcua-address-space@load_nodeset2.ts:144` (`const userDefinedExtensionObject = addressSpace2.constructExtensionObject(dataTypeNode, pojo);`seems to have moved to line 153 in the current repo version)
5. note that for the instance the resulting `pojo` is undefined

**Expected behavior**

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: PC
- OS version: 

    - ( x) Windows : version : 10
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- **node-opcua version:** :  2.59.0
  
  

- **Node:**
  `node --version` = 12.22.1

erossignon, 2021-12-03:
Can you provide a nodeset.xml file containing the DataType defined and the instantiated variable to ease the investigation.

OSjoerdWie, 2021-12-03:
An example is stored in [example.zip](https://github.com/node-opcua/node-opcua/files/7649872/example.zip)

The example contains the described 'Field1' and 'field2' with an instance having values 'value1' and 'value2' respectively. 

On the breakpoint described in line described in step 4 of the reproduction scenarioit, the value of `pojo` equals `{ field2: { } }`.

OSjoerdWie, 2021-12-06:
I realized that step 5 in the reproduction scenario wrongly referred to `undefined`. This only holds if the type would not have `field2`. Steps adjusted accordingly.

OSjoerdWie, 2021-12-07:
While going through the code, it seems that part may be due to the reference to `field.dataType` in `_makeExtensionObjectReader(...)` in `extension_object_parser.ts`. As it seems, the property `dataType` does not exist on the field object. Instead, this information seems to be stored in the `fieldType` property.

When `fieldType` is used instead of `dataType`, at least the value seems to get set correctly for my string based field. However, this only works for a field that starts with a lower case character.

OSjoerdWie, 2021-12-07:
With respect to my question in the above bug report (_Additional question: I am not sure if changing the suggested lines would completely solve; I still can't seem to access the string field's value. How would I need to do this then?_), I've tried to trace where it would be assigned.

In `load_nodeset2.ts`, the parsed extension object is assigned to the `postponedExtensionObject` property of `postTaskData` in a `task` that gets pushed to `postTask2`. The only occurence where it is read back from a `postponedExtensionObject` property, is in a task that gets pushed to `postTasks3`. However, when debugging, I found that this task never gets created, as the logic checks the absence of `data.variant` (`load_nodeset2.ts:1246` in my used version), which never seems to be the case.

OSjoerdWie, 2021-12-09:
Regarding my previous comment, the fact that `data.variant` is never absent relates to the following:

https://github.com/node-opcua/node-opcua/blob/67b48cb862575c70f2a850ffe50580a1f2844629/packages/node-opcua-address-space/source/loader/load_nodeset2.ts#L1222-L1231

First `this.parent.parent.obj.value` gets assigned an object, then `data.variant` gets assigned the same object via `this.parent.parent.obj.value`. It seems that the check on the absence should not be on `data.variant`, yet on `data.variant.value`. Initially, `data.variant.value` is set to `this.extensionObject`, which, due to the mechanism that postpones the creation of the extension object, equals `null`. Therefore, I expect that the check should be on `data.variant.value` instead, as this should trigger in the mechanism.

I've verified succesful assignment via the postponed mechanism by making said change.

OSjoerdWie, 2021-12-13:
Just for the purpose of testing (and naively hoping), I created a similar example with another modeler ([example2.zip](https://github.com/node-opcua/node-opcua/files/7706777/example2.zip)). Unfortunately, it gives me same results.

In the meantime, I've forked the repo and tried to work on a solution. The suggested fix from my [previous comment](https://github.com/node-opcua/node-opcua/issues/1096#issuecomment-990217584) seems to be legit. However, for the other pieces of the problem, I am not too sure on how to solve these, yet. You've created an impressive package (or actually, set of packages), so it is taking me quite some time to get acquainted and understand the various flows.

Changing the `dataType` attribute of the `Field` type into `fieldType` works for my input. Actually, in `_makeExtensionObjectReader()` I've instantiated `const dataType = field.dataType ?? (field as any).fieldType` and referred to `dataType` instead of `field.dataType` within that function.
However, while building the library, I notice that I get errors: `Error: Cannot find StructureType constructor for TrimmedString - it may be abstract, or it could be a basic type`. As `TrimmedString` is a sub type of `String`, I have to assume that the error is valid (it is not a `StructureType`), yet I don't understand yet why I now see it. I do understand why I didn't see it before, as type was always passed as `undefined`. At the same time, I don't understand why the typescript for the various included (default) node sets in the building process gets created properly from the XML either way (i.e. with either the `.dataType` (`=== undefined`) property or the `.fieldType` property)

erossignon, 2021-12-14:
Hi OSjoerdWie, thank you for trying hard to resolve this. The reading of custom Extension Object within NodeSet2.XML file is currently not supported and only partially implemented in a way that will require a massive overhaul on our side as you figured out, This point is not yet visible on our roadmap.
The current workaround is to set up manually the extension object after the server has been initialized.
 Feel free to contact us directly at contact@sterfive.com if you feel you need us to prioritize this item on the roadmap higher.

closed 2022-08-23 by erossignon