node-opcua/node-opcua #1099: Make all API functions accepting 'Certificate' accept both PEM and DER certificates

OSjoerdWie, 2021-12-06:
-->

**Is your feature request related to a problem? Please describe.**

When generating a self signed certificate, this certificate is generated in PEM format. However, when calling functions like `checkCertificate`, the `Certificate` argument is expected to be in DER format (`Certificate` is defined as `DER` and `DER` as `Buffer`). This will give problems when you pass the `Buffer` based result of a read file of a PEM formatted certificate. In particular in this case the error that is thrown is not helpful ("cannot find version block"). 

**Describe the solution you'd like**

The `Certificate` type is universal (`Buffer` or `string`). In case a `Buffer` is supplied, an automatic detection and, if needed, conversion is done. 

**Describe alternatives you've considered**

On interface level use descriptive type names (e.g. `CertificateDER`), such that it is directly clear to the consumer.

erossignon, 2021-12-13:
When load from file, the convention is that certificate are in binary format (equivalent to DER)

the `convertPEMtoDER` helper function provided by the node-opcua-crypto API will help.

the readCertficate(filename: string) takes a PEM or DER file and  returns a Buffer object that contains the raw certificate.

OSjoerdWie, 2021-12-13:
Ok. Thanks for the consideration and feedback.

closed 2021-12-13 by OSjoerdWie