node-opcua/node-opcua #1102: Server returns BadAttributeIdInvalid for nodeclass Object

logikamente, 2021-12-13:
In opcua server there is an object "DB410_LETTURA". 

[image]

I try to read it from client with nodeId in this way 

`    const dataValue2 = await session.read({
      nodeId: "ns=4;i=14",
      attributeId: AttributeIds.Value
      });
      console.log(dataValue2);`

Server returns BadAttributeIdInvalid 

`references of RootFolder :
   ->  Objects
   ->  Types
   ->  Views
DataValue {
  value: Variant { dataType: 0, arrayType: 0, value: null, dimensions: null },
  statusCode: ConstantStatusCode {
    _value: 2150957056,
    _description: 'The attribute is not supported for the specified Node.',
    _name: 'BadAttributeIdInvalid'
  },
  sourceTimestamp: null,
  sourcePicoseconds: 0,
  serverTimestamp: null,
  serverPicoseconds: 0
}
`

Why?

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (X) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

erossignon, 2021-12-13:
Objects do not expose Value attributes, only Variables do.

logikamente, 2021-12-14:
Thanks!
Ok, now if I access the JobNumber variable I get an array of OpaqueStructure. How do I explode this dataType ?
[image]

`const dataValue2 = await session.read({
      nodeId: "ns=4;i=17",
      attributeId: AttributeIds.Value
      });
      console.log(dataValue2.value);`

I obtain 

Variant {
  dataType: 22,
  arrayType: 1,
  value: [
    OpaqueStructure {
      nodeId: [NodeId],
      buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 97 more bytes>
    },
    OpaqueStructure {
      nodeId: [NodeId],
      buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 97 more bytes>
    }
  ],
  dimensions: null
}

erossignon, 2021-12-14:
Hi @logikamente ,
The extension object in an array are usually decoded automatically by node-opcua client, we would need to investigate deeper on your specific server.
This is now a support issue, I recommend that you join one of our support program at https://support.sterfive.com for dedicated professional support.  or contact us at contact@sterfive.com

erossignon, 2023-05-05:
Closing due to long inactivity

closed 2023-05-05 by erossignon