node-opcua/node-opcua #1103: EPERM: operation not permitted, mkdir on certificate path. Can I change path of PKI certificate?

logikamente, 2021-12-14:
Hi,
Can I change path of PKI certificate?

I try to add a cron tab from my web server wher is installed OpcuaClient, but I obtain this error when run ts-node opcua.ts (opcua.ts is my script opcua-client)

[image]

Can I change path of PKI certificate? I try this but don't work

`const certificateManager = new OPCUACertificateManager({
    rootFolder : '/var/www/clients/client0/web1/web/node_opcua/pki', //my folder
    automaticallyAcceptUnknownCertificate: false
    });

  async function main() {
    try {
      await certificateManager.initialize();
     .... 
     ....
     }
}`

Thanks

**Context**

- (X ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (X) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

logikamente, 2021-12-14:
Hi Andreas,

I try this but dont't work

`
const client = OPCUAClient.create({
    applicationName: "MyClient",
    connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpointMustExist: false
    
  });

const certificateManager = new OPCUACertificateManager({
    rootFolder : '/var/www/clients/client0/web1/web/node_opcua/pki',
    automaticallyAcceptUnknownCertificate: false
    });
    
  const endpointUrl = "opc.tcp://192.168.147.3:4840";

  async function main() {
    try {
      await certificateManager.initialize();
await client.connect(endpointUrl);
}
}`

AndreasHeine, 2021-12-14:
i am not sure if the client needs a OPCUACertificateManager at all -> i used it typically in opc ua servers!

erossignon, 2021-12-14:
You're almost there:  you need to createa clientCertificateManager  before and pass it as a argument to the OPCUAClient create method, so that the default clientCertificateManager  is not used.

```` javascript 

async function main() { 
  try {
    const clientCertificateManager = new OPCUACertificateManager({ 
        rootFolder : "/var/www/clients/client0/web1/web/node_opcua/pki", 
        automaticallyAcceptUnknownCertificate: false
    });
    await certificateManager.initialize(); 

    const client = OPCUAClient.create({ 
        applicationName: "MyClient", 
        connectionStrategy, 
        securityMode: MessageSecurityMode.None,
        securityPolicy: SecurityPolicy.None, 
        endpointMustExist: false,
        clientCertificateManager 
   });
   const endpointUrl = "opc.tcp://192.168.147.3:4840";

    await client.connect(endpointUrl);

  } catch(err){
      console.log(err);
  } 
}
```

make sure that the  `/var/www/clients/client0/web1/web/node_opcua` folder exists before hand and is writeable

AndreasHeine, 2021-12-15:
@erossignon what is the benifit of using a certmanager instead of using the servercert in clientcreate? just for my own curiosity :)

erossignon, 2021-12-15:
THe PKI is used to store and manage the certificates received by the client from connected server. ( or by the server from connected client) .(not only the client or server own certificate) . PKI are also used to manager CA Certificates and Certifiate revocation lists that are used during security validation.

logikamente, 2021-12-17:
Thank you a lot for your fast reply! It works!

closed 2021-12-17 by erossignon