node-opcua/node-opcua #1104: Question: mimic historic data using a data file

rgiacomin, 2021-12-17:
This is a question, not an issue.

I want to create an OPC UA server simulator, reading historical data from text files (one file per variable).
It should support both read raw historical data, as well as aggregations.

Is it somehow possible to tell the opc server it should read the data from the files?

AndreasHeine, 2021-12-17:
hey, from what i found in tha API this could do the trick:
https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua_address_space.addressspace.html#installhistoricaldatanode
https://node-opcua.github.io/api_doc/2.32.0/modules/node_opcua.html#ihistoricaldatanodeoptions
https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.ivariablehistorian.html

mikakaraila, 2021-12-18:
I made this couple of year ago just with simple csv files (Excel is nice to generate needed signals).
Sorry but I cannot share code because I implemented it for the company.
Tip: read values into the vector and use setInterval/setTimeout to update value from the vector.

rgiacomin, 2021-12-20:
@AndreasHeine , thank you for answer. Providing a IVariableHistorian as an option during installHistoricalDataNode really did the trick!!!

closed 2021-12-20 by rgiacomin