node-opcua/node-opcua #1107: NodeJS OPC UA Client - Extension Object Handling

mohitkumaragarwal, 2022-01-04:
**I'm submitting a question  (check one with "x")
 - (x ) support request 

**Current behavior**
Hello,

I have an OPC UA Server which is running and has few complex data types.

I am trying to use the OPC UA Client from Node JS - https://github.com/node-opcua/.....cua-sample and trying to read an extension object.

I have also referred to the NodeJS Handbook - node-opcuabyexample.pdf and the example says, to parse the Extension Object, it says just do the following: console.log(dataValue.toString());

But, when I try to do that, I get the following output for one of the structures:

New Value = { /* DataValue */
value: Variant(Array<ExtensionObject>, l= 2, value=[/* OpaqueStructure */ {
nodeId ns=1;i=5017
buffer =
00000000: 03 00 00 00 56 49 4e 08 00 00 00 45 59 58 31 32 34 35 36 ....VIN....EYX12456

},/* OpaqueStructure */ {
nodeId ns=1;i=5017
buffer =
00000000: 0d 00 00 00 53 4f 43 4b 45 54 5f 4e 55 4d 42 45 52 01 00 00 00 33 ....SOCKET_NUMBER....3

}])
statusCode: Good (0x00000)
serverTimestamp: 2022-01-04T09:35:12.501Z $ 526.000.000
sourceTimestamp: 1601-01-01T00:00:00.000Z $ 000.000.000
}

How DO I parse extension objects so that I can map respective values correctly. Currently the output is in the binary form but the NodeJS example does NOT any details on parsing the structure field accordingly.
**Expected behavior**

**Please tell us about your environment:**

  * (x) my request is related to node-opcua acting as a  OPCUA CLIENT 

  * (x) I have installed node-opcua as a package ( using npm install )
        
  * Device: Dell Laptop
  * OS version: Windows 10    
    - ( ) Windows : version : 20H2
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: OPC UA Flex SDK Server - Based on Matrikon Flex SDK.
     - Version: This is an application on top of Matrikon Flex SDK 5.0 Version
     - Manufacturer/Software vendor: Matrikon Flex SDK
     - link : https://
     
    
* **node-opcua version:**  :
https://github.com/node-opcua/node-opcua-sample

AndreasHeine, 2022-01-04:
have you tried latest version of node-opcua?

mohitkumaragarwal, 2022-01-04:
Hi,
I tried with latest base from the below link: https://github.com/node-opcua/node-opcua
Is there any specific version you would like me to try?

Best regards,

Mohit Agarwal
System Architect
R&D System Solutions
________________________________
AC MVI GI
Atlas Copco Industrial Technique AB

Visitor Address: Sickla Industriväg 19, Stockholm, 131 54, Sweden
Postal Address: 105 23, Stockholm, 131 54 Nacka, Stockholm

Phone: +46 (0) 87438290 - Mobile: +46 (0) 766961990
E-mail: ***@***.******@***.***>

Visit Atlas Copco at:   http://www.atlascopco.com<http://www.atlascopco.com/>
Follow us at:  Facebook<https://www.facebook.com/atlascopco> - LinkedIn<http://www.linkedin.com/company/atlas-copco> - Twitter<https://twitter.com/AtlasCopcoGroup> - YouTube<http://www.youtube.com/user/atlascopcogroup> - Instagram<https://www.instagram.com/atlascopcogroup/>
________________________________
Home of Industrial Ideas

AndreasHeine, 2022-01-04:
https://github.com/node-opcua/node-opcua/commit/d9c9930eb0a5a80322393364cdc59e8dbcfc9cf2 :
> In 1.04, the definition of extension Object is now attached the
> corresponding dataType object.
> In 1.03 the extension object definition was store into a embedded
> BSD in a DataTypeSystemType. This required a complex preparation
> on the client side.
> 
> DataType
>   | 1
>   |
>   | n
>   +-HasEncoding-> "Default Binary" (O)[DataTypeEncodingType]
>                      |
>                      +-HasDescription->"MyItemType" (V)[DataTypeDescriptionType]
>                                          |
> 					 +-ComponentOf->Schema(V) []
>                                               |
>                                               +-ComponentOf->OPC Binary(V)[DataTypeSystemType]
> 
> Up to recently, 1.04 servers was mainly mixing the 1.03 and 1.04
> approcah in the address space.
> New 1.04 server are now only exposing dataType definition inside
> the DataTypeDefinition attribute of the DataType Object and do not
> expose the DataTypeDescriptionType nor the Schema nodes
> 
> In this commit, we introduce support for this new setup.

would be interesting what matrikon uses 1.03 or 1.04!

mohitkumaragarwal, 2022-01-04:
Thanks for the response, Matrikon SDK which I am using is based on 1.04 and from the CS NodeSet perspective, I have removed the TypeDictionary node which is deprecated since NodeOPC earlier was not able to read the built-in type due to that.

So, I have 1.04 based OPC UA Server and trying to parse the CS specific extension object. Are there any specific APIs which I can refer or any example on parsing user defined structures on the client side?

Also, I have pulled the latest commit from the NodeOPCUA GIT.

Best regards,
Mohit Agarwal

d9c9930<https://github.com/node-opcua/node-opcua/commit/d9c9930eb0a5a80322393364cdc59e8dbcfc9cf2> :

AndreasHeine, 2022-01-04:
i guess this is what happens:

https://github.com/node-opcua/node-opcua/blob/90bb9139261a0edbb531081afcc6904c74e9ee52/packages/node-opcua-extension-object/source/extension_object.ts#L158

https://github.com/node-opcua/node-opcua/blob/90bb9139261a0edbb531081afcc6904c74e9ee52/packages/node-opcua-extension-object/source/extension_object.ts#L125

mohitkumaragarwal, 2022-01-05:
Thank you again, the link explains the decoding of extension object as opaque structure, but how can I make it to a meaningful structure? Any suggestion or example would be really helpful.

Best regards,

Mohit Agarwal
System Architect
R&D System Solutions
________________________________
AC MVI GI
Atlas Copco Industrial Technique AB

Visitor Address: Sickla Industriväg 19, Stockholm, 131 54, Sweden
Postal Address: 105 23, Stockholm, 131 54 Nacka, Stockholm

Phone: +46 (0) 87438290 - Mobile: +46 (0) 766961990
E-mail: ***@***.******@***.***>

Visit Atlas Copco at:   http://www.atlascopco.com<http://www.atlascopco.com/>
Follow us at:  Facebook<https://www.facebook.com/atlascopco> - LinkedIn<http://www.linkedin.com/company/atlas-copco> - Twitter<https://twitter.com/AtlasCopcoGroup> - YouTube<http://www.youtube.com/user/atlascopcogroup> - Instagram<https://www.instagram.com/atlascopcogroup/>
________________________________
Home of Industrial Ideas

https://github.com/node-opcua/node-opcua/blob/90bb9139261a0edbb531081afcc6904c74e9ee52/packages/node-opcua-extension-object/source/extension_object.ts#L158

https://github.com/node-opcua/node-opcua/blob/90bb9139261a0edbb531081afcc6904c74e9ee52/packages/node-opcua-extension-object/source/extension_object.ts#L125

AndreasHeine, 2022-01-10:
hey, sorry for the delay i was looking into:
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client-dynamic-extension-object/source/populate_data_type_manager.ts
but i have not manage to get it work without error!
@erossignon not sure why it tells me: "FIX YOUR APP BECAUSE OF SIMULTANIOUS REQUESTS"!? 
is it ment to be used from client side?

mohitkumaragarwal, 2022-01-10:
Hi,

No problems and thanks again. I have following the below steps:

  *   Node OPC UA: Taken the latest master branch code from https://github.com/node-opcua/node-opcua
  *   Node OPC UA Sample Client:
     *   Taken the latest master branch code from https://github.com/node-opcua/node-opcua-sample
     *   Replaced the node_modules in this folder with the newer version of  https://github.com/node-opcua/node-opcua
  *   Connected Unified Automation UaCpp Server with following endpoint and monitored a node below by updating the two variables in simple_client.js
     *   const endpointUrl = "opc.tcp://localhost:48010";
     *   const nodeId = "ns=3;s=ControllerConfigurations";
  *   From the command line, ran the following command: node .\simple_client.js
     *   I get the attached output from the command. Basically, I am getting the binary stream and now, I wanted to parse it as a meaningful structure.  So, I am not sure how to do that with NodeOPCUA.

Best regards,
Mohit Agarwal

https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client-dynamic-extension-object/source/populate_data_type_manager.ts

PS C:\DDrive\SourceControl\GIT_HUB\node-opcua-sample> node .\simple_client.js
08:57:13.548Z :opcua_client_impl             :247   Warning: endpoint_must_exist is now deprecated, use endpointMustExist instead
connected !
QualifiedName { namespaceIndex: 0, name: 'Views' }  
QualifiedName { namespaceIndex: 0, name: 'Objects' }
QualifiedName { namespaceIndex: 0, name: 'Types' }  
08:57:14.455Z :resolve_dynamic_extension_obje:64    Constructor =  BAConfigurations
08:57:14.456Z :resolve_dynamic_extension_obje:65    opaqueStructure =  ns=3;i=9
08:57:14.457Z :resolve_dynamic_extension_obje:66    opaqueStructure =  0x0a00000010000000416972436f6e646974696f6e65725f310000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f320000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f330000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f340000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f350000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f360000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f370000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f380000000000000000000052400000000000004e4010000000416972436f6e646974696f6e65725f390000000000000000000052400000000000004e4011000000416972436f6e646974696f6e65725f31300000000000000000000052400000000000004e400a000000090000004675726e6163655f31000000000000000000005240090000004675726e6163655f32000000000000000000005240090000004675726e6163655f33000000000000000000005240090000004675726e6163655f34000000000000000000005240090000004675726e6163655f35000000000000000000005240090000004675726e6163655f36000000000000000000005240090000004675726e6163655f37000000000000000000005240090000004675726e6163655f38000000000000000000005240090000004675726e6163655f390000000000000000000052400a0000004675726e6163655f3130000000000000000000005240
08:57:14.459Z :resolve_dynamic_extension_obje:67    00000000: 0a 00 00 00 10 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 31 00 00 00 00 00 00 00 00    ........AirConditioner_1........
 ...                                                00000020: 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 32    ..***@***.***@....AirConditioner_2
 ...                                                00000040: 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69    ..........***@***.***@....AirCondi
 ...                                                00000060: 74 69 6f 6e 65 72 5f 33 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00    ***@***.***@....
 ...                                                00000080: 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 34 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00    ***@***.***
 ...                                                000000a0: 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 35 00 00 00 00 00 00 00 00    ..***@***.***_5........
 ...                                                000000c0: 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 36    ..***@***.***@....AirConditioner_6
 ...                                                000000e0: 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69    ..........***@***.***@....AirCondi
 ...                                                00000100: 74 69 6f 6e 65 72 5f 37 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00 00 00 4e 40 10 00 00 00    ***@***.***@....
 ...                                                00000120: 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 38 00 00 00 00 00 00 00 00 00 00 52 40 00 00 00 00    ***@***.***
 ...                                                00000140: 00 00 4e 40 10 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 39 00 00 00 00 00 00 00 00    ..***@***.***_9........
 ...                                                00000160: 00 00 52 40 00 00 00 00 00 00 4e 40 11 00 00 00 41 69 72 43 6f 6e 64 69 74 69 6f 6e 65 72 5f 31    ..***@***.***@....AirConditioner_1
[... cut ...]
00000240: 00 00 00 00 00 00 52 40 09 00 00 00 46 75 72 6e 61 63 65 5f 38 00 00 00 00 00 00 00 00 00 00 52    ......***@***.***_8..........R
00000260: 40 09 00 00 00 46 75 72 6e 61 63 65 5f 39 00 00 00 00 00 00 00 00 00 00 52 40 0a 00 00 00 46 75    @....***@***.***
00000280: 72 6e 61 63 65 5f 31 30 00 00 00 00 00 00 00 00 00 00 52 40                                        rnace_10..........R@

})
   serverTimestamp: 2022-01-10T08:57:14.527Z $ 353.000.000
   sourceTimestamp: 2022-01-10T08:57:14.527Z $ 353.000.000
}
 closing session
 session closed
done!
PS C:\DDrive\SourceControl\GIT_HUB\node-opcua-sample>

closed 2024-07-16 by erossignon