node-opcua/node-opcua #1110: Zalgo Spam

scottbell, 2022-01-10:
**Current behavior**
A long infinite printout of spam text due to:
https://github.com/Marak/colors.js/issues/285

**Describe the bug**
Multiple packages in node-opcua depend up on the color.js library, e.g.:
["colors": "^1.3.3",](https://github.com/node-opcua/node-opcua/blob/master/package.json#L119)

**Step by step instruction to reproduce**
Install and run any application depending on node-opcua

**Expected behavior**
No infinite spam being printed out.

**Screenshots**
```
ț̵̶̯̮̱͍͕͍͔̙̩͍̱̥͉̟̤͈̙̤̘͗ͤͮ͂̄ͫ̋̆̓͐̀̕͟ͅͅe͔͉̣̠̝̼͖͉̣̞̞̗̭͎̖͖̭̪͇̦̻̫͍̖̰̙̜̗͎̳̺̐̾͊ͨ̃ͣ̄̎͒͌̃̒͢͡ͅs̴̞͕͙̯͇͚̹̪̺̦̠͖̗̫̩̺̠͚̘͈̻̰̹̙̣͇̻͇̬̼̦̟̮̗͉̹̜̝͈̭̩̝̪̫͔̭̠̳̮͉̠̰̤͔̣̖͎̟͙̩̙͕̰̈̈͒̾͌̓̏̏ͫ̀ͨ̉̓ͫ̊̆̄̎̀͘͢͡ͅͅt̡̧̢̖̹̝̩̹̱̬͕̯̦̯̞̠̼͈͚̭͉̙̰̻̯̙̝͍̬̪͉̙͔͓̩̞͍͚̖̳̝͓̻̻͚̞͖̺̫̞̯͔̲̰̫͍̻̰̬͇̰͊̋̄̊̈͆ͣͅi͉͇͚̰̯̩͉̦̗̹͙̼̣̖͇̲̹͇̺̭̮̓ͦ͆̏͑͛̎ͣͭ͌͆̄̂̿̍ͤͨ́͘ͅn̷̡̘̝͍͇̰̻̞̺̼͔̯̫͉͖͕ͧ͆ͪͭ̍g̷̶̡͙̣̼̯̞̲ͭ͌͂ͧ͒ͤ̒̈ͫͬ̀ ̨̤̤̣͕̠̝̰̰̗͚̜̤̘͈͙͕͈̜̝̹̳̾̃̔̍̂̊ͧ̋͐̉̽ͨ̿́̚͜͢t̶̢̨͕̦͈̭͇͙͔̥̪̩̯͔̝̫̱̣͎̠̘͔̱̯͕̺̭̟͈̮͓̖̘͙̜͙͎͇̙̭͇̤̫̰̱̺̫̮̫̝̳͛̉ͪͩ͊ͥͮ͗͒̑ͯ̅̃͜ͅͅͅͅͅe̢͎̫͍̥̫̻̻̫̖̬̹̫̗͇̻͚͉̜̣̙̫͍̰̬̟̺̥̗̦̫̻̙̭̺̫͚͕͉̯̲͎̹̰̦̱͎͚̬̗̹̲̻ͫ̋ͪ̅͒̄ͣ̚̕ͅͅs̷̛̲̱̪͇͉̼̫͕̭̩͖̠̪̟̝͚̣̘̬̲̻͇̲̠̲̩̗̞̄̄ͪ̀ͤͭ͋̀ͥ̿ͫ́̓͞t̛͔̦̖̳̠̮̲̙̳̣͇̭̖͚̫̫͇̙̖̳̞̣̘̩̼̼̯̱̫̟̠͚̮̻̥̥͎̹̱̰̪̩̪̗̝͎͚̦͖͙̮̱͙̭̳͕̟͑̎ͣ͑̏͂̂̔͋͊̍ͯ͊̀͟ï̷̖̝̹̱̻͎̻͖̳̩̬̼͕̰̳̱̼̜̜̮̞̽͗ͬ́͋ͯ̊̀͋̂́̀͝͝ǹ̏̑̐ͩ̌̉͊̒̌͋̉ͦͣ͊̐ͬ͛͢͢͡ ҉̤͕̗̣͔̬̦͇̗̙̦̣̠̣̭͓̥̝̬̖͖̺̹̜͎̭͇̬͕͉̦̞͕̗̪͕̻͚̼̯̱͖̺̤̣̞̮̱̜͍̜̘̭̞͙̘̜̘̭̜̤̭͓͡ͅͅgͩͨͤ͌ͤ͛̓ͭ ҉̫͎̲̹̹͎̣͓̻̲̮͙̭̞͔̹̀͢ ̢͕̳̺͈̳̳͍͙̥͍͚̟̞̘̯̹̥͖͉̪̺̪̬͈̤̲͕̮̝͈͎̜̼̎̇̈̀̾̓̑̎̀͂̇̑ͩͦ͂̀̀̚̕ͅͅͅt̸̲̗̝̩̠̦̦͍͍̙͙̱̱̩̪̭̜̣̩̦̘̺̜̰̦͍͎͕̦͉̭̩͕̪͓̱͙̼̮͔̟͍͇̗̰̥̤͕̖͖̳̖̯͔̤͇͙̣͚̺̰̯͖̥ͬ̽͌̋̓͐͛͋̄̄͐͐̄ͮ͋͜ȅ̮̜̜̪͙͕̪̦͙͇̫͕̹͙̖̫̞̭̟͈̦͔͇̥͕͚͓̫̪͙̥̳̣̮̙͎̳͖͙̘͕̣̞͉̆̅ͥ͆̏̈̾͘͟ś̷̸̩̱̩̖̮͕̪̱̜͚̼̹̦̫̘̹̖̥̫̠̲̥̄ͫ̈͗͛͑ͨ̏͒ͣ̑ͯͭ̏̃̉̄̓͟ţ̴̛̟͎̱̲̑͐ͫ͛̃͌ͤͤ̔̾ͯ̿͟͜ͅi̸ͭ̓̇̿͐̚ ҉̩̥̝̰̯̪͇̺̫̼̣̲͔̗̮͉̹̭̳̼̩̜̞̖̝́͞ͅn̶̨̫̯̱͙̘̟̳̱ͬ͌́͒͛̌ͯ͗ͫ̋̐ͭ̋̀g̷̸̶̢̼̳̠͖̞̪͍̫̫̰̹͍̙̩̞̙̲̠̰̫̺̪̜̞̣͓͉͇̰͓̗̼̥̙͈̯̺̹̬͈̪̯͔ͯ̈̿̈̀̋ͣ̅ͬ̀͘ͅ ̱̻͈̺̮̩̝̰̱͈̥̭̤̰̰̲̦̻͉͈̰̻͔͎̭̥͖̘̱̬̟̙̰̱̱̜̭̖̞̥̗͎͈̞̙͔̗̻̠̝͈̣̰͖̯̮̰͍͚͎̦ͬ̒ͬͯ͛́͡ͅͅͅt͌ͨ͊̆ͯͧ̋̏́͠ ҉͍͇̙̜̀͢e̶̾ͬ͗͌̉̍ͥ ҉̩̰̳̪̬ͅș̶͙̩͕̞͓̺̼͚̫̙̞̱̫͍̫̱̜͉͈̮͍͍̬̲̝̣̠̲͈̜̣̫̹̩͖̳̣̪̭̙͇̺̤̖̗͇̦̥̩̻̤͚̳̘̻͂ͭ̓̓͐͛̔̉̄̂ͮ̏͂̈ͩ͡t̛͓̲̣͑ͯͥ̀ͬͯ̾̑̇͂ͩͣ̄͗̈ͩͩ͜ͅi̵̭̝̮̣̱̯͇̭͖͇̪̯̰̞̤̦͉͙̦̺̗͔̱̝͇̜̼͍̫̯͇̳̺͍̯̯͚̖͔͉̳̳̫̝̰̜̐͗ͯ̅̀͢͞ͅͅͅn̝͎̺̳͉͚͉̲͈̲̦͍͙̥̖̙̲̬̼͚͈̲̙͓͇̘̟͇̗̟͎̠̥̞̦̘͉̟͍̗͇̣̯̤͖̙͇̞̮̙͈̲̙̹̩̻̑ͧ̃̿̒̐̍̏̓ͥ̾͆͌͆ͩ̐͊́̕̕ͅͅģ̵̡̭̙͙̭̙̬̻̠̗͉̲̟̩͙̣̖̠̘̮͉̰̞̦̘͇̻̩̲̦̞͇̪̗̮̣͖̮̳̪̳̮̱͔̤͓͖̙͌͐̈ͣ͂̉̇͒̒̍̔̃͊̈͛̉̚͠ͅ ̵̡̰̻̜͉̼̖̗̘͇͈̥̤͇̩͖͉̩̖̜̱̫͉̳̘̘͚͕̺͎̞͇̺̰̭͈̝͈̥̙̩̖̪͙͔̯̫͇̜̜͚̂͑̅ͭ̽̑̈̔ͫͣ͊̅ͭ̂̕̕͝t̛͛͒̃̀̿̃͒͒ͮ̈͐̀͋͠ ҉̬͍̘̥͎̹̝̬̘͔̮͈̜̹͓͇̙̣̯̩̳͍̥͖̳͕̰̝̰̼̣̪͙͙̖̺̫͇͚͎̤͚̩̮͖̺͟ͅͅẽ̵͓̻͕̹̘̙̼̹̟̹̞͖͈̣̖̱̳͉͉̅̊͋ͣ͘s̎ͬ͐̅̓ͭ̿ͪͩ̏͋̔͗͂ͣͥ̓͒ ҉̵̢̲͈̹̞̝̱̘̝̳̠̱̦̘̥̼̫͚͕͚̱̪̫̼̻͕̞̰͎̮̬͕͉̯̼̟̕t̷̶̢̖̜̱̖͐͑̂͋͟͝i̢̛̜̯̜̠̖̯͚͈̺̺̮̞̯̬̝̥̫̯͙̬̥̠̰̮̣͓͖̯̼̥̱̞̩͍͖̙̗͖̥̠̖̮̬̞̳͙͖̝̦͙͍͈̱̅͑́͆̀̇̂̐͒̍̂ͩ͌̽̀̿̈ͯ̕ͅͅņ̵̜͚͙̘̝͈͖͍̹͔̫̙̦̼̙̞̲̳̼̖͍̺͍͔̻̫͈̺͖̳͇̯̲̞͇̘̱̘̪͍͉̺̙̘͎̮͓̱͙̰̞ͮ̆͐̓͌ͧ̒ͥ̏̾͟͢ͅͅg̵̨̡̻̰̙̲͍̭̫͇̦̣̥̥͈̖̟͍̙̱͚̳̗̞͔͍̩̙̹͕͎̬̬̜̠̗̤̯̫̻̻͚̗̅ͫ͆ͬ̓̎ͪͯ̏ͨͯ̽ͫ̊̈ͭ̀̚͢ͅ ̨͙̦̭͍̻͕̯̤͉̠̖̖̪͚͈̜̠̯̺̖̩͔̱͕̜͙̮͚̬̱̻̽̉͒͊̔̂͌ͨͤ̂̽̎ͥ̉ͧ͠t̵̻͎̙͎̝͉̬̰̲̫̭̰͓͚͔̘ͯͮ̈͌ͪ̀͘ẽ̢̡͍̪͇̤̪̠͇̲̹͎̦̣̗͖̙͂ͫ͌̈ͬͩ̑͊̈̋̀͐͘s̡̪̼̥͙̰͖ͤ̈ͮ̅̏͐̀̚ṯ̸̡̛̗̜̬͍͇̻̰͍̝̜̯̖͓̲̖͕̙͍̯̫̰̞̘̱̙͓̙̺͈͇̱͇̣͈̼̺̰͍̜̠̘̪͛̍ͯ́͂ͫ̅̓͆̒̎̕ͅͅǐ͂̍̑̈ͦ̚͡͝ ҉̳̲͕͈̖̣̠̪̼̜nͩ͆̈ͣͥ́̾ͨ͌ͨ͑̌̾̔ͩͬ̚̕͟ ҉̫̣̟̙̥̩̞͕̺͖̫̬̪͓͓̫̘͈̻̗̮̗̩͖̜̖̬̠̹̱̜͔͖̝̹̗̖̞̥̳͇̯̜͍̟̮̜̺͔̩̬̪͍ͅǧ̵̺͓̯̗̬͍͙̞͎̜̗̙̗̳̰͎͓̼͓̟̫̙̦̩̹̣̹̹̪̹͍̥͚̺͈͎̠̖͈̻̖̦͇̝̖̬̰̱̬͇̪̬̳̇́̌̀͘͢ ̡̡̢̭͕͔̩̰̞̹͔̮̣͈͙͔̦̘̮̲̖̻̘͎̮̠̲̩̮̱̰̟̭͕̰̬̫̦͇̗̮̼̜͈̞̠̥ͮͮ͗̔̅̋̄̔ͅt̶̢̡̧̬̮̩̼̥̪̫͖͈̠̯̹͕̺͖͔̲̦̯̦̻͕͓̲͎͈͓͍̽̀́͒̆̈̈̂̍͂̄ͯ̀ͫ̑ͣ͡ͅͅͅḙ̷̼̤̩̭̗̼̼͓̘͉̬̩͖̲͈̟̥̳̙̲̙͓̭̟̣̜͕͖͙̗̜̘̝̣̦̘̯̺̫̗̦̹͙͉͕̤̘͉͚̠͗̅̌̉̀͜͠s̞̝͕̥͚̩̥̭̘͍̘̭̰̙̙̥͖̞̯̬̲̝̖̹̰̰̙͇̼͙̦̰̻͉̭̜̫̗̺̝̭̖͐̐ͬ̋̔̉ͨͦ̏̀̚͟͞͝t̶̅̉̿̆ͥ̔̂̂͠ ҉̞̺̣̣̲͖͍̤͚̫͔̙͖̻̯͙̻͖̘͍̯̹̠̬̘͔̗͖͓̻̖͔̫̼̣̫̘̱͕̥͙͕͎̳̘̘̩̳̠͉̼͍̤̬͉̟͙̘͉̩̖͘ͅͅͅͅi̵̧̧̳͕̟̦̼̥͍̥̮̲̱͓̙̰̟͇̤̣͉͈̝͍͖̤͙̯̼̦͙̖̫̲͇̻̣̰̮̞͈ͣ̏ͭ̾ͫͫ̀̽̐̉͆̉ͧ̔̔͂̑̒ͅn̶̷̻̘͙̱̱͙̣͎̞͕̝̮̝̯͚̲͒̾̏͆͒͐̏̈̇͒ͭ̓g̮͎̰͇̬̜̯̤̥̝͉̗̻̳͚̩̥̫͉͎̱͔̰̯̹̪̘̥̬̜̗̞̬̯̝̲͔̟̞̬͚̘̘̗̹̬͌͋ͨ̂̌̇ͭ͊̄̍̈́ͥ̀͘͜
ţ̛ͭͧ̏̈͑ͬ́ͣ̓̌̽ͪͥ̀ ҉̵̞̹̘͍̺͚̼͚̬̮̱̘͓̝̣̗̫̭̙̜̘̰̤̳͎̤̲̩̹͙͖͈͉̣̳̝̬͉͙̠̫͕̯̞̤̺͎ͅͅȩ̵͖͕͚̮̘̗̭͚̝̭̖͉̖͚͕̥̜͇̱̰͙̖̮̺̤̱̗̤̜͎͓͔͔̦̪̬͈̱̫̥̝͈̭̫̱̟͉̰̤̮̰͓̩͇̖͙̘̙̤̃̽̓ͭͨ̋́͟͡ͅͅşͭͫͦ̈̎ͮͬ͗͐ͮ͂̋̂ͥ̚ ҉̘͍̗̫̪̮̘̰̝̫͙͔̳͙̥̻͚̻̞̬͍͎̭̼̗̜̜͔͓̞̝̹̘̼ͅt̴̷̢͓͇͍̪̬͓͙͎͎̥̱̞͚̞͖̣̬̬̪̙̹̩̭̍ͦ͌̉ͣ̈͋ͧ͂͑̊͌ͣ͒͊͜ͅį͕͎̲̫̏̋͒͊̆̐̀͘͠ṋ̶̗̪͔̥͕̰͍̤͖̮̞̖̣̪̞̪̳̖̲̠̰̞̥̲͎͙̯̥͎̥͔͖̼̜̙̞̲͔̫̝̃̽͆ͮ̑ͫ͆͋̇́̄ͪ͌̐̓͢ͅĝ̛̫̣̳̯͍̠͚͚̘̫͕̤͉̬͓̹͓̭̯̝̞̱̪͓̠̤̙̗̹̳͍̪͈̤̗̗̖̝̪͍̱̥͉̠̬̼̍̐͗ͣ̈̕͢͞ ̷̴̧̛̭̼͔̬͉̭̟̖͎͙̟̩̪̥̺̰̜͚̮̼̜͖̝̪͚̹̲͙̟͉̣̲͈͈͕͉̳̳̻̯̮̟̱͕̱̯̯̟̝̘͍̙̩̘̞̦͎̘̺̼̮̠̇ͤͥ̆̏̑̚t̴͓̠͖̙̱̯͎̳͇͇͙͙̦̰̤͈͚̣̘̲̹̣͔̜͙̮̖͒̍̈ͯ͗̐́́͠e̸̸̯͇̘͈͈͚͔̗̞̣̲̲̹̦̙͎̥͚̭̼̭̱͙̥͔͕̫̜͍̼̮͇͉̰͍͔̝͚͔͔͖ͦ́̽̽̆̒̔̓̆̃ͥͯͪ̓̄ͪ̄ͧ́́ͅͅṡ̶̴͓͇͔̝͙̥̯̜͈̣͎̲̪̮͍̩͎̯̤̼̙͇̣̯͉̝̲̬̠̬̺̗̜̼̫͈͖͚͍̣̞͈̘̟̹̜̳̰̩͗̐͗͊̂̎̚͞ͅͅͅͅt̷̬̦̘̟̙͓͖̪̫̬̖̣̞͈̹̼̘̦̳̺͍͓̳̙̯̳̗̬̘̟̪̝̞ͩ̌̄̌ͤͮ̃ͮ̃̏͗ͣ͐͐̏ͨ̿̅͘̕ͅͅi̵̵̤͔͉͙̞̙̬̺̳̭͉͉͙̹͈̭͖̥̰̳̫͍̻̹̝͚̙̟̟͚̰͖̬̠̜͖̯̭͈̦̬̟̣̯̿̔̾̌ͯ͐̽̀ͥ̑̐̀̕͝ͅͅn̛̙͓̻͙̫̖̤̬̟͖̰̪̤͈͓͍͓͎͕̩̣̩̫̟̳̫̮̹̺̭͙̺̹̬͈͔͚̟̺̦͓̘͎̆̌̈̇͊̉̊̉̄ͥ̍̋̽̚͡ͅg̶̶̸̝̱̞̯͓͎͎̝̼͖̺̳̞̟̻͉̫̼̥̦̘̟̓͌͒̈̎͗́̒́ͤ̏͒͆̓ͩ͆̒ͭ͟ ̴̨̜͕͍̝̩̘͍̩̖̜͉͎͕̺̘͔̓ͭ̀ͧͭ͑ͯ̅̓ͥ͊̊ͣ̔́́̚͠ͅt̴̢̼̱̦͍̫̦̳̲̪̳̦̟̪̬͎͖̟̪̪͎̖̱̪̤͍̲̟͓͍͉̣̝͈̞̭̫͇̰̙̘͇̫̥̻̣͖̗̲̠̙͔͓̤̟̳͒̑ͣ͐́̑͒ͫ̚̚ͅe̵̸̡̩̺̗͕̯̤̥͓͓̣̩͍͓͓̬̼͚͖̗̜̤̖̥̙̩͕̜̭͙̯͔͉͇̰̪̝̮͕̯͌̏̀̀̋̃ͯͬ̏͡s̵̡̧̛̭͕̦̦̮̭͚̙̗̝̭̥̞̹̤̹̞ͩ̈͊ͧ̐̑ͧ̓͋̈ͣ̋͞ͅt̷̢̗̲̩̫̙͎͚̊́̓̅ͮ̋̍̎͐ͦ͜͜͞i̷̴̯̗̱̳͍̞̮͎̞̣̤̝̬͓͇̯̬͓̤͇̤͕͙̞͕ͯ̌ͨͧͮͯ́̀n̶̢͉̦̳̘̲̤̫͖̪͇̼̱̫͇͇͕̻͔̮̺͔̪̦̗̱̹̬͍̻͈̥̺̼͓͙̱̙͙̘͉̤̙̭̭͖̞̱̘̳̗͓̖̘̦̥̺͍̲̬͉̦͕ͮͭͣ̊̂͛̔̆ͮ̅͐̄́͢͡ͅͅg̴̛͖̝̳̱̻͙͙̻͔̮̗̱͕̠̙͉̯̲͇̼̞̗̙͕̱̘̟̠͖̖͖̙̜͖̖̝̻̫̩̲̝̘͔̖̣̜̣̩̩̟͇̖̦̻͕̬̑͑ͬͭ͐̂͊̔̎͊ͪ̽̀̚̚ ̵̵ͮ̔̂ͪͨ̐́͢ ҉̦̣͙͚̣̰͓̜̮͚͍̟̘̻͕̫͍͍̬͓̠͍̬̞̖͖̦̹̩̹͚͔̫̭̖͔̘̱̼̮̺͈̮̱̳̯̹̼̮͔͙͉͚̟ͅͅt̸̡̡͔̼͖̥̭̝̦̘͎̰̯̺̦̞͙̩̫̼͕̱͔̤̳̣̬̟͌͒̔ͬ̅̅ͨ͂ͫͧ̊ͥ̎ͪ̚̚͝͝ͅͅͅè̶̶͉̹̣̩̬̯̮͉̠̺͉ͯ͒̔̋ͦ̍͊͋͜s̵̸̸̨̳̟̤̟̩͙̺̜͔̰͉̰̦͓̮̟̹̩̩̺̲͓͉̹̋̆̅̋ͥͤ̆ͤ́̈͢ͅͅt̷̡̠͓͙͖̖̼̖͍͇̺̤͎̠̝͇̗͉͔̙͕̠̮͙̤̞̟̠̼͙̜̺̹̹͎͚̥̙̥͚̳̭̩̝͇͓̖̳̙͎̥͖͎̜̣̯̺̖̲̱͔͔͍̿̽͑̀ͦͦ̆̎ͦͮ͌ͮ͗̅̍̋́̚͡ͅͅͅi̵̿͐̓̾ͧͩ͑͌̀̀́͞ ҉͍͍̻̳͓͚͉̤̬͙̭̘̙̱͔͖̯̩̲̯̹̘̺̩͈̰̺͇͙͙̭̳͚̯͎̰̩̰̙͚̩̟̳̲͕̯̗͉̫ͅͅn̨̢̧̄ͭ̂̏͛͆̄͂̏ ҉̻̦̺̝̫̱̱͖̞́g̴̴͖͈̬̰͍̼̪̺̠̜̦̤̣̭̯͓̫͈͓̟͓͈̰̥̥̱̪̙̼͚̠̮͚̯͖͖̬̻̦̜͉̻̟͇̝̣̗͖̦͇̟̟̗͚̼͔̺̻ͫ̍̂̇̉̓̆̋̏̈͑̈̔̈̾͐͘͝͡ͅ ̨̗͍̗̖̮̗̦̗̣̰̦̯̯̭̞̗̲̞̙̗̫̯̠̺͎͔̻͈̤̟̪̘̣͇̣̗̘͓̰̝̰̘͎̤̻̰͈̠̀͌͛̈ͥ̈̏͟͝͝t̨̟̲̙̼̮͍̰͙̝͕͕̺̣̫̪̲͎̗̲̘̬̮̜̥̳̺̼͔͔̲̻͖͈̻̥̰̏̓̈͑ͧ̄̈̇̓ͩͪ͑̔̐̕ͅͅͅę͇̜͍̹̖̟̟̱̣͎̜͙̮͈͓̯͖̫̝̦̦͈̻̦͒̃̌͆ͨͩͤ͒̂̃̕ͅs̴̸̢̡̯͔͔̟͇͖̟̘̞̖̩̱̖̝͓̭̳̞̹̘̅ͯͩͬ̎̾̐͆͒̇̋̍̾̂̇͌̚͝t͌̊ͭ̓ͣ̎ͪ ҉̷̪̣̬̱͔̲̪̥̳̘̱͎͈̞̻͇̹̘̩̣͓̮͎͉̤͔̝͔͇̠͕͔̱̺̠͖͇͔͈͈̩̝̤͙̹̩̺͍̳̯̟͖̫̳͕̞͎̻͇̜͖̫̫̀̀ͅͅͅï̮̪͙̳̟̮̖͚̟̗̻̣̰̤̬͍͙̬̳̗̞̻͕̮̥̟͕̖̺̫͉̠̟̭̺͎͔̲̘̙͉͙̼̞̲̤͓̳̣̰̮̙̪̘̜̲̤̳̭͈̖̝̈ͧ̉̉̔͗́ͤ̿̔̈̊̈̀͠ͅͅn̖̳̞͓̩̠̻̩̗̹͇͚̘͍̯̖̪̻̻͕̬͍̥̘̹̖̲͎̝̲̼͕̜͌ͯ̓̅̾ͨ̀͡ͅͅg̴͕̻̜̙͛͑̊͋̔ͭͨ̽͋ͭ͗̔͗̏̾ͨ̈ͬ̀͜ ̝͈̘̫̦̺̰̹̣̗͚̝̝̘̣̮̰̝͍̦͕̑ͩ͂ͮ̽̑͋̈̚͢͜͡t̵̴̡͕̟̠̳̜͉̟̪̘̥̪͈̼̦̫̱̣̯̯̮͇̺̹̻ͭ͊̇̇ͦ̓̓ͫ́͌ͪ̄̌͂̕͟ȩ̤͈̝̫̯̖̻̹͈̪̮̱̭̺̹̪̝̘͍͕̳̞̻͎͈̘͉ͧͧ̈̿ͦ͆̕͞͡s̴̶̞͚̗͈̗̤͕͍̲̙̩̞͖͈̼͖ͣ̌ͧ͂̃̏͐͆̾̈͂̓͊͊̐͑͟͢͠ͅt̶̪̲̝̱̯̬̫̱̼̘̗̼̪͔̞͓̜͔̤̯͖͈͇̦͉̻̝̪̟͙̼͓͔̠͔̲̩̥̆̔͑͗́͘͜į̶̢͓̦̪̺̤̣̼͍̪̖͚̻̺̹̯̮̺͈̦̖̩̠̝̳̦̠̺̽͐ͣ̍́̍̾ͭ̂̐ͤ͗̉n̵͇̝̙͕̰̱̥̳̙̙͎͉̫̖͎͈̪͕̯͉̪͍͇̱̟͕̦͇̖͙̝̬̮̱͔ͭ̉͋͂̿͛ͬ̎͛̽͌͗̎ͦ͞͡ͅg̡̮͇̰̝͇̼͈̼̘̻͍̖̣̞̰̲̟̮̦͚͍̲̭̗̗͈ͭ͐͐̎ͯ̄́͆ͯ̉ͥ̑ͭ͛͘ͅ ̷̷̨̱̻̳̼̱͙͈̫̪͓͉͔̻̯̗̼̫͎͍͉̬͔̮͉͈̱̘̱̯͙͇̟̰̞̥̻͈͓̭̟̠̬̖̝̰̦͎̺͈̝̝̜̼̜̋ͣ̎̌̓ͤ͐̋͒͑ͥ̌ͬ̊̅ͭ̆ͤͅͅt̵̸͇̹̪̳͉̻̥̣̫̠̯̬̪̺̞̮̻͚̣̹͇̫̪̪̲̠̘̱̘̪͈̮̦͖͓̞̪̙͙̳̲̖͓̝͍̪̘̰̦̗̹̪̘̮̣̪̹̱̥̲͍͓͚̰ͭ̓͛͗͆̈͊̾̀̀͜ͅe̛̻̳̼̼̥̤͖̬̖̦͙͔̠͈̱̼̱̫͇͔͚̺̦̜͇͈͈̼̘̮̘̤̥̯͙̝̜̟̳̮̼̫͇̣̠̗̱͖̹͚̼̊̏͊̽̾̿ͨ̎ͥͨͮ͜ͅͅs̢̢̛̯̬̯͕̖̖̹̖͚̫̞͔̘̥̭̯̗̳̳̺̤͇̬̰̤͔̼͕̻̘͕͖̟͔͚̮̱̯̭̫̖͍̰͚̪̝͎͓̬̈̇ͥ͊̓ͩ̓͋́ͩͮ̉͛̕ͅt̛̝̰͖̪͎̻̣̜͔̲̹͔̻̱̤̳̤̠͇̭͈͔̲̖̤̎̆̎ͥ̋̂ͥ̃̎ͣ̅̽ͯ́́ͅi̵̦̦̟̣̘̟͓̜͉̥͕̘̠̥͔̞̗̮͓̠̯̙͈͍͔̠̫̥̪̪̜͚̭̭̤͕̘͙͇͙̮̹̹̪̥̫̘̬̹͈͕͖͎͓͈͎̜̝͍̰̪̓ͨͩͬ̂͆̒ͣ̀͟͞ͅͅͅņ̲̺̻͇̙͓͕͚̱̞̲͉͉͔̠̘̬̹̰̞̹̤͇̲̠̪̱͈̞̤̻͉̳̤̣͉͆ͣͭ͂ͤ̽̀͜ͅg̸̵̭̫̜͕̱͕̝͎̯͙͎͈̟͔̝̲̹̟͍̲͍͈͚͍͚͙̟̭̰̯͇̩̗͉͍̙̺̥͙̩͚̯̬͎̿̊͂̃͒̒̎͑ͨ͋͒ͦ͒̓̓ͥ͛ͮ̚͘͝ͅ
t̳̻͔̩̤̺͕̘̰͇̪̫̳̺͓̤̘̹̟̫͂ͣ̋̊͌̾̉̎ͪ̿ͩ̄̐̀ͮ́͞͠ͅͅͅȩ̵̸̖̪͎̰̙͙͎̗̙̮̖͕̖̱̳̭͖͙̳͔̦ͥ̈̔̉̆ͯs̛̮͖̖̗̳͚̳̜̻͚̬̝͕̤̼̮͎̼͚̪̹͕͙̹̘͑͗ͬ̈̿̾̉̈ͩͪ̕t̴̡̛̤̩̼̦̞͈̩͈͔̹̳̻͎͕̤̙͈̗̭̫̮͉̫̙̖̳͑ͥ̈͛̿̈ͨ͆̉̃̒ͭͯ̐̍̃ͥ̐̚̕͟iͥ̍ͣ̓͆̄̃̽͋ͪ̐̚͠͡ ҉̪̬̳̯͖͖̀n̶̡͙̬̣͖̺̲͍͔̯̞̺̲̞̘͔̫̮̱̞̰̲̟̭̺̏̉ͯ͒͛̌̅̽̚͠͝ͅg̸̵̵̢̳̠̪̹̲̼͙͍̭̹̥̥̟͕̼̓ͪ͊̈͐͂ͯͣͪ͂̈͘ͅ ̸̢͔̱͔̩̗͇̜̠̞̬̯̤̗̫̪̮̻̯̩̟̣̥̯̜̮͉̦͈̞̯͕͚̰̺͖͎̰͕̻̦̙̳̪̯̘͕̻̥̞̙͔̜̝̙̩̹̫̪̥̙̭̬̝̖͊̽͒͗͑ͣ̔͋ͦ͌̆̋ͅͅͅt̴̢̮̟̳͕͚ͮ̓ͯͫ̏͘ͅḛ̡̨̻͙͎̘̹͚͉̞̬̼̞̺̗̤̟͇̉͑̌̆ͯ̓͋̒̏ͤ̋̂̉̈͒̀ͅs͎͔̞̖̮͕̖̗͕̬̥͈̭̰̱͓͉̣̤̠̰̣̻̫̱̪̰̹̲̘̝̖̻͇̺̗̝̳̱̻̭̭̪̼̣̖͖͎̭ͧ̇̓͋ͧ̎́͞t̷̡̻̹̹̤̖̗͚͈̒̐ͤͧ̔ͣ͜͞ͅį̴̨͎̘̣̺̞̙̤̗̦̻̘̦̖̦̝̻͇̃ͫ̓ͧ̆ͮ͌̌̎̀͑̃ͤ̍ͬn̶̨̧͈̯̱̜̖̙͎͙͇͓̯͚͇̣̱̠͎̖̼̘͕͉͈̯̮̬͙̳͈͚̖͎̮͇͙̯̹̘̯̘̦ͯ͛ͨ͑ͪ̂͑͋̆̇̉ͣ̇ͦ͝ͅͅgͭͤ͗̀̋̈ͪ̋͌͞ ҉̳̼̺̗͔̭̖̟̦͇̖̖̹͇͈̮̞̤̦͓͙̲̩̝̕ ̨͖̜̮̝͙̻͔͖̩̮͎͚̤̮̼͎͎̜̬͓̫̻͙͉̝̩͉̮̲͉͈͈̭͙̰̞̺̭͚̮̳̮̜̲̮̫̝̤͙̭͈̯̺̹̲̓̔̔ͫ̏͝͡͠t̷̴̵̟̜͓̞̘͎͖̘͖̝̜̫̞̫͎̪͎̯̭̠̦̼̫̦̩̻̫̱͚͉̞̗̪̩͈̦̦̱̥ͬ̽͂̃̏̐̎̿ͮ̈ͫ̆̓͐̂ͪ̆ͫ͗ͅͅé̘̟̖̩͙̹͓̫̦̘̯̻̩̲̤̜͎̦͈̫͇̪̲͔͕̜̤̖̫̟̹̠͙͇̻̣̂̌̈̾̈̃ͥͨ͜͟͡ͅs̨͕̗͙̯͓͇̲̳̘͉̘̬̟̤͉̞̥̺̯̦͕̺̞͇̑̔̊ͫ͌̇̉͒̏̏ͩ̽ͩ͘͝͡t̴̴̵̛̘̗͔̘̙̪͈̲̱̝̳̹̠͙̮͊͑ͮ́͊͂̄́̈ͣͧͬ͆͠i̶̱̪̝̝̲̩̼̻̓ͦͪ͐ͪͮͥͬͨͪ̊ͦ͆ͭ͂̕͢n̵̡͍̻̰͓̜̫̓̈̊̂̍ͩ̓͑̂̉ͮͦ͘͞g̨̛̲͕̦͍͚͓̫̤͔̤͎̦͖̭̝͑̎͛ͯ̾̏ͥ̓̌̚͢͡ ̴̡̦̗̹̖̜̘̠̭̻̳̹͚̘̭͚̙̹͇͈̥̜̝̜̰̰̦̥̣͚̗͙̯̣̗͙͓̰̫̝̜̮̩̻͎̮͈̭̟̲̪͎̰͕̬͓̖̟͇͔̮͓̰̘̌̂̏̍̓͗ͫ͆ͭ͗ͧ̏̽ͩ̂̑ͦ́ͅt̸̴̩͈̣̝͔̱̜͎̪̦̭͙̬͉͔̼͈̫͍̰͕̞̼̠͉̥̼͖͖̳̹͓̻̤̮̭̪̹̬̭̪̥͓̜̜̬̝̙͔͕̰̰̬̟̬̼͍̦̣͇͖̭̹̥̗͆̐̓̾̈͂̈͗ͦ͌̍ͬͦͨ͋͗̽͝ę̶̧̖̮̖̻͙̙̮̻̥̪̺̣͙̦̻̣͓̦̦͖̲̻̹̭͈̯͖̑̌̄̎̋̅̏͜͡sͥ͌ͮ̂̈̚ ҉̢̖̰̫͕̘͚͞͝t̷̨̡̡̺̻͎̻͉̮͙̏ͬ̑̈͒̽̈͞i̶̧ͬ̆̽̓ͦ͑͗ͯ̐̒̈͒̀̊͌̉̿̾͛ ҉̧̱͎̗͈̹͓͎̞͚̱̭̟͎̲̺̦͖͇̘̫̺̥̺̝͚͚͖͈̻̞͎̥̙̼̟̲͕̝̯͇̘̦̝͚̗̞̭̤͇͓͖̙̬̟̼̝̯̲̩͇͖̩ͅn̗̬̝̫͖̻͓͚͙̩͎͙̜͉̭͉͍͇̞̉̀̽̅ͧ̀́͟ͅͅg̸̢̮̝̜̳̮͈̗͈̝̠̰͕͙̗̖̪̦̭̺̫̩̪̳̺͔̩̫̠̰̣͇̻̳͉͕̘̰̳̳̱̺̺̬̩̪̦̮̟̭̫͔̰͉͍̘͙͇͕͉͚͍͚͎͉͓̰͆ͥͥ̈ͦͪ̀̈͒ͯ̊ͮͫ̓̊̄͌͆́̚͝͞ͅ ̵̢͍̪̱̟͓͈̞̺̞̞̣̮͇͙̞̩͚͉͈͇̬̮̜̪͕͇̗̰̞̦̂ͫ̀̓͊̐͗̉͗̎͛ͅͅt̷̛͗ͮ̌̄ͪ̒̈͋͛̐̚ ҉͍̲̩̩̹͚̳͉̖ͅe̸̢̛̬̦̪̰͉̱̯͖̖͔̼̟̜̹̤͕͓͔̩̭̜̟̱̦̰̻̠̘ͭ̆̇ͤ̈ͯ͆ͣͩ̚͞͞ş̵̡̧͖͈̞̠̜̝̻̦̳̯̝̺̫̝̹̜͉̺̣͈͈̝̞͚̪͖̗̙͎̣͍̞͚̻̦̫̘̗͖̗̤͂̊̌̿̀́̀̚ͅͅt̶̤̮̮̘͕͚͚̲̹͎͇͖̪̱̤͙̣̳͇̫̬̻̼̝̻̭͕̲͍̫̯̜̮͖̰̭̥̱̥̣̩̦͎̪̖̪̤͎̯͇̮̖̠͖̬͇̩̗̝͑ͬ͂̀͛ͨ̔̈͟i̷̯̬̝͚̦̥̻̣̟ͬ̾̅̋͛̆ͦ̓ͣ̑ͫͨ͐̌̈ͤ̿̀̚͠n̸͚̘̝͈̰̤̩͚͈̦͙̯͈̗̮̙͓̺̯͈͉̺̺̮͈͚̤͔͈̘͖͌͊ͦͫ̽͐̊͛͋ͤͤͫͮ̚͢͜͡ͅͅģ̛̱͈͙̦̹̣̻͎̲̩̻̻̱̗͚̮̞̥͚̙̬̞͔͙̹͇̗̳̣͎̖̗͇̱̼̫̪͈͎͓̜̻͖͖͔̹ͩ͐̾͑̐̑ͭ͑̑̓͗͛̐ͥͦ̚̚͡͝ͅͅ ́ͭͤͪ̅ ҉̹͈̠̼̯̠͎̲͉̲͈̞̮̩͉̻͓̞̘̯̩̪͔̼͍̬͈̹͍͕͇̘͈͉̪͕̤͎̭̬̞̻̮̦̯̤̗͈̥̻̯̗̜̘̥̖̯̕͘͟͡ͅͅt͈̭͍̮͇̙̝̫͙̲̻̬̰̺̥̝͛ͦ̓ͣ͊̿͛̑ͮ̈͑ͯ̅ͭ͋ͦ̈̆ͮ͡͝e̢̻͓̳͇͇̬̫͉̲͇̯͖͓̝̼̘̥͓͖̭̹̣͖̺̯͈̫̩̗̣͎̐ͮ̈̀̒̄͝s̴̰̱̬̞̼̰̙͕͓̝̰̫̘͖̝̪͎̯̭̼̮̬̀ͩ̾̏͆̉̋̅͒ͭͯͨ̈͆̇͐͢͞ͅͅt̸͈͈̺̠̪̙̟͍̝̫̜͉̲̤͕͍̲̹̦̺̟̫̫̟̗͙͚̹͚͉͕̼̯̺ͯ̑ͨͬ̆͊̑̿̍ͧ̔ͨ͒̉̔̚͝͠i̷̢͎̥͔̗͚̟̯̜͕͔̜̭̳̣̖͙̞͙̣͉̰͖͇̱̺̬̤̳͔̱͓͔̹͖͕̥̯̖͎̮̩͔̣̜͖̠͚͚̮̹̯̗̲̻̲̦̱̻̗̦ͯͪ͌̑̐͐̄ͤ̑̃ͨͦ͑́̚͜͝͝ņ̥̤̹̩̝̩̪͖̥̥̬͉͉̐͆ͦ̓̉͂̅̒̀̕͢g̶̛̣̤̺̺̜̖͖͎̼͉̲̝͚̪̲̺͇̘̭̺͈͍̀̊͗̓̅͆̓͆̈ͭ̽ͮ̌ [... cut ...] ̤̣̳̥̳͔̘̻̬͖̪̠̝̤̼̼̹̗̬̝͙̝͔̻͈̰̩ͅg̴̷͎̣̘͕̱̰̪̪̠͔̙̖̝̹̣̫̠̲̺̘͆̆͂̇ͩͫ̊̑͂͗ͮ͑̿͌͌̃́

```

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- (X) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- (X) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- **Node:**
  `node --version` All

erossignon, 2022-01-14:
Hi Scott, I can't reproduce.
Can you be more specific ? 

````
dependencies:
@cucumber/cucumber 7.3.1
├─┬ cli-table3 0.6.0
│ └── colors 1.4.0
└── colors 1.4.0
cli-table3 0.6.0
└── colors 1.4.0
colors 1.4.0
node-opcua-pki 2.13.0
└─┬ cli-table 0.3.9
  └── colors 1.0.3

devDependencies:
literate-programming 1.1.0
└─┬ literate-programming-cli 2.1.0
  └── colors 1.4.0
litpro 2.1.0
└─┬ literate-programming-cli 2.1.0
  └── colors 1.4.0
tracer 1.1.5
└── colors 1.4.0
````

scottbell, 2022-01-14:
I think it’s been repaired upstream from us, so no need to worry now! I’ll close the ticket.

closed 2022-01-14 by scottbell