node-opcua/node-opcua #1115: require('node-opcua') v2.64.0 fails due to missing file `nodesets/nodesets`

fedefrancescon, 2022-02-08:
**Current behavior**
When doing a fresh install on `node-opcua`, upon requiring the modules, node throws an error complaining about a missing file `node_modules/node-opcua-nodesets/nodesets/nodesets`

**Describe the bug**

```shell
> $ echo "require('node-opcua')" | node
/tmp/TEST/node_modules/node-opcua-nodesets/dist/index.js:14
            throw new Error("Please make sure that nodeset can be found in " + path.join(dirname, "../nodesets"));
            ^

Error: Please make sure that nodeset can be found in /tmp/TEST/node_modules/node-opcua-nodesets/nodesets
    at constructNodesetFilename (/tmp/TEST/node_modules/node-opcua-nodesets/dist/index.js:14:19)
    at Object.<anonymous> (/tmp/TEST/node_modules/node-opcua-nodesets/dist/index.js:50:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/tmp/TEST/node_modules/node-opcua/dist/index.js:106:29)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
```
It seems that the file `nodesets` inside `nodesets` folder is required but doesn't get created upon installation.
Is this required to be manually done? In previous versions this seems to not cause any issues.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. `mkdir /tmp/test-opcua && cd /tmp/test-opcua`
2. `npm install node-opcua`
3. `echo "require('node-opcua')" | node`

**Expected behavior**
`node-opcua` loads as expected.
Or have a quick guide if the creation of that file need to be done manually, I've tried to look at the doc but probaly i've missed something.

**Screenshots**

**Gist**

 
gist:

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- () my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- (X) I am using an application that uses node-opcua

    - ( ) node-red
    - (X) other : our server application directly uses `node-opcua`

- Device: PC
- OS version: Arch Linux / Debian Bullseye

    - ( ) Windows : version : ______
    - (X) Linux : version : Debian 11
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
v2.64.0

- **Node:**
  `node --version` = v17.3.0
  `node --version` = v14.19.0
  `node --version` = v12.6.0

Graos, 2022-02-08:
where did you get hold of V2.64.0 when the latest release of node-opcua is [v2.63.2] ?

fedefrancescon, 2022-02-08:
On public npm: https://www.npmjs.com/package/node-opcua
Got published 2 days ago

Graos, 2022-02-08:
All the nodeset are placed in 
[image]

your error window says a different path /tmp/TEST/node_modules

than your installation location example /tmp/test-opcua

fedefrancescon, 2022-02-08:
Yes, I copy-pasted one of my previous test that was using a different path that the latest that i reported (`/tmp/test-opcua`)

Anyway here is the full output with the same paths as in the request.

```shell
federico@archer /tmp/TEST                                                                                                                                                                                [16:23:42]
> $ mkdir /tmp/test-opcua && cd /tmp/test-opcua

federico@archer /tmp/test-opcua                                                                                                                                                                          [16:27:58]
> $ npm install node-opcua

added 254 packages, and audited 255 packages in 18s

28 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities

federico@archer /tmp/test-opcua                                                                                                                                                                          [16:28:24]
> $ echo "require('node-opcua')" | node
/tmp/test-opcua/node_modules/node-opcua-nodesets/dist/index.js:14
            throw new Error("Please make sure that nodeset can be found in " + path.join(dirname, "../nodesets"));
            ^

Error: Please make sure that nodeset can be found in /tmp/test-opcua/node_modules/node-opcua-nodesets/nodesets
    at constructNodesetFilename (/tmp/test-opcua/node_modules/node-opcua-nodesets/dist/index.js:14:19)
    at Object.<anonymous> (/tmp/test-opcua/node_modules/node-opcua-nodesets/dist/index.js:50:33)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1149:10)
    at Module.load (node:internal/modules/cjs/loader:975:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:999:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/tmp/test-opcua/node_modules/node-opcua/dist/index.js:106:29)
    at Module._compile (node:internal/modules/cjs/loader:1097:14)

Node.js v17.3.0
```

fabioconedera, 2022-02-22:
I have the same problem. Any solutions?

Graos, 2022-02-23:
I would suggest to try with v2.63.0. It works fine fetching the required nodeset from the node_modules folder.

fedefrancescon, 2022-02-23:
Actually `v2.63.2` from NPM registry works fine.

There is a little dis-alignment between github releases and NPM registry versions. Is this supposed to be?

Anyway, could be considered something like removing the broken version from the registry? Now we know the problem and it's good. Just to prevent other peoples from getting in troubles with this.

Graos, 2022-02-23:
I agree. When i try to install node-opcua from NPM it shows v.264.0 as the latest. But in the node-opcua Github under the releases i see only v2.63.2 as the latest release. I am not an expert on this so i can't really comment.

erossignon, 2022-02-27:
fixed in @2.64.1

closed 2022-02-27 by erossignon