node-opcua/node-opcua #1117: subjectAltName for user certificate

keremsubasi, 2022-02-10:
Hello, 

I'm having I problem while connection the OPC UA Server of my PLC to the OPC UA Client node.

These are the Security & Authentication Settings of the Server:
Security Mode: Sign&Encrypt
Security Policy:  Basic256Sha256
Authentication: X.509 Certificates

The server provides this files: a server certificate (.der), a certificate revocation list (.crl) and a rootca certificate (.der).

I stored the server cert in C:\Users\k.subasi\AppData\Roaming\node-red-opcua-nodejs\Config\PKI\trusted\certs, the rootca cert in *\issuers\certs and the crl in *\issuers\crl.

I used the client cert, which was already in *\own folder and imported it into the server.

Did I do that correct so far?

Than I created a user certificate and key. Now I get this error form my command prompt:

[image]

I tried to do it with a config file with the alt_name URI = urn:Kerem:NodeOPCUA-Client. No I get no error in the command prompt but in nodeRED:

[image]

How do I create a correct user certificate? I tried many ways but it didn't work.

Thanks

mikakaraila, 2022-02-10:
Take a look:
https://security.stackexchange.com/questions/74345/provide-subjectaltname-to-openssl-directly-on-the-command-line

keremsubasi, 2022-02-11:
With this configuration I get the error BadIdentityTokenInvalid from UaExpert.

The strange thing is that I got no error with the other config and in the OPC UA online reference it says that only the things I used are required: https://reference.opcfoundation.org/v104/Core/docs/Part6/6.2.2/. 

Now I get this info as well:
[image]

Can you tell me how you created a user certificate an how the config file exactly looked like? 
Thanks

erossignon, 2023-05-05:
check https://github.com/node-opcua/node-opcua-pki; 
Although you can create your own certificate using bare openssl, node-opcua-pki is a recommanded tool to generate OPC UA compatible cerficates.

closed 2023-05-05 by erossignon