node-opcua/node-opcua #1119: Question: rejected valuechange by client gets into ServerHistory

AndreasHeine, 2022-02-18:
#### If a client writes a variable value (like in "BindVariableOptionsVariation1") and i return a "Bad" status code -> the rejected value somehow gets into my in memory history (screenshot!)

i started to dig into it but i haven't manage to find it...

followed the code from "address_space_historical_data_node.ts" to "ua_variable_impl.ts"

assuming either:
the "value_changed" event got emitted !? (https://github.com/node-opcua/node-opcua/blob/9cfb90077559b8d7f817d5fe21156e5ee7c06789/packages/node-opcua-address-space/src/historical_access/address_space_historical_data_node.ts#L799)
or
does it only work with a timestamped_get!? (like in "BindVariableOptionsVariation2")

[image]

```ts
let setpoint = 50
const myHistoricalSetpointVar = namespace.addVariable({
    browseName: 'MyHistoricalSetpointVar',
    componentOf: showcaseHA,
    dataType: DataType.Double,
    userAccessLevel: "CurrentRead | CurrentWrite",
    value: {
        get: function(): Variant {
            return new Variant({
                value: setpoint,
                dataType: DataType.Double
            })
        },
        set: function(this: UAVariable, value: Variant): StatusCode {
            // Check in the backend, if value is in range
            if (value.value <= 100 && value.value > 0){
                setpoint = value.value
                return StatusCodes.Good
            } else {
                // Setpoint not valid!
                return StatusCodes.BadOutOfRange
            }
        }
    }

    addressSpace?.installHistoricalDataNode(myHistoricalSetpointVar, {
        maxOnlineValues: 500,
    })
})
```

a short hint would be awesome ;)

erossignon, 2023-05-05:
I have modified the behavior of the setter so that the returned StatusCodes.BadOutOfRange gets assigned to the underlying recorded DataValue. The value will still be recorded to the history but will be flagged with statusCode=BadOutOfRange.  UAExpert will ignore it to display the diagram.

I have also added a flag to the `addAnalogDataItem({instrumentRange: {low,high}})` to control the behavior of filtering of outof band value during a write operation leads.

This will be available in node-opcua@2.103.0 onward.

AndreasHeine, 2023-05-12:
@erossignon i will test it next week manually! currently the build of the containers does not work...

closed 2023-09-28 by AndreasHeine