node-opcua/node-opcua #1126: Set sessionTimeOut but not take effect

pro-xiaoy, 2022-03-23:
Im create a seeion connect 
by setting options: 
```js
const options = {
  applicationName: "MyClient",
  connectionStrategy: connectionStrategy,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
  endpointMustExist: false,

  applicationName: "Test",
  clientName: "Test",
  requestedSessionTimeout: 60 * 60 * 1000, 
  endpointMustExist: false,
  keepSessionAlive: true,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
};

OPCUAClient.create(options);
```
but always error logs tips me
```bash
03:33:12.164Z :client_subscription_impl      :81    [NODE-OPCUA-W09] The subscription parameters are not compatible with the session timeout !
 ...                                                                  session timeout    = 30000  milliseconds
 ...                                                                  maxKeepAliveCount  = 100
 ...                                                                  publishingInterval = 1000 milliseconds"
 ...                                                
 ...                                                                  It is important that the session timeout    ( 30000 ms) is largely greater than :
 ...                                                                      (maxKeepAliveCount*publishingInterval  =  100000 ms),
 ...                                                                  otherwise you may experience unexpected disconnection from the server if your monitored items are not
 ...                                                                  changing frequently.
03:33:12.166Z :client_subscription_impl      :101   [NODE-OPCUA-W12]  An ideal value for maxKeepAliveCount could be 23.
 ...                                                                  An ideal value for publishingInterval could be 1000 ms.
 ...                                                                  This will make  your subscription emit a keep alive signal every 23000 ms
 ...                                                                  if no monitored items are generating notifications.
 ...                                                                  for instance:
 ...                                                                    const  client = OPCUAClient.create({
 ...                                                                        requestedSessionTimeout: 30* 60* 1000, // 30 minutes
 ...                                                                    });
```
The complete code is as follows
```js

const connectionStrategy = {
  initialDelay: 1000,
  maxRetry: 1
}
const options = {
  applicationName: "MyClient",
  connectionStrategy: connectionStrategy,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
  endpointMustExist: false,

  applicationName: "Test",
  clientName: "Test",
  requestedSessionTimeout: 60 * 60 * 1000, 
  endpointMustExist: false,
  keepSessionAlive: true,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
};

const subOptionParams = {
  requestedPublishingInterval: 1000,
  // 临界情况是23
  requestedMaxKeepAliveCount: 100,

  requestedLifetimeCount: 500,
  maxNotificationsPerPublish: 0,
  publishingEnabled: true,
  priority: 0
}

const monitorParams = {
  samplingInterval: 100,
  discardOldest: true,
  queueSize: 10
}

const opcClient = OPCUAClient.create(options);
const session = await opcClient.createSession();
const subscription = ClientSubscription.create(session, subOptionParams);
```

How to settings **session timeout** Max time

pro-xiaoy, 2022-03-23:
Hello, I have completed the above problems by myself and forced change. Im try settings session Timeout
```js
  session.timeout = 60 * 60 * 60 * 1000;
```
And then hava new problem when I try subscription.monitorItems more than 1000+ nodeIds
my program reports such logs
```bash
(node:1) UnhandledPromiseRejectionWarning: Error:  serviceResult = BadTooManyOperations (0x80100000)
    at process_request_callback (/app/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:53:15)
    at ClientSecureChannelLayer._on_message_received (/app/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:506:9)
    at MessageBuilder.<anonymous> (/app/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:132:18)
    at MessageBuilder.emit (events.js:400:28)
    at MessageBuilder._decodeMessageBody (/app/node_modules/node-opcua-secure-channel/dist/source/message_builder.js:210:26)
    at MessageBuilder._feed_messageChunk (/app/node_modules/node-opcua-transport/dist/source/message_builder_base.js:192:18)
    at PacketAssembler.<anonymous> (/app/node_modules/node-opcua-transport/dist/source/message_builder_base.js:53:68)
    at PacketAssembler.emit (events.js:400:28)
    at PacketAssembler.feed (/app/node_modules/node-opcua-packet-assembler/dist/packet_assembler.js:51:18)
    at MessageBuilder.feed (/app/node_modules/node-opcua-transport/dist/source/message_builder_base.js:86:35)
(Use `node --trace-warnings ...` to show where the warning was created)
```
How to reslove it

BR
Thx

erossignon, 2022-04-14:
BadTooManyOperations  => One of your client command to the server (read/write/browse etc ... ) has too many items and need to be split with several transactions. 
You have to read the server limits and make sure not to exceed those limits in your transactions.

[image]
[image]

Sterfive provides a commercial component called `@sterfive/node-opcua-optimized-client` that does this for you behind the scene so you don't have to worry about transactions size in your client application. please contact us for further info.

erossignon, 2022-06-11:
I would not recommand that you change manually the session time out on the client side.

Session timeout is negociated between the client and the server;

in your case your client requests a 1h timeout

     requestedSessionTimeOut:  60*60*1000

but the server imposes 
  
     sessionTimeout:  30000 /*  30 seconds */

So the only option that you have is to adjust the subscription parameters

I would recommend:

```
  requestedPublishingInterval: 1000,  // 1 second
  requestedMaxKeepAliveCount: 10,   // will be every 10 * 1000 

  requestedLifetimeCount: 500,
  maxNotificationsPerPublish: 0,
  publishingEnabled: true,
```

so that 
      requestedPublishingInterval: 1000 * requestedMaxKeepAliveCount: 10  = 10000

which is now less than  ` sessionTimeout:  30000 /*  30 seconds */`

closed 2022-06-11 by erossignon