node-opcua/node-opcua #1127: (Caught) Exception while processing Certificate Revocation List (CRL), causes CRL to be ignored resulting in 'BadCertificateRevocationUnknown' error.

vembacher, 2022-03-24:
**Current behavior**

My certificate revocation list (CRL) is recognized, processing it causes a caught exception. This results in a `BadCertificateRevocationUnknown` error when connecting to a server using a certificate signed by a CA.

**Describe the bug**

CRL is failed to be processed and added to certificate manager CRL store.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

Move CRL into crl folder (same behavior both in `trusted/crl` and `issuer/crl`.
When processing [this CRL](https://gist.github.com/vembacher/97609fbd8001c9b568e8666a4ae2a38f) the exception occurs at:

https://github.com/node-opcua/node-opcua-crypto/blob/898d90923863c1fbc21b4b76c9290c3044d16cfc/source/explore_certificate_revocation_list.ts#L89

https://github.com/node-opcua/node-opcua-crypto/blob/898d90923863c1fbc21b4b76c9290c3044d16cfc/source/explore_certificate_revocation_list.ts#L56

https://github.com/node-opcua/node-opcua-crypto/blob/898d90923863c1fbc21b4b76c9290c3044d16cfc/source/asn1.ts#L236

**Expected behavior**

CRL is processed and added to the CRL store. 

**Code**

The code that is used can be found here, it does not use the certificate manager at all, so I assume it is not an issue caused on that side.

https://github.com/frangoteam/FUXA/blob/fcf73b2153cd12442c55e8a629500bc2a8f3176f/server/runtime/devices/opcua/index.js

**Context**

- [x] my request is related to node-opcua acting as an OPCUA CLIENT
- [x] I have installed node-opcua as a package ( using npm install )
- Device: n/a
- OS version: Linux
    - [x] Linux : version : same behavior on Manjaro and Docker container with tag `node:14`

- Description of the other OPCUA system I am trying to connect to:

    - Name: 
      - self maintained OpenPLC fork
      - using open62541
    - Version: 
      - latest commit of OpenPLC fork
      - open62541 v1.2.4
    - Manufacturer/Software vendor: https://github.com/open62541/open62541
    - link to the manufacturer or product site: https://github.com/vembacher/OpenPLC_v3
    - **Note:** no connection problems with multiple other OPC UA stacks.

- **node-opcua version:**
v2.65.1
- **Node:**
  `node --version` = v14.19.1 and v12.22.10

**Note:** If this is the wrong repo to create this issue, feel free to close and I will re-open in the other repos. I just saw that most issues are reported here and the issue occurs when using the higher level interface.

erossignon, 2022-03-26:
It looks like the CRL is using a obsolete format V1, which is not supported by node-opcua-crypto;
We only support CRL version 2 or above .... 

The workaround is to regenerate your CRL with  a more recent format. 

```
Certificate Revocation List (CRL):
        Version 1 (0x0)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: CN = PlcTestbedCA, O = PlcTestbed
        Last Update: Mar 14 11:04:33 2022 GMT
        Next Update: Mar 14 11:04:33 2023 GMT
No Revoked Certificates.

-------------- 
expected 

```
Certificate Revocation List (CRL):
        Version 2 (0x1)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = US, ST = Arizona, O = OPC Foundation, CN = ctt_ca1I, DC = xxxxx
        Last Update: Nov 15 09:04:30 2019 GMT
        Next Update: Nov 13 09:04:30 2024 GMT
        CRL extensions:
            X509v3 CRL Number:
                3
Revoked Certificates:
    Serial Number: 03
        Revocation Date: Nov 15 09:04:16 2019 GMT
    Serial Number: 04
        Revocation Date: Nov 15 09:04:19 2019 GMT
    Serial Number: 07
        Revocation Date: Nov 15 09:04:27 2019 GMT
    Serial Number: 08
        Revocation Date: Nov 15 09:04:30 2019 GMT
````

erossignon, 2022-03-28:
fixed in node-opcua@2.66.1 and node-opcua-crypto@1.10.0

closed 2022-03-28 by erossignon