node-opcua/node-opcua #1132: Loading nodeset XML file with StructureType will cause exception

mikakaraila, 2022-04-06:
**Current behavior**

**Describe the bug**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. ....Start server with testsave.xml
2. ....
3. ....

**Expected behavior**

**Screenshots**
[image]

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - (X ) node-red
    - ( ) other : please specify

- Device: _Laptop_____
- OS version: ______

    - (X ) Windows : version : _10_____
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 2.66.3
  
  

- **Node:**
  `node --version` = 12.16.2

mikakaraila, 2022-04-06:
I have emailed corresponding testsave.xml by email for the investigation.

mikakaraila, 2022-04-06:
This way it will work:
[image]

[image]

There are commands: getNamespaceIndex and getNamespaces.

mikakaraila, 2022-04-07:
Latest Prosys UA Simulation server support export nodeset:
https://downloads.prosysopc.com/opcua/simulationserver/Prosys_OPC_UA_Simulation_Server_5_Release_Notes.html

New: Exporting namespace as a NodeSet file (available in Professional Edition only). Right-clicking an instance namespace (namespace with no types) and clicking Export Namespace…​ will save the namespace in a file using the NodeSet format defined in the OPC UA specification .

erossignon, 2022-04-19:
fixed in node-opcua@2.67.0

closed 2022-04-23 by erossignon