node-opcua/node-opcua #1133: Sampling interval equal to zero do not publish information

gabrielfpvh, 2022-04-07:
**Current behavior**

Define an OPCUA Server with a variable changing every 30 seconds.
Connect an OPCUA Client and subscribe to the variable configuring the **sampling interval** of the monitoring item to **zero**.
Using node-opcua version 2.66.3

**Describe the bug**

The client do not receive data updates and addording to the specification: https://reference.opcfoundation.org/v105/Core/docs/Part4/7.21/
"The value 0 indicates that the Server should use the fastest practical rate."

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Define an OPCUA Server with a variable changing every 30 seconds.
2. Connect an OPCUA Client and subscribe to the variable configuring the **sampling interval** of the monitoring item to **zero**.

**Expected behavior**

"The value 0 indicates that the Server should use the fastest practical rate."

**Screenshots**

**Gist**

 
gist: 

Sample server of the documentation: https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/sample_server.js

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (*) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (*) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - (*) other : UA Expert

- Device: VM
- OS version: alpine

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - (*) Other : specify : docker container node:12.20.1-alpine

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =12.20.1

AndreasHeine, 2022-04-07:
like i wrote on gitter:

please check with wireshark! i tested it against -> opc.tcp://opcua3.umati.app:4840 (node-opcua: 2.66.3)
[image]

gabrielfpvh, 2022-04-11:
Hi @AndreasHeine, this issue is a different one.
@erossignon is testing the issue in the sample server: https://github.com/node-opcua/node-opcua/commit/d453e7401de1c3a448be85f5fdcb9a40599cfee6
But is something related to the server implementation itself.

In this issue, apparently the server is not sampling a variable value when the configured sampling interval is set to zero. Thus, the client is not receiving updates since changes are not being detected.
But, according to the OPC UA specification indicates "The value 0 indicates that the Server should use the fastest practical rate."

This issue is easily avoided by setting a "minimumSamplingInterval" greater than zero. But, in my point of view, it is missing a point of the specification.

Please, do not hesitate in asking for more evidence or information. I'm glad to provide it.

erossignon, 2022-04-19:
Setting minimum sampling interval to > 0 is the right thing to do in your case, minimumSamplingInterval=0 should be reserved for trigger variables. Trigger variables are not compatible with getter/setter.

Here is the design intend:

Setting minimumSamplingInterval=0 in the UAVaraible indicates that whenever the value will change the monitored item will be notified. In this case, the variable acts as a trigger and is not suitable for sampling. Using a getter here is not appropriate in this case. setValueFromSource should be used instead.

A continuously changing value, however, can use a getter, the most up-to-date value will be fetched when the sampling interval will occur. But this case requires a specific minimum sampling interval  that is not zero.

This is why I have adjusted the sample.

closed 2022-04-19 by erossignon

gabrielfpvh, 2022-05-03:
Thanks for your answer @erossignon.
I understand the server implementation must define the minimum samplingInterval according to each deployment case. In other words, according to each server capabilities.

There is something I'm still not getting:
- After your explanation, I don't understand the application of a samplingInterval equal to zero.
- Does the OPCUA specification define any variable type such as trigger? Isn't it relataded to the application level?