node-opcua/node-opcua #1134: Side effect due to #1096 improving extension object loaded in nodeset2.xml

Graos, 2022-04-08:
Node-OPCUA version: 2.66.1
Loaded a nodeset file with an extension object instance variable of the type (StructWithAllDataTypeElements) which contains all the data type elements with its value initialized.
Following values are initialized to the respective data type elements in the extension object
boolElement                   /* Boolean             */: true
 byteElement                   /* Byte                */: 255
 stringElement                 /* String              */: OPCUASERVER
 floatElement                  /* Float               */: 99.99
 uint16Element                 /* UInt16              */: -1
 uint32Element                 /* UInt32              */: 65536               0x
10000
 int16Element                  /* Int16               */: 32767
 int32Element                  /* Int32               */: 2147483647
 stringElement2                /* String              */: OPCUASERVER
 stringElement3                /* String              */: OPCUASERVER

when reading the extenion object variable instance using the read attribute service from OPC UA client. It successfully reads but when the same variable is subscribed the OPC UA server goes into the error mode with the below prints.

Why the node-opcua server thinks 65535 value for uint16 as out of range. ???

*************************************Error prints from Node-OPCUA server*****************************************************
Error encoding variant RangeError [ERR_OUT_OF_RANGE]: The value of "value" is ou
t of range. It must be >= 0 and <= 65535. Received -1
    at new NodeError (node:internal/errors:371:5)
    at checkInt (node:internal/buffer:72:11)
    at writeU_Int16LE (node:internal/buffer:722:3)
    at Buffer.writeUInt16LE (node:internal/buffer:730:10)
    at BinaryStream.writeUInt16 (\node_modules\node-o
pcua-binary-stream\dist\binaryStream.js:110:21)
    at encodeUInt16 (\node_modules\node-opcua-basic-t
ypes\dist\integers.js:23:12)
    at encodeArrayOrElement (\node_modules\node-opcua
-schemas\dist\source\dynamic_extension_object.js:63:13)
    at _internal_encodeFields (\node_modules\node-opc
ua-schemas\dist\source\dynamic_extension_object.js:186:17)
    at encodeFields (\node_modules\node-opcua-schemas
\dist\source\dynamic_extension_object.js:222:5)
    at EXTENSION.encode (\node_modules\node-opcua-sch
emas\dist\source\dynamic_extension_object.js:314:9) {
  code: 'ERR_OUT_OF_RANGE'
}
Variant(Scalar<ExtensionObject>, value: { /*StructWithAllDataTypeElements*/
 boolElement                   /* Boolean             */: true
 byteElement                   /* Byte                */: 255
 stringElement                 /* String              */: OPCUASERVER
 floatElement                  /* Float               */: 99.99
 uint16Element                 /* UInt16              */: -1
 uint32Element                 /* UInt32              */: 65536               0x
10000
 int16Element                  /* Int16               */: 32767
 int32Element                  /* Int32               */: 2147483647
 stringElement2                /* String              */: OPCUASERVER
 stringElement3                /* String              */: OPCUASERVER
};)
09:27:28.065Z :opcua_server                  :1443  Internal error in issuing re
sponse
 ...                                                please contact support@sterf
ive.com { /*ReadRequest*/
 ...                                                 requestHeader
   /* RequestHeader       */: {
 ...                                                   authenticationToken
   /* NodeId              */: ns=0;b=Qz9WISq8vbTBm8iXx9l9hw==
 ...                                                   timestamp
   /* DateTime            */: 2022-04-08T09:27:28.036Z
 ...                                                   requestHandle
   /* UInt32              */: 1011140               0xf6dc4
 ...                                                   returnDiagnostics
   /* UInt32              */: 0               0x0
 ...                                                   auditEntryId
   /* UAString            */: null
 ...                                                   timeoutHint
   /* UInt32              */: 10000               0x2710
 ...                                                   additionalHeader
   /* ExtensionObject     */: null
 ...                                                 }
 ...                                                 maxAge
   /* Double              */: 0
 ...                                                 timestampsToReturn
   /* TimestampsToRetur   */: 2 /*(Both)*/
 ...                                                 nodesToRead
   /* ReadValueId      [] */: [
 ...                                                   { /*0*/
 ...                                                     nodeId
   /* NodeId              */: ns=2;i=6192
 ...                                                     attributeId
   /* UInt32              */: AttributeIds.NodeId/* 1 */
 ...                                                     indexRange
   /* NumericRange        */: NumericRange:<Empty>
 ...                                                     dataEncoding
   /* QualifiedName       */: <null>
 ...                                                   },
 ...                                                   { /*1*/
 ...                                                     nodeId
   /* NodeId              */: ns=2;i=6192
 ...                                                     attributeId
   /* UInt32              */: AttributeIds.NodeClass/* 2 */
 ...                                                     indexRange
   /* NumericRange        */: NumericRange:<Empty>
 ...                                                     dataEncoding
   /* QualifiedName       */: <null>
 ...                                                   },
 ...                                                 .... TRUNCATED ..... (NODEO
PCUA_DEBUG_MAXLINE_PER_MESSAGE=25
node:internal/process/promises:246
          triggerUncaughtException(err, true /* fromPromise */);
          ^

RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be
>= 0 and <= 65535. Received -1
    at new NodeError (node:internal/errors:371:5)
    at checkInt (node:internal/buffer:72:11)
    at writeU_Int16LE (node:internal/buffer:722:3)
    at Buffer.writeUInt16LE (node:internal/buffer:730:10)
    at BinaryStream.writeUInt16 (\node_modules\node-o
pcua-binary-stream\dist\binaryStream.js:110:21)
    at encodeUInt16 (\node_modules\node-opcua-basic-t
ypes\dist\integers.js:23:12)
    at encodeArrayOrElement (\node_modules\node-opcua
-schemas\dist\source\dynamic_extension_object.js:63:13)
    at _internal_encodeFields (\node_modules\node-opc
ua-schemas\dist\source\dynamic_extension_object.js:186:17)
    at encodeFields (\node_modules\node-opcua-schemas
\dist\source\dynamic_extension_object.js:222:5)
    at EXTENSION.encode (\node_modules\node-opcua-sch
emas\dist\source\dynamic_extension_object.js:314:9) {
  code: 'ERR_OUT_OF_RANGE'

********************************************************************************************************************************* 
This issue is not seen in 2.65.0 because in that version the Node-OPCUA server does not load the extension object variable elements value from the load set file. This issue is introduced after the change #1096 improving extension object loaded in nodeset2.xml in 2.66.1

erossignon, 2022-04-10:
uint16Element / UInt16 /: -1  is a invalid value for UInt16

erossignon, 2022-04-10:
I've added a fix so that values are clamped inside their valid range. In your case, -1 will be adjusted to be 0

Graos, 2022-04-11:
Thanks. You are right i did not notice the nodeset file contains -1 as value for uint16 type. I will close the case.

closed 2022-04-11 by Graos