node-opcua/node-opcua #1137: OPCUA client crashes if Temp directory is missing

Yaro96, 2022-04-13:
**Current behavior**

Running a node opcua-client through pm2 on a Windows server, sometimes I encounter this unhandled error, which blocks completely the execution.

**Screenshots**

[image]

Inside "AppData\Local\Temp" the expected folder (in this case "2") is missing (there is "3" instead).  The issue can be resolved if I create the expected folder ("2") in "AppData\Local\Temp" after that, the program creates an "empty_config.cnf" and runs smoothly

[image]

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Virtual Machine
- OS version: Windows Server 2019

- Description of the other OPCUA system I am trying to connect to:

- **node-opcua version:** : 2.66.3
  
  

- **Node:**
  `node --version` = v16.14.0

erossignon, 2022-04-13:
This seems to be a PM2 issue similar to https://github.com/Unitech/pm2/issues/3365

An easy workaround is to make sure that the %TEMP% folder exists.
Alternatively, you can also set the TEMP to a specific folder that you control

PM2 provides a way to inject environment variables into the process:

https://pm2.keymetrics.io/docs/usage/environment/

erossignon, 2022-05-15:
@[Yaro96](https://github.com/Yaro96) can we close this issue now ?

Yaro96, 2022-05-23:
yes, thanks for the support

closed 2022-05-23 by Yaro96