node-opcua/node-opcua #1138: private key file issues

SteveJay7, 2022-04-20:
opcuaClient.create gives me the following error:
"cannot locate private key file ...\node_modules\node-opcua-client\certificates\PKI\own\private\private_key.pem"

I don't need encrypted communication so I used the following parameters:
securityMode: MessageSecurityMode.None,
securityPolicy: SecurityPolicy.None,

Shouldn't that result in ignoring certificates? So not sure if this is a bug or if there are other settings I need to bypass encryption?

I also tried to specify a relative path to the included certificate file by setting this:
certificateFile: "./certificates/client_selfsigned_cert_2048.pem",

... but still I get the error that it can't locate the key file. Wondering what I did wrong here.

Thanks for your help!

**Context**

my request is related to node-opcua acting as an OPCUA CLIENT
I have installed node-opcua as a package ( using npm install )

- **Windows 11**
- **node-opcua version:** :  2.21.0
- **Node:**
  `node --version` = 12.16.2

erossignon, 2022-04-23:
Please provide a gist that reproduces the issue, to facilitate the investigation 
Something like:

```` javascript
const { OPCUAClient, AttributeIds } = require("node-opcua");
(async () => {
  const client = OPCUAClient.create();
  const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";

  await client.connect(endpointUrl);
  const session = await client.createSession();

  const dataValue = await session.read({
    nodeId: "i=2258", // currecnt Tim
    attributeId: AttributeIds.Value,
  });
  console.log("Server time is ", dataValue.value.toString());
  await session.close();
  await client.disconnect();
})();

````

SteveJay7, 2022-04-28:
Hi Etienne,

the issues seems to be related to yarn. I recognized that if I add any module with yarn, like "yarn add electron-devtools-installer", the PKI folder in the node-opcua-client/certificates/ folder gets deleted, which then causes the error mentioned above. after re-adding the PKI folder and its contents, everything works again.

Do you have any idea why the PKI folder gets deleted with every yarn activity? I even tried to remove all postinstall scripts from my package.json but that didn't help.

Thanks,
Stefan

erossignon, 2022-04-29:
@SteveJay7 it is very difficult to answer your question without seeing the code that you are using to create your client.
OPCUA Client uses certificates that are stored outside the node_modules folder by default. 
Please contact us for professional support i

closed 2022-04-29 by erossignon