node-opcua/node-opcua #1143: "Object has more than one parent" when instantiating type from companion sepcification

mb-software, 2022-05-05:
**Current behavior**

I have an OPC UA companion specification, where one objecttype has a childobject and a subfolder. the subfolder again has a HasAddin reference to the childobject. I have also prepared a minimal nodeset2.xml file with the same constellation to reproduce the issue (see attachement).

When I instantiate this objecttype by code, I get a "object 'childobject' has more than one parent" error.
When the object is instantiated in the nodeset2xml it is however working fine.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. create a server and load the nodeset.xml from the attachement
2. instantiate an instance of type 'testtype'

```
const nodesettest = 'modeltest.xml';
(async () => {
  // Let's create an instance of OPCUAServer
  const server = new OPCUAServer({
    port: 4334, // the port of the listening socket of the server
    resourcePath: '/UA/MyLittleServer', // this path will be added to the endpoint resource name
    buildInfo: {
      productName: 'MySampleServer1',
      buildNumber: '7658',
      buildDate: new Date(2014, 5, 2),
    },
    nodeset_filename: [
      nodesets.standard,
      nodesets.di,
      nodesettest,
    ],
  });
  await server.initialize();
  console.log('initialized');

  const { addressSpace } = server.engine;
  const namespace = addressSpace.getOwnNamespace();
  const nstest = addressSpace.getNamespaceIndex('http://yourorganisation.org/modeltest/');

  const testType = addressSpace.findObjectType('testtype', nstest);

  // Extruder instance with optionals test
  const test = testType.instantiate({
    browseName: 'testobj1',
    //nodeId: 's=testobj1',
    organizedBy: addressSpace.rootFolder.objects,
  });

  server.start(() => {
    console.log('Server is now listening ... ( press CTRL+C to stop)');
    console.log('port ', server.endpoints[0].port);
    const { endpointUrl } = server.endpoints[0].endpointDescriptions()[0];
    console.log(' the primary server endpoint url is ', endpointUrl);
  });
})();
```

**Expected behavior**
should give no error and create an instance of the object like the instance in the nodeset (testobj1)

**Screenshots**

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - (x ) Windows : version : ____2.68.0__
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

erossignon, 2022-06-11:
NodeOPCUA displays a warning (not an error) when a node is referenced (Aggregated) by more than one parent  node
[image].
This warning is here to help  us detect inconsistencies in models.
You can ignore the warning or get the  model file adjusted.

Please contact@sterfive.com to investigate deeper your specific case.

mb-software, 2022-08-02:
Thanks for your answer.
Unfortunately the problem is not only a warning that I could ignore, but a wrong behaviour when instantiating the type via code.

Please try to instantiate the testtype like in my code above. Then change the value of myobj. You will notice that the value of subfolder.myobj won't change, meaning that you get two independent instances of myobj instead of one instance with two references.

Changing the model file unfortunately is also not an option since it's an official companion specification published by the OPC Foundation.

xenonforlife, 2022-08-09:
This is indeed a modelling decision, which a particular standardisation group takes. It is getting more and more popular to re-reference elements of an OPC UA information model. A classic example is the MachineryComponentsType object (DefaultInstanceBrowseName: Components) in the OPC UA for Machinery Companion Specification. This object is supposed to contain other objects connected via the HasComponent reference. Those other objects may already be connected to some other (parent) objects in the information model with some HierarchicalReference (thus making it have more than one parent) i.e. same instance referenced multiple times.

erossignon, 2025-04-07:
Creating UANode that have 2 different  parents (HasChild) is an anti pattern that breaks the concept of object oriented modeling and cause trouble with automatic wrapper code generator. 
The recommanded approach is to anchor the child node under one single parent and use a different type of Reference to link it with a second Node (suchas Organizes or any other suitable semantics).

closed 2025-04-07 by erossignon