node-opcua/node-opcua #1144: Cannot find StructureType constructor for "DT -"

teropes, 2022-05-11:
**Current behavior**
node-opcua client thows exception when it is browsing data types of Siemens S7-1500 PLC OPC UA server. It outputs alot of schema and data type related output and finally throws exception with output:
`Cannot find StructureType constructor for "DT - it may be abstract, or it could be a basic type`

After enabling some debug logging to `datatype_factory.js` before the exception, it seems that our client is processing datatype with name of `DT: LM DRIVE` when it throws exception.

For reference, here are couple previous datatype names and how their `typeName `is parsed correctly for the `getStructureTypeConstructor` function and then this `DT: LM DRIVE` which throws exception:
```
processing  "dtMeasure"."lim"
getStructureTypeConstructor typeName: "dtMeasure"."lim"
processing  "dtMeasure"."value"
getStructureTypeConstructor typeName: "dtMeasure"."value"
processing  dtMeasure
getStructureTypeConstructor typeName: dtMeasure
processing  DT: LM DRIVE
getStructureTypeConstructor typeName: "DT
Cannot find StructureType constructor for "DT - it may be abstract, or it could be a basic type
```

Stack trace:
[image]

UA Expert client does not give any errors and for example subscription to variables using `DT: LM DRIVE` datatype work as expected.

Datatype looks like this with UA Expert:
[image]

**Describe the bug**
While I agree that prefixing datatype name with `DT: ` might be bad practice and it would in our case lead to nodeId like `ns=3;s=DT_"DT: LM DRIVE"`, I don't think the library should throw exception / stop working on it IF it is valid according to the OCP UA specification. In case that kind of naming is clearly against the OPC UA specification, then it is expected behaviour from the node-opcua library. 

**Step by step instruction to reproduce**
Define OPC UA server with datatype like `ns=3;s=DT_"DT: Test"` and run node-opcua client against it.

**Expected behavior**
If it is not against OPC UA specification, library should not throw exception / stop processing further. 

**Context**
- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

    - ( ) Windows : version : ______
    - (x) Linux : version : CentOS 8 x86_64
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:
    - Name: Siemens S7-1500 PLC OPC UA Server

- **node-opcua version:** :
  v2.67.0

- **Node:**
  `node --version` = 16.14.2

teropes, 2022-05-11:
Actually I was not able to re-produce this issue with another Siemens PLC by adding `DT: TEST` datatype there. It does not seem to cause any errors.

erossignon, 2022-06-11:
So let's close, 
Thank you for your support.

closed 2022-06-11 by erossignon