node-opcua/node-opcua #1147: Can't browse subfolder

doteich, 2022-05-31:
I'm trying to browse the classic Siemens TIA OPC structure
[image]
However while browsing from Root further down I get stuck at the subfolder "DataBlocksGlobal" (ns=3;s=DataBlocksGlobal) as this only returns me a single Variable, while UA Expert shows all the available sub-objects. 

This: ` 
session.browse('ns=3;s=DataBlocksGlobal', (err, browseResult)=> {
if (!err) {                
       for(const reference of browseResult.references) {
            console.log( "   -> ", reference.toString());
        }
            }
            console.log(err)
        })
`
Only Returns: `   ->  { /*ReferenceDescription*/
 referenceTypeId               /* NodeId              */: HasProperty (ns=0;i=46)
 isForward                     /* UABoolean           */: true
 nodeId                        /* ExpandedNodeId      */: ns=3;i=5202
 browseName                    /* QualifiedName       */: Icon
 displayName                   /* LocalizedText       */: locale=null text=Icon
 nodeClass                     /* NodeClass           */: 2 /*(Variable)*/
 typeDefinition                /* ExpandedNodeId      */: ns=0;i=68
};
null`
If a go one step further and browse nodes in the subfolder, it works again. 
Is this a bug or is there any way to browse such kind of subfolders?

erossignon, 2022-06-04:
The simplified form only follows HasChild References.

you need to use the complete form of the session. browse method:
```
session.browse({
     nodeId: "ns=3;s=DataBlocksGlobal",
     referenceTypeId: "HierarchicalReferences",
     browseDirection: BrowseDirection.Forward,
     includeSubtypes: true,
     nodeClassMask: 0,
     resultMask: 63
}, callback)
````

doteich, 2022-06-07:
Thanks for your quick response. I changed the browse params to following: 
` session.browse({
            nodeId: 'ns=3;s=DataBlocksGlobal',
            referenceTypeId: "HierarchicalReferences",
            browseDirection: BrowseDirection.Forward,
            includeSubtypes: true,
            nodeClassMask: 0,
            resultMask: 63

        }, (err, browseResult) => {
            if (!err) {

                for (let reference of browseResult.references) {
                    console.log(reference.toString());
                }
            }
            console.log(err);
        })`

Still i only get following browsing results:
`{ /*ReferenceDescription*/
 referenceTypeId               /* NodeId              */: HasProperty (ns=0;i=46)
 isForward                     /* UABoolean           */: true
 nodeId                        /* ExpandedNodeId      */: ns=3;i=5202
 browseName                    /* QualifiedName       */: Icon
 displayName                   /* LocalizedText       */: locale=null text=Icon
 nodeClass                     /* NodeClass           */: 2 /*(Variable)*/
 typeDefinition                /* ExpandedNodeId      */: ns=0;i=68
};
null`

Is there any other option/param i can set?

doteich, 2022-06-07:
Can be closed. Setting the appropriate NodeClass returned me all the values. 
Sample code, if anybody faces the same issue: 
`session.browse({
            nodeId: "ns=3;s=DataBlocksGlobal",
            referenceTypeId: 33,
            browseDirection: BrowseDirection.Forward,
            includeSubtypes: true,
            nodeClassMask: NodeClass.Object,
            resultMask: 63

        },`

closed 2022-06-07 by doteich