node-opcua/node-opcua #1148: Option automaticallyAcceptUnknownCertificate apply BadCertificateTimeInvalid

tuxnsk, 2022-06-01:
Is it possible to ignore not only the BadCertificateUntrusted error, but also the BadCertificateTimeInvalid error when setting automaticallyAcceptUnknownCertificate. Or is there another method for this?

erossignon, 2022-06-03:
certificate time must be valid, there is no option to bypass this requirement.

tuxnsk, 2022-06-04:
Unfortunately, the largest Russian manufacturer of automation equipment does not understand this and only provides access with an expired certificate.

opc.tcp://opc.owencloud.ru:4843

erossignon, 2022-06-04:
Sorry, there is nothing we can do. security requires a set of validation steps that must be complied with.

closed 2022-06-04 by erossignon

PatersonKopp, 2023-07-11:
But why isn't there an option to disable this check? I also know a lot of manufacturers with expired certificates on their servers. I know it not a good practice but for some use case it absolutly make sense to ignore those invalid certificates.