node-opcua/node-opcua #1153: missing pem file

lightMorning, 2022-06-08:
I met a problem，my project doesn't generate pem file

`An error has occured :  [Error: ENOENT: no such file or directory, open 'C:\Users\DELL\AppData\Roaming\node-opcua-default-nodejs\Config\PKI\rejected\NodeOPCUA-Client@iTQo??-s?[9c81bfc884a79d083ede54cbd114e25300b74406].pem'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'open',
  path: 'C:\\Users\\DELL\\AppData\\Roaming\\node-opcua-default-nodejs\\Config\\PKI\\rejected\\NodeOPCUA-Client@iTQo??-s?[9c81bfc884a79d083ede54cbd114e25300b74406].pem'
}`

node version：14.19.3
npm version：6.14.17
This project automatically installs openssl in node14

I tried upgrading node to 16 and it didn't work. How to solve this problem please

lightMorning, 2022-06-09:
I changed my computer name to English and it works fine

closed 2022-06-09 by lightMorning

tbn-mm, 2024-06-21:
If you applicationName contains slashes, the same error is thrown because the applicationName is interpreted as path that does not exist.