node-opcua/node-opcua #1160: Meaning of "Invalid channel" error from client.createSession()?

AgileInteractive, 2022-06-29:
`const client = OPCUAClient.create({
    endpointMustExist: false,
    securityMode: MessageSecurityMode.SignAndEncrypt,
    securityPolicy: SecurityPolicy.Basic256Sha256
});

await client.connect(endpointUrl);

const session = await client.createSession({
        userName: data.userName,
        password: data.password,

    },
    function(err) {
        console.log(err)
    }
);`

When trying to connect to a server with user/pass and Basic256Sha256, I get this error "Invalid channel". Other OPC-UA softwares connects to the server (not a node-opcua server) without any problems.

erossignon, 2022-07-11:
Make sure not to mix async/await pattern with the old callback pattern

you should not provide a callback function to the API if you intend to use `await`

- also you need to pass the type: UserTokenType.UserName to the parameters

```` javascript
const session = await client.createSession({
        type: UserTokenType.UserName,
        userName: data.userName,
        password: data.password,
    });

````

bramvdbruggen, 2022-07-19:
I'm also getting the `Invalid Channel` error, mine is on a `BrowseRequest` tho.
The problem I'm facing is that it's only happening after I wait a little before running the function again.

```ts
const client = OPCUAClient.create({});
await client.connect(connectionstring);
this.session = await client.createSession({
    type: UserTokenType.UserName,
    userName: username,
    password: password,
});

await this.session.browse({ nodeId: `ns=2;s=FileSystem${path}` });
```

erossignon, 2022-07-19:
The session times out before you have a chance to send the browse command.
You need to setup the client with one of the 2 keepalive technics that we recommend.

bramvdbruggen, 2022-07-19:
That was a fast reaction and a simple fix. Thanks!

simultsop, 2022-08-01:
> @[bramvdbruggen](https://github.com/bramvdbruggen) 2 keepalive technics that we recommend.

Any reference/url for this?

bramvdbruggen, 2022-08-01:
@simultsop I have not found any url for this.

But I fixed it by adding `keepSessionAlive: true` to the `OPCUAClient.create` function.

```ts
let client = OPCUAClient.create({
    keepSessionAlive: true,
});
```

bramvdbruggen, 2022-08-03:
However after some time the next error occurs.
`client_session_keepalive_manag:149    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest`

Is there some time limit on the keepalive or am I missing something else?

cbudai, 2022-08-03:
I have the same message when I disconnect the OpcUa Server to see what will happen if the server goes down in production, but first i have this message:

14:01:02.983Z :client_session_keepalive_manag:149    warning : ClientSessionKeepAliveManager#ping_server  Transaction has timed out ( timeout = 60000 ms , request = ReadRequest)

And then:

14:01:13.002Z :client_session_keepalive_manag:149    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel after performing transaction on ReadRequest

`
 const client = OPCUAClient.create({
    applicationName: "test",
    connectionStrategy,
    clientCertificateManager,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpointMustExist: false,
    keepSessionAlive: true,
    requestedSessionTimeout: 15000,
  });

  client.on("connection_failed", () => {
    console.log(`Client failed to connect.`);
  });
  client.on("connection_lost", () => {
    console.log(`Client lost the connection.`);
  });
  client.on("start_reconnection", () => {
    console.log(`Client is starting the reconnection process.`);
  });
  client.on("reconnection_attempt_has_failed", (_, message) => {
    console.log(`Client reconnection attempt has failed: ${message}`);
  });
  client.on("after_reconnection", () => {
    console.log(`Client finished the reconnection process.`);
  });
  client.on("backoff", (attemptNumber, delay) => {
    console.log(
      `Client connection retry attempt ${attemptNumber} retrying in ${delay}ms.`
    );
  });
  client.on("close", () => {
    console.log(`Client closed and disconnected`);
  });
  client.on("timed_out_request", (request) => {
    console.log(`Client request timed out: ${request.toString()}`);
  });

  await client.connect(endpointUrl);

  const session = await client!.createSession();

  const subscription = await session.createSubscription2({
    requestedPublishingInterval: 250,
    requestedMaxKeepAliveCount: 50,
    requestedLifetimeCount: 6000,
    maxNotificationsPerPublish: 1000,
    publishingEnabled: true,
    priority: 10,
  });

  subscription.on("keepalive", () => {
    console.log("keepalive")
  });

  subscription.on("keepalive_failure", (state) => {
    console.log(
      `Session encountered a keepalive error: ${
        state !== undefined ? state.toString() : "No state provided."
      }`
    );
  });
  subscription.on("session_closed", (statusCode) => {
    console.log(`Session closed. Status code: ${statusCode.toString()}`);
  });
  subscription.on("session_restored", () => {
    console.log(`Session restored.`);
  });

  subscription.on("terminated", () => {
    console.log("terminated"
  });

  subscription.on("status_changed", () => {
    console.log("status_changed");
  });

  subscription.on("error", () => {
    console.log("error");
  });

  const itemsToMonitor: any = [];
...
`

bramvdbruggen, 2022-08-05:
I found [this other issue](https://github.com/node-opcua/node-opcua/issues/697#issuecomment-552935113) that explained what is happening. In my case this error appears over night when the machines running the opcua server will be shut off, therefore the connection is lost and there is no session to keepalive.

Solving is just as easy as connecting and creating a session when the keepalive fails.
```ts
async connectClient() {
    await this.client.connect(this.constring);
    this.session = await this.client.createSession(this.user);
}
```
And then:
```ts
this.session.on("keepalive_failure", () => {
    this.connectClient();
})
```

The transaction timeout problem that you are having @cbudai is one I don't have a clue about at the moment.

cbudai, 2022-08-08:
Thanks @bramvdbruggen I have tried what you say but it does not work well. In my case it gives me the next error: "client connected". Then I try to disconnect the client, but when disconnecting the client it also closes the session and I not have the keep alive failure for when the OpcUa Server is working again.

I also put the connect of the client, and it stays in backoff. So the backoff skips until the server is back online and everything is fine. When the server is online, the client connects and creates the subscription, but not the session.

```ts
this.session.on("keepalive_failure", () => {
  this.restartConnection()
  logger.warn("keepalive_faillure")
});

  restartConnection = async () => {
    try {
      await this.client.disconnect()
      await this.client.connect(this.endpointUrl)
      this.session = await this.client.createSession()
      await this.createOpcUaSubscription()
      this.opcUaSubscriptionCreateMonitorItems();
    } catch (error) {
      console.log("OpcUa: restartConnection", error)
    }
  }
```

morphace, 2022-12-06:
We experience the same issue - is there a solution for it, or a workaround?

AgileInteractive, 2022-12-06:
> We experience the same issue - is there a solution for it, or a workaround?

My second post worked around it with a callback instead of await

erossignon, 2023-05-01:
@morphace , 

the fix is to use async/await form and not to mix callback with async:async

```
// WRONG CODE : This code is mixing async/await and callback form
const session = await client.createSession({
        userName: data.userName,
        password: data.password,

    },
    function(err) {
        console.log(err)
    }
);
```

correct:
```javascript
try {
  this.session = await client.createSession({
     type: UserTokenType.UserName,
     userName: username,
     password: password,
  });
} catch(err) {
   console.log("Session failed to be created, err.message);
}
```

or 
```javascript

client.createSession({
    type: UserTokenType.UserName,
    userName: username,
    password: password,
}, (err, session) =>{
    if (err) {
         console.log("Session failed to be created, err.message);
    } else {
        // do something with session 
    }
});
```j

closed 2023-05-01 by erossignon