node-opcua/node-opcua #1162: After a "password" change, a reboot of the OPCUA server results in "maximum number of Sessions supported by Server has been reached".

weekaung, 2022-07-08:
**Current behavior**
We have implemented "security" with 
securityPolicies: [
        SecurityPolicy.Basic128Rsa15,
        SecurityPolicy.Basic256
    ],

    securityModes: [
        MessageSecurityMode.Sign,
        MessageSecurityMode.SignAndEncrypt
    ],

    certificateFile: server_certificate_file,
    privateKeyFile: server_certificate_privatekey_file,

Also, our system allows the user to change the default password, which the server will automatically do a restart with the new password.

**Describe the bug**
However, when there are "clients" already connected to the server, the client would try to do a re-connect continuously, resulting in the server using up the "maxAllowedSessionNumber", after which the server is un-usable as clients are unable to login.
More specifically, the client which is connected is a "Prosys OPC UA Browser" which we are using for testing.

**Expected behavior**
Why is the "node-opcua" using up the session when the same client is trying to reconnect? Shouldn't it be using the same session and disable re-connect after a fixed amount of retries?

**Screenshots**
[image]

[image]

- ( ) my request is related to node-opcua acting as an OPCUA SERVER
- ( ) I am using an application that uses node-opcua

- **node-opcua version:** :
  "node-opcua": "^2.70.3"

- **Node:**
  `node --version` = v16.15.0

weekaung, 2022-07-11:
02:20:12.352Z :server_end_point              :625   PREVENTING DDOS ATTACK => maxConnection =1
02:20:12.352Z :server_end_point              :645   Closing channel  241
02:20:38.397Z :server_end_point              :625   PREVENTING DDOS ATTACK => maxConnection =1
02:20:38.398Z :server_end_point              :631   All channels are in used ! let cancel some
OPCUAServerEndPoint#_on_client_connection The maximum number of connection has been reached - Connection is refused

Getting the above errors, when I change password and reboot server. The Prosys OPC UA Browser still trying to re-connect with old password.

weekaung, 2022-07-11:
This seems to be a clear "bug" in the system. On changing the credentials of the currently logged in OPCUA client and doing a OPCUA server reboot, the system will crash ( the client would use up the "maxAllowedSessionNumber" so that other clients cannot connect ).

AndreasHeine, 2022-07-11:
@weekaung hey, for me it sounds like the prosys-client has a to "aggressive" reconnect strategy!

sidenote: the prosys-client cant re-use the session because the of the userchange

weekaung, 2022-07-15:
@AndreasHeine Hi Andreas, thank you for replying to my problem. Yes, it seem so prosys-client has a very "aggressive" reconnect strategy. However, I connected the Prosys-client to another OPC UA server (developed by another vendor) and their server did not run out of "session" as a result of the password change, it was able to handle it and the result is that the client cannot log on, which is a behavior I would expect.

AndreasHeine, 2022-07-15:
can you provide a wireshark trace? so we can see what happens.

weekaung, 2022-07-15:
@AndreasHeine Attached is the Wireshark capture. Thank you so much
[ProSys-OPCUA.pcapng.gz](https://github.com/node-opcua/node-opcua/files/9119042/ProSys-OPCUA.pcapng.gz)
.

AndreasHeine, 2022-07-15:
Its encrypted...

weekaung, 2022-07-15:
It's gzip compressed. This is the zip compressed.
[ProSys-OPCUA.pcapng.zip](https://github.com/node-opcua/node-opcua/files/9119168/ProSys-OPCUA.pcapng.zip)

AndreasHeine, 2022-07-15:
no the communication between client and server ^^ so in the trace you don't see the actual requests ;)

weekaung, 2022-07-15:
Is there anyway I can set it to not use encryption?

AndreasHeine, 2022-07-15:
use SecurityPolicy "None"

weekaung, 2022-07-15:
Just give me a moment, I will try again. I think I know how.

weekaung, 2022-07-15:
@AndreasHeine 
Hi Andreas,
I couldn't capture un-encrypted packets. I have downloaded the sample_server.js (https://github.com/node-opcua/node-opcua)/documentation/sample_server.js) and made added the "userManager" to check user credentials. I am encountering the same problem.
I will attach the code which you will be able to see the problem more clearly. The default credentials are admin / password. After using "Prosys" client to connect using the above credentials, stop the server and change the password and restart ~ this problem will surface.
Thank you.
[MyLittleServer.pcapng.zip](https://github.com/node-opcua/node-opcua/files/9119622/MyLittleServer.pcapng.zip)
[sample-server.zip](https://github.com/node-opcua/node-opcua/files/9119623/sample-server.zip)
[image]

AndreasHeine, 2022-07-15:
cant spot a issue on node-opcua side... if your server changes the user and/or password its correct that the client cant connect anymore (and if he does that too often its looks like a bruteforce attack) i used UAExpert by the way and after i changed the user in the client i was able to connect again!

weekaung, 2022-07-18:
@AndreasHeine 
Hi Andreas,
Thank you once again for taking the time to look into this problem.
I have made a video to show you what exactly the problem is, maybe you will be able to replicate that in your environment.
I am using the same sample code I have uploaded last week.
Thank you,
David.

PS: Because the recording is too large, I have uploaded it to YouTube.
Below is the YouTube link.

https://youtu.be/87j3gn-yd8s

weekaung, 2022-07-19:
@AndreasHeine 
Hi Andreas,

Yes, it could be that the  "Prosys Client" is misbehaving. It could be any rogue client out there, the "server" however should be able to prevent it from using up server resources so much so that other clients cannot connect again. 

As I have mentioned previously, we have tried this same scenario with another vendor's product (Moxa, to be more specific) and their OPCUA server was able to prevent this from happening, so I am sure that there is a way around this problem, but our technical team do not have the resources or ability to look into "NodeOPCUA" source code to fix the problem.
Your help will be very much appreciated or if the core team behind "NodeOPCUA" would take the time to look into this issue. 

This is a very real world problem that I am sure other people will encounter.
Thank you so much.
David.

PS: MOXA OPCUA link below:
https://www.moxa.com/en/products/industrial-edge-connectivity/opc-ua-software/mx-aopc-ua-suite#resources

erossignon, 2022-07-30:
@weekaung,
>  our technical team do not have the resources or ability to look into "NodeOPCUA" 
You  the author at contact@sterfive.com for professional services and we will be able to dig into this for you.

erossignon, 2022-08-09:
This should now be fixed now in @2.73.0, @weekaung can you confirm ?

weekaung, 2022-08-09:
@erossignon 
Hi Etienne,

Made a few tests and yes the bug has been fixed.

Thank you so much,
David.

closed 2022-08-09 by weekaung