node-opcua/node-opcua #1163: Problem when version of Companion Specification includes "RC" (Release Candidate)

DrHaraldWeber, 2022-07-11:
**Current behavior**
Cannot start server because of NodeSet version with characters inside

**Describe the bug**
Using the model
`<Model ModelUri="http://opcfoundation.org/UA/PlasticsRubber/ImmToRobot/" PublicationDate="2022-07-01T00:00:00Z" Version="RC 1.00.04">`

Throws the error: 
`05:55:57.333Z :generateAddressSpaceRaw       :36    generateAddressSpace:  Loading xml file  ./nodesets/opc40079forumati.xml  failed with error  Invalid Version: NaN.0.4`

When changing the Version to "1.00.04" it works.

**Expected behavior**
It should be possible also to use the server for Specs that are in the Release Candidate status.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - (x) Windows : version : _10_____
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - (x ) Other : specify : __Gitpod____

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
 2.71.0
  

- **Node:**
  `node --version` = v16.15.1

GoetzGoerisch, 2022-07-11:
@erossignon according to spec this is just a string.

https://reference.opcfoundation.org/Core/docs/Part6/F.2/

_UANodeSet_:
Version - String - The version of the model defined in the UANodeSet. This is a human readable string and not intended for programmatic comparisons.`

closed 2022-07-12 by erossignon

GoetzGoerisch, 2022-07-12:
Thank you @erossignon!

commit 6ea4d82b (2022-07-11): fixes #1163 parsing informative version string in nodeset and trying to convert to semver whenever possible
commit 582eb499 (2022-07-11): fixes #1163 parsing informative version string in nodeset and trying to convert to semver whenever possible
commit 9cc5ccf2 (2022-07-11): fixes #1163 parsing informative version string in nodeset and trying to convert to semver whenever possible
commit 54887662 (2022-07-12): fixes #1163 parsing informative version string in nodeset and trying to convert to semver whenever possible