node-opcua/node-opcua #1164: Problem with Guid in Structure (inside DataSetMeataData)

DrHaraldWeber, 2022-07-11:
**Current behavior**
Cannot start server because of NodeSet version with characters inside

**Describe the bug**
Using a model which contains a DataSetClass with description according to DataSetMetaData 

[Opc.Ua.PlasticsRubber.ImmToRobot.NodeSet2.xml.zip](https://github.com/node-opcua/node-opcua/files/9081316/Opc.Ua.PlasticsRubber.ImmToRobot.NodeSet2.xml.zip)

throws the error: 
`performPostLoadingTasks Err  =>   Cannot find reader for dataType ns=0;i=14 fieldTypename=Guid 
 Error:  Cannot find reader for dataType ns=0;i=14 fieldTypename=Guid`

Although the PusSub functionality is not needed in the test server, it is part of the Companion Specification.
The DataSetClasses are included under Objects -> Server -> PublishSubsribe -> DataSetClasses

It seems that there is a problem when a structure contains a Guid.

(Note: The linkes NodeSet creates a second problem reported as issue #1163)

**Expected behavior**
Guid should be possible in a structure.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - (x) Windows : version : _10_____
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - (x ) Other : specify : __Gitpod____

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
 2.71.0
  

- **Node:**
  `node --version` = v16.15.1

GoetzGoerisch, 2022-07-11:
@erossignon have a look at this commit: https://github.com/DrHaraldWeber/Sample-Server-node-opcua/commit/9895d549fa4e1d47aae93107d41d5287f61cc208

closed 2022-07-12 by erossignon

GoetzGoerisch, 2022-07-12:
Thank you @erossignon!

commit 59853e4b (2022-07-11): fixes #1164 - loading guid in extension object from nodeset2.xml file
commit ddd8362f (2022-07-12): fixes #1164 - loading guid in extension object from nodeset2.xml file