node-opcua/node-opcua #1165: Add a configuration parameter to specify OpenSSL directory on Windows

Jswk1, 2022-07-11:
**Is your feature request related to a problem? Please describe.**

Hello. I've been using this library in my project and recently we found out (it's not stated anywhere publicly), that the OpenSSL is required during runtime (https://github.com/node-opcua/node-opcua/issues/1009). This became a problem when one of my clients deployed my application in offline environment. It was not possible to connect to the OPC UA Server nor to download OpenSSL even though it was installed on the server. The issue came from path that is checked by `node-opcua-pki`:

[image]

As you see it is either loaded from APPDATA or from the place where the process was started. I was wondering if it was possible to specify a client parameter to be able to provide path to OpenSSL directly. In my case it is installed separately on my client env. and is freely available under `C:/Program Files/...etc` but it is not checked. 

**Describe the solution you'd like**
Add a paramter in `OPCUAClientOptions` and forward it to `node-opcua-pki`.

burgerni10, 2022-07-12:
You can create a specific clientCertificatManager by specifying the rootFodler of your certificates : 

```javascript
 const clientCertificateManager = new OPCUACertificateManager({ rootFolder: `${this.encryptionService.certsFolder}/opcua` })
 // Set the state to the CertificateManager to 2 (Initialized) to avoid a call to openssl
 // It is useful for offline instances where downloading openssl is not possible
 clientCertificateManager.state = 2
```

Then, you can use this clientCertificateManager when creating your client

```javascript
const options = {
  endpointMustExist: false,
  keepSessionAlive: this.keepSessionAlive,
  clientCertificateManager: this.clientCertificateManager,
}

const client = Opcua.OPCUAClient.create(options)
```

Note that the structure of your folder must match the structure expected by OPCUA Certificate Manager.

I used selfsigned package to generate self signed certificates

Jswk1, 2022-07-13:
Hello @burgerni10 

Thank you for the input. I will check your approach. For now I am rolling with a workaround for those interested - I use pm2 to host my application process and I overwrite the environment variable `LOCALAPPDATA` for this specific process that hosts node-opcua. Not the most elegant but it works.

erossignon, 2023-05-05:
[Jswk1](https://github.com/Jswk1), there is no need to have openssl installed if the server or client certificate and private key are created beforehand and if the `OPCUACertificateManager` is configured appropriately. 
openssl is only used to create default key pair (privateKey/self signed certificate) if they are not provided.

closed 2023-05-05 by erossignon