node-opcua/node-opcua #1167: Encoding variant Error after update to 2.72.0

AndreasHeine, 2022-07-13:
node-opcua: "2.72.0"

Error:
```
Error encoding variant Error
    at assert (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-assert\dist\index.js:11:21)
    at BinaryStream.writeByteStream (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-binary-stream\dist\binaryStream.js:280:40)
    at encodeByteString (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-basic-types\dist\byte_string.js:20:12)     
    at encodeVariant (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-variant\dist\variant.js:196:13)
    at Variant.encode (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-variant\dist\variant.js:90:9)
    at encodeDataValue (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-data-value\dist\datavalue.js:58:25)
    at encodeArray (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-basic-types\dist\array.js:22:9)
    at ReadResponse.encode (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-types\dist\_generated_opcua_types.js:19079:50)
    at MessageChunker.chunkSecureMessage (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-secure-channel\dist\source\message_chunker.js:60:17)
    at ServerSecureChannelLayer.send_response (C:\Users\andre\Documents\SampleServer-node-opcua\node_modules\node-opcua-secure-channel\dist\source\server\server_secure_channel_layer.js:414:29)
Variant(Scalar<ByteString>, value: 0x)
```

Reproduce:
1. 'npm run start' -> https://github.com/AndreasHeine/SampleServer-node-opcua
2. Connect with UA Expert
3. Error!

### Issue looks like related to the TypeDictionarys in MachineTools and IA Nodesets!

for Reference:
@GoetzGoerisch

[image]

closed 2022-07-13 by erossignon

erossignon, 2022-07-13:
You need 2.72.1 just released

GoetzGoerisch, 2022-07-13:
Already in https://github.com/AndreasHeine/SampleServer-node-opcua/pull/243

GoetzGoerisch, 2022-07-14:
Thank you ❤️  @erossignon fixed with `2.72.2`

commit 71c22f9d (2022-07-13): bug(numerical-range) fix buffer adjustment (fixes #1167)