node-opcua/node-opcua #1169: node-opcua have different behavior on linux and windows

VergilDMC00, 2022-07-13:
goodmorning, I have a problem with node-opcua, I created a software that uses node opcua to connect to a PLC, the problem is that when i run my software on a pc with OS windows when i turn off the connession wheter i fisically disconnect the cable from the PLC or turn off the wifi on my pc, automaticaly the software close the connection and try to reconnect. My problem is that  when i run the same software on a linux device, the only two thing that happen are the request timeout and the expeitation of the secure Channel token

Steps to reproduce the behavior:

1. .... run  the program and let it go for few minutes
2. .... turn off connecction
3. .... let the programm run for a long time

**Expected behavior**

i expect the same behavior the software have on my windows device, when i shutdown connection it has to try to recconect until i stop the programm or turn on the `connection`

//PROGRAM

const opcua = require("node-opcua");
const async = require("async");   
const client = opcua.OPCUAClient.create({endpointMustExist: false});

let ip = "[address:port of the device you want to connect to]"
let read = "[address of the item you want to monitor]"  example: ns="";s=""

client.on("connected",()=>{console.log("SUCCESSUFULY CONNECTED")})
client.on("connection_failed",()=>{console.log("CONNECTION FAILED")})
client.on("start_reconnection",()=>{console.log("START RECONNECTION")})
client.on("connection_reestablished",()=>{console.log("CONNECTION REESTABLISHED")})
client.on("timed_out_request",()=>{console.log("TIMED OUT REQUEST")})
client.on("connection_lost",(err)=>{console.log("LOST CONNECTION TO ",ip)})
    
async.series([
    function(callback)
    {
        const endpointUrl = "opc.tcp://" + ip;
        console.log({endpointUrl})
        client.on("backoff", (retry, delay) => 
            console.log("still trying to connect to ", endpointUrl ,": retry =", retry, "next attempt in ", delay/5000, "seconds" )
        );
        client.on("abort",()=>{console.log("conncted to ", ip)})
        // step 1 : connect to
        client.connect(endpointUrl, function (err) {
            if(err)
            {
                console.log(" cannot connect to endpoint :" , endpointUrl );
                callback(err)
            } 
            else 
            {
                console.log("connected !");
                callback()
            }
        });
            
    },
    function(callback)
    {
        client.createSession(function (err, session) {
        if (err) {
        console.log("Session creation error: ",err);
            callback(err);
        }
            the_session = session;

            the_session.on("disconnect",()=>{console.log("SESSION DISCONNECTED")})
            the_session.on("close",()=>{console.log("SESSION CLOSED")})

            callback();
            console.log("Session created");
        });
    },        
    // step 3 : browse
    function(callback) 
    {
        the_session.browse("RootFolder",
            function (err, browseResult) 
            {
                if (!err) 
                {
                    console.log("Browsing rootfolder: ");
                    for (let reference of browseResult.references) 
                    {
                        console.log(reference.browseName.toString(),reference.nodeId.toString());
                    }
                } 
            });        
        callback()
    },   
    function(callback)
    {  
        const maxAge = 0;
        const nodeToRead = {
          nodeId: read,
          attributeId: opcua.AttributeIds.Value,
        };
        the_session.read(nodeToRead, maxAge, function (err, dataValue) {
        if (!err) 
        {
            var resObj = {
              value: dataValue.value.value,
              valueDesc: dataValue.statusCode._name,
            };

            const subscriptionOptions = {
                maxNotificationsPerPublish: 1000,
                publishingEnabled: true,
                requestedLifetimeCount: 100,
                requestedMaxKeepAliveCount: 10,
                requestedPublishingInterval: 1000,
            };

            the_session.createSubscription2(
                subscriptionOptions,
                (err, subscription) => {
                  if (err) 
                  {
                    console.log({err});
                  }
                  else
                  {
                    the_subscription = subscription;
                    the_subscription
                      .on("started", () => {
                        console.log(
                          "subscription started for 2 seconds - subscriptionId=",
                          the_subscription.subscriptionId
                        );
                      })
                      .on("keepalive", function () {
                        console.log("subscription keepalive");
                      })
                      .on("terminated", function () {
                        console.log("terminated");
                      })
                      .on("error", function () {
                        console.log("error");
                      })
                      .on("internal_error", function () {
                        console.log("internal_error");
                      })
                      .on("close",()=>{console.log("SUBSCRIPTION CLOSED")})
                      .on("disconnect",()=>{console.log("SUBSCRIPTION DISCONNECTED")});

                    // install monitored item
                    const itemToMonitor = {
                        nodeId: opcua.resolveNodeId(read),
                        attributeId: opcua.AttributeIds.Value,
                    };
                    const monitoringParamaters = {
                        samplingInterval: 100,
                        discardOldest: true,
                        queueSize: 100,
                    };
                    const monitoredItem = the_subscription.monitor(
                        itemToMonitor,
                        monitoringParamaters,
                        opcua.TimestampsToReturn.Both,
                        (err, monitoredItem) => {
                             monitoredItem.on("changed", (dataValue) => 
                            {
                                console.log("resObj: ",resObj)
                                console.log("valore = ", dataValue.value.value);
                                console.log("-------------------------------------");
                            });
                        }
                    );
                }        
            });
        }});
    }
])

//

**Log of the error*
when i shutdown the connection this is the log of the error

subscription keepalive
subscription keepalive
TIMED OUT REQUEST
TIMED OUT REQUEST
TIMED OUT REQUEST
TIMED OUT REQUEST
TIMED OUT REQUEST
11:55:13.096Z :client_secure_channel_layer   :865   Invalid socket => Communication has been lost, cannot renew token

**Context**

- (X ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - (X ) other : nodejs

    - ( ) Windows : version : ______
    - (X ) Linux : version : i dont know the version im sorry
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- node-opcua version: 2.71.0

- **Node:**
  `node --version` = 16.16.0

VergilDMC00, 2022-07-18:
@erossignon where do i modified the open/close state?

robertsLando, 2023-05-08:
I have this issue too, suggestions about how to fix this? This is causing my opcua client to stay stucked and keep connecting/reconnecting until I recreate it

erossignon, 2023-05-08:
@robertsLando, this issue is closed , and will not be addressed anymore.
Please open a new bug issue and fill the template, providing gist to reproduce the issue. 
Alternatively, sterfive can provide consulting service to help you review your implementation code in private if your company is subscribing to the ["NodeOPCUA Support Membership" program](http://support.sterfive.com)

erossignon, 2025-04-07:
@VergilDMC00 @robertsLando  this is now fixed in [node-opcua@2.151.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.151.0)