node-opcua/node-opcua #1170: OPC UA Non-Compliance: Endpoint description information returned by node-opcua server does not match.

Graos, 2022-07-18:
NODE-OPCUA stack version : 2.72.2
NODEJS version : v16.15.1

Current Behavior:
 
Endpoint description information returned by node-opcua server in the GetEndPoint Response does not match with the Endpoint description information returned in the Create session response.
Snapshot of wireshark capture GetEndPoint response where the Product URI is returned
[image]

However in the below CreateSession response wireshark capture snapshot Product URI is empty

[image]

**Expected behavior**
As per the OPC UA compliance spec, endpoint description returned in the GetEndpointResponse and in the CreateSessionResponse by OPC UA server should match exactly.

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

JeanArhancet, 2022-08-25:
Hello, 

I tried to reproduce from the master branch but I can't (the endpoint client has a value in this branch https://github.com/node-opcua/node-opcua/compare/master...JeanArhancet:node-opcua:test/1170)

Graos, 2022-08-26:
Please read my description again and confirm. I have mentioned ProductUri("nodeOPCUA-server") is returned in GetEndPointResponse but not in CreateSessionResponse. Have you tried checking he ProductUri in CreateSessionResponse ? Does the server sends a proper string ?

JeanArhancet, 2022-08-27:
You're right @Graos, I opened a PR to fix this issue

PR #1188 (2022-08-28): Fix producturi in session
Hi,

This is my first MR for this project :) 

Fix #1170
closed 2022-08-28 by erossignon