node-opcua/node-opcua #1171: monitorItems raising error message : Cannot read properties of undefined (reading 'value')

kevinjjjjj, 2022-07-20:
**Current behavior**
      the_subscription.monitorItems(
        itemsToMonitor,
        monitoringParamaters,
        TimestampsToReturn.Both,
        (err, monitoredItemGroup) => {
          monitoredItemGroup.on("changed", function (dataValue) {
            console.log(dataValue.value.value);
          });
          callback();
        }
      );

monitorItems is raising the following error message:
3:47:35.472Z :client_monitored_item_group_im:111   [NODE-OPCUA-W20] the monitoredItem.on('changed') handler has raised an exception.
 ...                                                error message : Cannot read properties of undefined (reading 'value')

**Describe the bug**

Attempt to use .monitorItems() to monitor multiple nodes is causing an error to occur.   

**Step by step instruction to reproduce**

let itemsToMonitor = [];

the_session.browse(
        "ns=3;s=85/0:Simulation", //NodeID
        function (err, browseResult) {
          if (!err) {
            for (let reference of browseResult.references) {
              // add each node to the array
              const itemToAdd = {
                nodeId: resolveNodeId(reference.nodeId.toString()),
                attributeId: AttributeIds.Value,
              };
              itemsToMonitor.push(itemToAdd);
            }
          }
        }
      );

const monitoringParamaters = {
        samplingInterval: 100,
        discardOldest: true,
        queueSize: 10,
      };

      the_subscription.monitorItems(
        itemsToMonitor,
        monitoringParamaters,
        TimestampsToReturn.Both,
        (err, monitoredItemGroup) => {
          monitoredItemGroup.on("changed", function (dataValue) {
            console.log(dataValue.value.value);
          });
        }
      );

**Expected behavior**

I am trying to monitor multiple nodes using monitorItems but it keeps raising the same error.
Code is a modified version of the example code on github. monitorItems seems to work well if I pass on a single item from the itemsToMonitor array (e.g. itemsToMonitor[1]) but it keeps raising error if I try to input an entire array. 

Looking at the document [https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua.clientmonitoreditemgroup.html#on](url) 
 monitoredItemGroup.on seems to take on extra arguments when compared to ClientMonitoredItem.on but I couldn't find any examples on this. 

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT

- ( ) I have installed node-opcua as a package ( using npm install )

- OS version: ______

    - ( ) Windows : version : 10

Honza123321, 2022-07-20:
Hi @kevinjjjjj ,

I had a similar problem like you and I found out, that If you add another parameter to function in "on changed handler", you can get the data for single item. 

I did it like this:
```
subscription.monitorItems(
 itemsToMonitor,
 monitoringParamaters,
 TimestampsToReturn.Both,
 (err, monitoredItems) => {
   monitoredItems.on("changed", function(items ,data){

    console.log("---------------------------------------------------");
    console.log("Node ID: ", items.itemToMonitor.nodeId.value);
    console.log("Node value: ", data.value.value);
    console.log("---------------------------------------------------");

    });
 }); 
```
I hope it will help you.

Have a nice day.

kevinjjjjj, 2022-07-20:
Hi @[Honza123321](https://github.com/Honza123321)

yes, you're 100% right. After posting this I had a play around with monitorItems and found that I need to add [ClientMonitoredItemGroup](https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua.clientmonitoreditemgroup.html) to make it work!

Hope this gets included in the client example for people trying it out for the first time in the future.

erossignon, 2022-07-22:
Thank you [Honza123321](https://github.com/Honza123321)

closed 2022-07-22 by erossignon