node-opcua/node-opcua #1174: Cannot find dataType Definitions Error. Program gets stuck after period of time.

david114, 2022-07-26:
Hi,

I'm currently running into a problem when reading large ExtensionObjects from a S7-1500 with Custom DataTypes.
At the first read process of those ExtensionObjects I'm getting a lot of error messages, telling me that the library can't find dataType Definitions. However, after those error logs, the read process works just fine and I can successfully access my data. 

Then, after a longer period of time, I'm experiencing a weird behaviour where the library will just print out the whole NodeSet Schema to the console and get stuck. I then have to kill restart the program and everything works fine again for like 1 hour.

Here is the error output:

```
17:20:35.217Z :populate_data_type_manager_104:50    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:34:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.240Z :populate_data_type_manager_104:177   err= Error
 ...                                                    at assert (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:175:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.293Z :populate_data_type_manager_104:50    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:34:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.294Z :populate_data_type_manager_104:177   err= Error
 ...                                                    at assert (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:175:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.347Z :populate_data_type_manager_104:50    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:34:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.348Z :populate_data_type_manager_104:177   err= Error
 ...                                                    at assert (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:175:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.441Z :populate_data_type_manager_104:50    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3500
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:34:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.443Z :populate_data_type_manager_104:177   err= Error
 ...                                                    at assert (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:175:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.732Z :populate_data_type_manager_104:50    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15889
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:34:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
17:20:35.734Z :populate_data_type_manager_104:177   err= Error
 ...                                                    at assert (C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-assert\source\index.ts:12:21)
 ...                                                    at C:\Users\me\Desktop\my-opcua-client\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:175:19
 ...                                                    at Generator.next (<anonymous>)
[... cut ...]
             { nodeId: 'ns=3;s="020_FIFO_OPCUA"."Actuals"' }
           ]
        );
        await writeToDatabase(processData);
        setTimeout(() => mainLoop(), 500);
      } catch (error : any) {
        logger.error(error.message);
        setTimeout(() => mainLoop(), 5000);
      }
    };
    mainLoop();

Is there something, I'm doing terrible wrong?
On the server-side there doesn't seem to be a problem. 
Support for <=1.3 dataTyles is enabled.

**I'm using version 2.72.2**

Help is highly appreciated!

erossignon, 2022-07-31:
The ExtensionObject definition is loaded out of the server during the first operation (like reading) that requires it.

If the server is restarted and the extension object definition has changed, the client needs to be restarted.
In your case, the app is periodically reading a set of nodes to store them somewhere; the best approach  is to perform a full connection everytime.

````
import { OPCUAClient, MessageSecurityMode, SecurityPolicy, DataValue } from "node-opcua";

const endpointUrl = "opc.tcp://localhost:4840";

async function writeToDatabase(dataValue: DataValue[]) {
    /** */
}
const mainLoop = async () => {
    try {
        const client = OPCUAClient.create({
            applicationName: "my-opc-client",
            securityMode: MessageSecurityMode.None,
            securityPolicy: SecurityPolicy.None,
            endpointMustExist: true
        });

        const processData = await client.withSessionAsync(endpointUrl, async (session) => {
            const processData = await session.read([
                { nodeId: 'ns=3;s="020_FIFO_OPCUA"."Setpoints"' },
                { nodeId: 'ns=3;s="020_FIFO_OPCUA"."Actuals"' }
            ]);
            return processData;
        });
        await writeToDatabase(processData);

        setTimeout(() => mainLoop(), 500);
    } catch (error: any) {
        setTimeout(() => mainLoop(), 5000);
    }
};
mainLoop();
````

closed 2022-08-09 by erossignon

cdac2019, 2022-09-08:
[image]
I am facing such type of issues
Help!

teropes, 2022-09-18:
I have faced similar issues with Siemens S7-1500 as the OPC UA server. Root cause in one of our problem cases was the following: In one of the DBs enabled in OPC UA server, there was some bit more 'exotic' Siemens datatype used and it seems that it didnt get correctly transformed into valid OPC UA datatype. I can't remember which datatype specifically it was, but maybe if you browse around your enabled DBs you might be able to figure it out. Drop out DBs from the Siemens interface one by one to pinpoint the issue and go from there.

Verschnupft, 2022-09-21:
@teropes, do you remember if the datatype causing this was a user defined type or a "normal" siemens datatype?

I tried to create a new, empty project for testing but I even encountered an error if I just try to read a struct with one integer variable.

cdac2019, 2022-09-28:
@david114  are you able to resove this issue ? is that happyen becuse of  session or siemens side datatype

david114, 2022-09-28:
> @david114 are you able to resove this issue ? is that happyen becuse of session or siemens side datatype

Unfortunately, I have no direct access to the Siemens PLC via TIA-Portal and I'm also not allowed to change any of the PLC's structure anyways. I'm relying on other people to fix this issue. @Verschnupft seems to have full access to his PLC though.

Verschnupft, 2022-09-28:
Hey,
I woun't be able to test things further until january/february (after the semester).
I don't think that my colleagues will find time to solve this issue since we don't have projects that need this functionality in the near future. 
Nevertheless I'm going to send this thread to my boss so he could contact you.

cedenoVF, 2022-10-13:
Hi All,

I have a similar issue, see below:
[image]

I am using standard datatypes from Siemens.