node-opcua/node-opcua #1175: serviceResult = BadTooManySubscriptions

rickypid, 2022-07-27:
My application when subscribing multiple items simultaneously throws an exception. The reason is clear to me as my server manages a maximum of 10 simultaneous publication requests, is there a solution to avoid this problem already integrated in this library to subscribe a maximum of items and queue the following items and avoid overloading the server?

- **Exception**

```
06:46:34.586Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
06:46:34.586Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  14
06:46:34.586Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 10
06:46:34.587Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
06:46:34.587Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  13
06:46:34.587Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 10
06:46:34.588Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
06:46:34.588Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  12
06:46:34.588Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 10
06:46:34.589Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
06:46:34.589Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  10
06:46:34.590Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 10
node:events:505
      throw er; // Unhandled 'error' event
      ^

Error:  serviceResult = BadTooManySubscriptions (0x80770000)

.....

  response: ServiceFault {
    responseHeader: ResponseHeader {
      timestamp: 2022-07-27T06:46:59.538Z {
        high_low: [ 30974340, -1414770561 ],
        picoseconds: 137500000
      },
      requestHandle: 283,
      serviceResult: ConstantStatusCode {
        _value: 2155282432,
        _description: 'The server has reached its maximum number of subscriptions.',
        _name: 'BadTooManySubscriptions'
      },
      serviceDiagnostics: DiagnosticInfo {
        symbolicId: -1,
        namespaceURI: -1,
        locale: -1,
        localizedText: -1,
        additionalInfo: null,
        innerStatusCode: ConstantStatusCode {
          _value: 0,
          _description: 'The operation succeeded.',
          _name: 'Good'
        },
        innerDiagnosticInfo: null
      },
      stringTable: [ '' ],
      additionalHeader: null
    }
  },
  request: CreateSubscriptionRequest {
    requestHeader: RequestHeader {
      authenticationToken: NodeId { identifierType: 1, value: 2447453622, namespace: 0 },
      timestamp: 2022-07-27T06:46:35.792Z {
        high_low: [ 30974340, -1652231936 ],
        picoseconds: 0
      },
      requestHandle: 283,
      returnDiagnostics: 0,
      auditEntryId: '',
      timeoutHint: 60000,
      additionalHeader: null
    },
    requestedPublishingInterval: 100,
    requestedLifetimeCount: 100,
    requestedMaxKeepAliveCount: 10,
    maxNotificationsPerPublish: 1,
    publishingEnabled: true,
    priority: 1
  }
}

Process finished with exit code 1

```

- **node-opcua version:** :

```
    "node-opcua": "^2.72.2",
    "node-opcua-client": "^2.72.2",
```

- **Node:**
  `node --version` = `v16.15.1`

AndreasHeine, 2022-07-27:
how many subscriptions do you use in your client application?
usually one should be enough!

rickypid, 2022-07-27:
From the log I would say about 15

```
06:46:34.586Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
06:46:34.586Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  14
```

AndreasHeine, 2022-07-27:
a subscription can have many monitored items! so you just need one in most cases, only if you hit the boundary of the "MaxMonitoredItemsPerSubscription" you need more then one!

AndreasHeine, 2022-07-27:
In simple words:
A OPC UA Subscription works like a "Mailbox" which gets emptied in a defined interval by the Client.
The OPC UA Client sends a publishrequest, the server takes all notifications since the last publishrequest.
and send them in the publishresponse to the client.
If there is no Notification in the "Mailbox", after a while the Client will get a "keep-alive" back and then the client will send another publishrequest.
The OPC UA Client can add "MonitoredItems" to the Subscription which will generate Notifications.

gkerry, 2022-07-28:
I got same WARNINGs as yours. But the program still running without throw error information.
There are more than 20 devices on my server channel. So I have to subscribe to more than 20 nodes corresponding to it?
I'd love to know how to fix these warnings.

AndreasHeine, 2022-07-28:
@gkerry  
20 nodes (20 MonitoredItems) in one Subscription?

mikakaraila, 2022-07-28:
It can be also server limitations. I have made client & server combinations for 100 000 variables that uses 20 subscriptions each containing 5000 variables without any problem, but you should check server diagnostics with UaExpert => Capabilities and then OperationLimits. New v1.05 will define more options to limit server for DoS attacks.

[image]

And compare above to current used ones:
[image]

gkerry, 2022-07-28:
> @gkerry 20 nodes (20 MonitoredItems) in one Subscription?

I'm not sure if that's one subscription. Sorry to my bad expression in English
In this node-opcua client project,  monitoredItem.on "changed" Event. with  my  NodeArry(23 or more Items) foreEach method. then...

AndreasHeine, 2022-07-28:
its a KEPServerEx right?

gkerry, 2022-07-28:
> its a KEPServerEx right?

yes, indeed.  and what's @mikakaraila 's Server ?  I didn't find these options you mentionded.

AndreasHeine, 2022-07-28:
@gkerry @mikakaraila
KEPServerEx does not support Session diagnostic...

gkerry, 2022-07-28:
> @mikakaraila KEPServerEx does not support Session diagnostic...
Huh? Does that mean I can ignore it?😂

AndreasHeine, 2022-07-28:
```js
const opcua = require("node-opcua");

async function main() {
    const endpointUrl = "opc.tcp://127.0.0.1:49320";
    const client = opcua.OPCUAClient.create({endpoint_must_exist: false});
    client.on("backoff", () => console.log("backoff"));
    await client.connect(endpointUrl);
    const session = await client.createSession({});

    const subscription = await session.createSubscription2({
        maxNotificationsPerPublish: 9000,
        publishingEnabled: true,
        requestedLifetimeCount: 10,
        requestedMaxKeepAliveCount: 10,
        requestedPublishingInterval: 1000
    });

    const nodeListToMonitor = [
        {
            attributeId: opcua.AttributeIds.Value,
            nodeId: opcua.resolveNodeId("ns=2;s=Beispiele für Simulation.Funktionen.Ramp1")
        },
        {
            attributeId: opcua.AttributeIds.Value,
            nodeId: opcua.resolveNodeId("ns=2;s=Beispiele für Simulation.Funktionen.Ramp2")
        },
        {
            attributeId: opcua.AttributeIds.Value,
            nodeId: opcua.resolveNodeId("ns=2;s=Beispiele für Simulation.Funktionen.Ramp3")
        },
        {
            attributeId: opcua.AttributeIds.Value,
            nodeId: opcua.resolveNodeId("ns=2;s=Beispiele für Simulation.Funktionen.Ramp4")
        }
    ]

    const parameters = {
        discardOldest: true,
        queueSize: 10,
        samplingInterval: 2000
    };

    const monitoredItems = nodeListToMonitor.map((item) => {
        const mItem = opcua.ClientMonitoredItem.create(subscription, item, parameters, opcua.TimestampsToReturn.Both);

        mItem.on("changed", (dataValue) => {
            console.log(`${item.nodeId} Changed on `, dataValue.value.toString());
        });

        return mItem
    })

    setTimeout(async () => {
        monitoredItems.forEach(async (item) => {
            await item.terminate();
        })
        await subscription.terminate();
        await session.close();
        await client.disconnect();
    }, 10000)
}
main()
```

could you try to reproduce it with that template?

gkerry, 2022-07-28:
> could you try to reproduce it with that template?

Okay, I'll give it a try

rickypid, 2022-07-28:
Hi guys,

This is a my current situation:

```mermaid
graph LR
    A--- B
    subgraph PLC
    A[OPC UA Server]
    end
    subgraph Backend API
    B[OPC UA Client]
    C[Socket.io Server]
    B --- C
    end
    subgraph Clients
    C -->|One| D[Socket.io Client]
    C -->|Two| E[Socket.io Client]
    C -->|Three| F[Socket.io Client]
    end
```

How my application works:

The backend establishes the connection to the OPC UA server without subscribing to any elements. When a customer connects, he requests the subscription of one or more elements (Through Socket.io).

When this happens, the error I have described occurs, but only at the beginning of the request when the client requests 15 elements at the same time. The OPC UA server has no problems in subscribing even hundreds of elements, the only problem is when there is a contemporaneity in the request. So if there was the possibility of limiting the number of simultaneous subscriptions and queuing the next ones as soon as the first ones are really subscribed, the problem would not exist.

AndreasHeine, 2022-07-28:
@rickypid  do you create a subscription for each socket io client?

rickypid, 2022-07-28:
Yes, because when the client disconnects from the backend, all its monitored items are removed by removing the matched subscription.

AndreasHeine, 2022-07-28:
then you can never server more socket io clients then the servers "MaxSubscriptionCount" which starve the server... you create a lot of traffic on the wire!

you should rethink on using just one subscription! add and remove monitoreditems dynamically!

AndreasHeine, 2022-07-28:
@gkerry you have a different issue! could you create a own issue with the versions of node, node-opcua, kepserverex...

rickypid, 2022-07-28:
This problems is appared when i upgrade a node-opcua package:

[image]

With the previous version the application had no throws an exception but only a warning.

```bash
[NODE-OPCUA-W22] Warning : your opcua client is sending multiple requests simultaneously to the server CreateMonitoredItemsRequest 
 ...                                                  please fix your application code
```

I didn't understand what your solution would be, create a subscription for each item?

If your concern about traffic and therefore the creation of many subscriptions linked to you have clients is not too worrying as the clients are never many for each backed, max 2/3 clients

AndreasHeine, 2022-07-28:
> I didn't understand what your solution would be, create a subscription for each item?

you completely misunderstood me...

gkerry, 2022-07-28:
@AndreasHeine 
ok, Thank you all the same

AndreasHeine, 2022-07-28:
@rickypid 

[image]

and if you add socket io clients you just hook onto the "EventEmitter" and add the necessary items

rickypid, 2022-07-28:
@AndreasHeine Thanks,

Your design is exactly my application, with only one difference: there is a subscription for each client.

Log of OPC UA Server when one client is connected to backend API:

[image]

Log of OPC UA Server when two clients are connected to backend API:

[image]

gkerry, 2022-07-28:
> and i cant reproduce....
it's a weird issue

AndreasHeine, 2022-07-28:
@rickypid PLCs have very limited ressources... especially the smaller models! from my own experience i can tell you the one subscription solution is working well (also for Siemens PLC's) the only thing which makes it complex is if you hit the "MaxMonitredItemsPerSubscription" count then you need another Subscription (adding/removing items, checking if your at the limit in one of the subscription ^^)

rickypid, 2022-07-28:
The PLC in question is really very powerful as it is a Soft PLC, that is, it runs on a core of a PC processor, I have a lot of experience with OPC UA on PLCs and I agree with you that on small / normal PLCs there are many limitations, in this case I believe that there are no such problems.

I opened the issue as with the old version I had that warning but the program did not issue exceptions.

Secondly, I wanted to know if there was a possibility (or a feature already integrated in your library) that it would queue requests and limit the number of concurrent requests, queuing the next ones. If this is not possible I take care of creating a queue in the API backend that does this.

I didn't want to take any more precious time from you. 😃

rickypid, 2022-07-28:
@AndreasHeine  I have created a queue in the API backend that does this,

Thank for your soppurt!

closed 2022-07-28 by rickypid

erossignon, 2022-07-30:
> Secondly I want to know if there was a possibility (or a feature already integrated in your library)  that it would queue requests and limit the number of concurrent requests, queuing the next ones

Sterfive, the company that develops NodeOPCUA , provides a commercial modules that provides this feature.  this is called "node-opcua-optimized-client".

This library takes into account the actual limit of the connected server and split multiple read/write/browse etc. into appropriate chunks and reassembles the response so that it is seamless for the client.
It also provides a queue mechanism to make sure that only a limited number of transactions are processed by the servers;
It also combines small semi-asynchronous reads and writes into single operations so you can reduce transaction round trip to the server.
It also holds any transaction such as read/write/browse etc .. when the server is disconnected and provides a mechanism to resume them if needed when the server is reconnected.  

Please contact me at contact@sterfive.com for more info if you are interested,