node-opcua/node-opcua #1178: installing node-opcua from source fails

tmtron, 2022-08-02:
**Current behavior**

[Installing-node-opcua-from-source](https://github.com/node-opcua/node-opcua#installing-node-opcua-from-source) does not work.

I follow the instructions on Windows 10 in a GitBash

**Describe the bug**
`pnpm install` fails:

```
> generateWithPNPM
> pnpm recursive run generate

Scope: 89 of 90 projects
packages/node-opcua-types generate$ tsc -b tsconfig.pre.json && ts-node source/generate && tsc -b
packages/node-opcua-types generate: ../node-opcua-utils/source/watchdog.ts(5,24): error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.
packages/node-opcua-types generate: ../node-opcua-utils/source/set_deprecated.ts(6,24): error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.
packages/node-opcua-types generate: ../node-opcua-utils/source/index.ts(5,24): error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.
packages/node-opcua-types generate: Failed
undefined
E:\tmp\node-opcua-test\node-opcua\packages\node-opcua-types:
 ERR_PNPM_RECURSIVE_RUN_FIRST_FAIL  node-opcua-types@2.72.2 generate: `tsc -b tsconfig.pre.json && ts-node source/generate && tsc -b`
Exit status 2
 WARN   Local package.json exists, but node_modules missing, did you mean to install?
 ELIFECYCLE  Command failed with exit code 1.
 ELIFECYCLE  Command failed with exit code 1.

```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

```
 $ git clone https://github.com/node-opcua/node-opcua.git
 $ cd node-opcua
 $ npm install -g pnpm 
 $ pnpm install
```

**Expected behavior**

Installation should work as described in the Readme file

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- (x ) I have installed node-opcua from source ( using git clone)
    - (x ) Windows : version : Windows 10 64 bit

- **node-opcua version:** :
  
  

```
$ git rev-parse HEAD
47a877cded9cbe11610ef583b6d59eb9b4f27a60
```

- **Node:**
  `node --version` = `v16.15.0`

tmtron, 2022-08-03:
the same issue occurs when I try the installation in WSL2

```
uname -a
Linux DESKTOP-XXX 4.19.128-microsoft-standard #1 SMP Tue Jun 23 12:58:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

node -v
v16.15.0
```

erossignon, 2022-08-03:
There migth be a typo in the readme file, please try 
```
   $ pnpm recursive install
````

closed 2022-08-03 by erossignon