node-opcua/node-opcua #1185: Connecting to a server with security policy in place

kevinjjjjj, 2022-08-15:
**Current behavior**

node-opcua client is unable to access a server with security policy set in place. 

**Describe the bug**

Connection is actively rejected by the server. 

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Anonymous login is prohibited and security policy is set to Basic256Sha256 on the server.
2. Try to connect with the following instance of a client:
`client = OPCUAClient.create({
      endpointMustExist: false,
      securityMode: MessageSecurityMode.SignAndEncrypt,
      securityPolicy: SecurityPolicy.Basic256Sha256,
      requestedSessionTimeout: 10000,
      clientName: "webHMI",
      // applicationName: "NodeOPCUA-Client", optional..
      // certificateFile: "certificates/cert.pem",
      // privateKeyFile: "certificates/key.pem",
      // serverCertificate: crypto_utils.readCertificate(
      //   "trusted_certificates/cert.pem"
      // ), crypto_utils is a separate library
    });`

notice that I have commented out the certificateFile, privateKeyFile, and serverCertificate parameters and I'm guessing this is the cause of the issue. I have created a self signed certificate for the client side and also have a cert from the server side as well but I am not quiet sure on what is going on with the crypto_utils library and there is no documentation regarding this issue on the Github. Is this still relevant or has there been a change on this?

3. create a session:

`var userIdentity = {
            userName: "user",
            password: "password",
          };
          //client.createSession(userIdentity, function (err, session)
          client.createSession(userIdentity, function (err, session) {
            if (err) {
              return callback(err);
            }
            the_session = session;`

4. Following error message is displayed when trying to connect:
`failure  Error: The connection may have been rejected by server,
Err = (_socket has been disconnected by third party)`

**Expected behavior**

able to connect to the server with security policy in place.

**Screenshots**

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT

- ( ) I have installed node-opcua as a package ( using npm install )

- Device: PC
- OS version: ______

    - ( ) Windows : version : 10

- Description of the other OPCUA system I am trying to connect to:

    - Name: NX102
    - Version: 1.27
    - Manufacturer/Software vendor: Omron
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 2.73.1

  
  

- **Node:**

  `node --version` = 18.2

erossignon, 2022-08-20:
* you have to make sure that the client certificate is added to the trusted certificate list of the server.

kevinjjjjj, 2022-08-25:
> * you have to make sure that the client certificate is added to the trusted certificate list of the server.

Yes, this had been done prior to this problem occurring and the connection worked well when securityMode and securityPolicy were set to 'none'.

Going slightly tangent, is there a way to generate a self signed certificate from the client side? I noticed that certificate has been automatically generated at \AppData\Roaming\node-opcua-default-nodejs\Config\PKI\own but I would like to tweak some of its properties but I can't figure out a way to generate another certificate with custom properties.

Many thanks in advance.

erossignon, 2022-08-25:
you can use the node-opcua-pki package, it comes with a command line tool to facilitate PKI and certificate creation.

kevinjjjjj, 2022-08-25:
thank you, I tried the instruction on this page: https://github.com/node-opcua/node-opcua-pki 

I've tried both npx and running with node command but can't seem to get it to work. Is there an instruction on how to run this demo in windows environment?

erossignon, 2022-08-31:
It works on linux,windows and mac ,
The easiest way is to start it using npx, like this:
 ```
npx node-opcua-pki --help
npx node-opcua-pki certificate --help
```

closed 2023-11-15 by erossignon