node-opcua/node-opcua #1190: BindVariableOptionsVariation1 getFunc can not be use async function

MichaelTsengXD, 2022-08-31:
**Current behavior**
BindVariableOptionsVariation1 getFunc can not be use async function

**Describe the bug**
In server, BindVariableOptionsVariation1 getFunc can use async function?
I try to use async function in BindVariableOptionsVariation1 getFunc, but occur "Bind variable error:   : the getter must return a Variant or a StatusCode... value_check.constructor.name  Promise"

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Red frame, I use get: async () => { // async function get value to value };
[image]

**Expected behavior**

**Screenshots**
Code
[image]

Error Message
[image]

**Gist**

 
gist:

**Context**

- ( O ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( O ) Windows : version : _Win10_
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ___Kepware___
    - Version: ___V6.11.764.0___
    - Manufacturer/Software vendor: ___PTC___
    - link to the manufacturer or product site: https://www.kepware.com/en-us/

- **node-opcua version:** :

  
  

- **Node:**

  `node --version` = v14.20.0

AndreasHeine, 2022-08-31:
you need to look in BindVariableOptionsVariation2 -> timestamped_get -> !!! callback-pattern !!!

```js
timestamped_get: function(
    this: UAVariable, 
    callback: (err: Error | null, dataValue?: DataValue) => void): void {
    somePromiseReturningTheValue
    .then((val) => {
        callback(null, new DataValue({
            value: new Variant({
                value: val,
                dataType: dtype
            }),
            statusCode: StatusCodes.Good,
        }));
    })
}
```

MichaelTsengXD, 2022-08-31:
> you need to look in BindVariableOptionsVariation2 -> timestamped_get -> !!! callback-pattern !!!
> 
> ```js
> timestamped_get: function(
>     this: UAVariable, 
>     callback: (err: Error | null, dataValue?: DataValue) => void): void {
>     somePromiseReturningTheValue
>     .then((val) => {
>         callback(null, new DataValue({
>             value: new Variant({
>                 value: val,
>                 dataType: dtype
>             }),
>             statusCode: StatusCodes.Good,
>         }));
>     })
>     .catch((err) => {
>         //...
>     })
> }
> ```

BindVariableOptionsVariation2 -> timestamped_get can use, but occur "asyncRefresh error: Variable is   ns=2;s=Variable01 2:Variable01"、"refresh call failed 2:Variable01 ns=2;s=Variable01 Cannot create property 'dataType' on string 'value' "...

Code
[image]

Error Message
[image]

How do I fix my code ? Thank!

AndreasHeine, 2022-08-31:
you can not use async/await syntax directly...!

AndreasHeine, 2022-08-31:
timestamped_get expects a function like this:
function(this: UAVariable, callback: (err: Error | null, dataValue?: DataValue) => void): void

just use getValue(key) as a promise:

```js
getValue(key).then((val) => {
    callback(null, new DataValue({
        value: new Variant({
            value: val,
            dataType: dtype
        }),
        statusCode: StatusCodes.Good,
    }));
})
.catch((err) => {
    //...
})

```

MichaelTsengXD, 2022-09-07:
> You need to use the second form to get async method working as you expect
> 
> ```ts
> export type GetFunc = (this: UAVariable) => Variant;
> export type SetFunc = VariableSetter | null;
> 
> export type VariableDataValueGetterSync = (this: UAVariable) => DataValue;
> export type VariableDataValueGetterPromise = (this: UAVariable) => Promise<DataValue>;
> export type VariableDataValueGetterCallback = (this: UAVariable, callback: CallbackT<DataValue>) => void;
> 
> export type VariableDataValueSetterWithCallback = (this: UAVariable, dataValue: DataValue, callback: CallbackT<StatusCode>) => void;
> export type VariableDataValueSetterWithPromise = (this: UAVariable, dataValue: DataValue) => Promise<StatusCode>;
> 
> export type TimestampGetFunc = VariableDataValueGetterSync | VariableDataValueGetterPromise | VariableDataValueGetterCallback;
> export type TimestampSetFunc = VariableDataValueSetterWithCallback | VariableDataValueSetterWithPromise;
> 
> export interface BindVariableOptionsVariation1 {
>     get: GetFunc;
>     set?: SetFunc;
>     timestamped_get?: undefined;
>     timestamped_set?: undefined;
>     historyRead?: HistoryReadFunc;
> }
> export interface BindVariableOptionsVariation2 {
>     set?: undefined;
>     get?: undefined;
>     timestamped_get: TimestampGetFunc;
>     timestamped_set?: TimestampSetFunc;
>     historyRead?: HistoryReadFunc;
> }
> ```
> 
> timestamped_get can be a VariableDataValueGetterPromise `(this: UAVariable) => Promise<DataValue>;`
> 
> in this case you need to return a DataValue ( with Statuscode and timestamps) instead of a Variant

> timestamped_get expects a function like this: function(this: UAVariable, callback: (err: Error | null, dataValue?: DataValue) => void): void
>
> just use getValue(key) as a promise:
>
> ```js [...]

MichaelTsengXD, 2022-09-07:
Thanks for everyone's reply, this issue has been resolve.

closed 2022-09-07 by MichaelTsengXD