node-opcua/node-opcua #1191: env-paths dependencies is missing from a couple of packages.

crispinboylan-int, 2022-09-02:
**Current behavior**

> node index

node:internal/modules/cjs/loader:958
  throw err;
  ^

Error: Cannot find module 'env-paths'
Require stack:
- /usr/src/app/device-controller/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js
- /usr/src/app/device-controller/node_modules/node-opcua-certificate-manager/dist/index.js

**Describe the bug**

Programs using node opcua crash with missing env-paths module.

Seems it is used in the following places:

node-opcua-certificate-manager/source/certificate_manager.ts:import envPaths = require("env-paths");
node-opcua-server-discovery/source/opcua_discovery_server.ts:import envPaths = require("env-paths");

but never mentioned in the corresponding package.json.

- **node-opcua version:** : 2.76.0

  
  

- **Node:**

  `node --version` = 18.7.0

GoetzGoerisch, 2022-09-02:
Was removed here: 3c7743b73ccb9e24f80638b08e304b6b46f74379

erossignon, 2022-09-03:
Well spotted, let me fix this,

erossignon, 2022-09-03:
fixed in node-opcua@2.76.2

closed 2022-09-03 by erossignon