node-opcua/node-opcua #1193: Invalid error message on unlimited maxMessageSize (maxMessageSize = 0)

dani162, 2022-09-06:
**Current behavior**

The OPC-UA Client receives a acknowledge message with the max message size set to unlimited (=0) and set the maxMessageSize of the message chunker to that received value.

If the OPC-UA Client now sends a request it always print error messages, because the message size of these messages is greater than zero (see screenshot below)

maxMessageSize = 0 is described as unlimited:
https://reference.opcfoundation.org/v105/Core/docs/Part6/7.1.2/#7.1.2.4

https://github.com/node-opcua/node-opcua/blob/177f468a9bcdedb98fab3d3fb2f87a8af0378e0e/packages/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts#L1184-L1186

https://github.com/node-opcua/node-opcua/blob/156520cbcb95c0f059ffa1aff743cc23b29ae95c/packages/node-opcua-secure-channel/source/message_chunker.ts#L208-L212

**Describe the bug**

The OPC-UA client prints invalid error messages

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Connect the OPC-UA Client to a OPC-UA Server with a unlimited MaxMessageSize (=0)
2. Send a Request to the OPC-UA Server

**Expected behavior**

No invalid error messages 

**Screenshots**

Error Message:
[image]

Acknowledge message:
[image]

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT

- (x) I have installed node-opcua as a package ( using npm install )

- OS version:

    - (x) Windows : version : 11

- Description of the other OPCUA system I am trying to connect to:

    - Close Sourced OPC-UA Server

- **node-opcua version:** : 2.76.2

- **Node:**

  `node --version` = v16.17.0

antoniosteiger, 2022-09-07:
I have the exact same error. Fix suggestion would be appreciated.

erossignon, 2022-09-07:
Well spotted, will be fixed in next build

commit 660067e5 (2022-09-07): fix #1193 - client should not complain about over sized messages, if AcknowledgeMessage.MaxMessageSize is set to Zero by the server
commit 2b9fb880 (2022-09-08): fix #1193 - client should not complain about over sized messages, if AcknowledgeMessage.MaxMessageSize is set to Zero by the server
closed 2022-09-08 by erossignon