node-opcua/node-opcua #1194: connection lost event triggering is taking long time up to 40 seconds

raptorsj, 2022-09-07:
**Current behavior**

**Describe the bug**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. ....
2. ....
3. ....

**Expected behavior**

**Screenshots**

**Gist**

 
gist:

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 2.71.0
- 

  
  

- **Node:**

  `node --version` = 16.14.2

erossignon, 2022-09-08:
You should not kill the client but rather properly close the session and disconnect the client using the provided API.

Abruptly killing the client will cause the server to keep the session object on the server side alive for a duration that is defined in the `requestedSessionTimeout: 30000` parameter above. (the server may decide to adjust the value to a more appropriate one) 

The server keeps the session open on its end in case the disconnection was due to a network outtage to allow the communication to resume seemlessly.

erossignon, 2023-05-04:
Some improvements have been made toward detecting socket timeout inside [node-opcua@2.101.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.101.0) 
The problem should be fixed now and detection 
This problem was apparent especially when running node-opuca client inside a docker container

@raptorsj , can you give it a try and tell me if the problem is now fixed on your end with node-opcua@2.102.0 ?

raptorsj, 2023-05-04:
will check and update

erossignon, 2024-03-06:
@raptorsj: any update ?

Rednas83, 2024-08-11:
I also get the `ETIMEOUT` after 40sec on the first attempt.
[image]
Even after setting it to 1sec
```
const options = {
  applicationName: "opcua",
  connectionStrategy: connectionStrategy,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
  endpointMustExist: true,
  keepSessionAlive: true, // https://github.com/node-opcua/node-opcua/issues/341
  requestedSessionTimeout: 1000, // https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.opcuaclientoptions.html
}
```

That is a very long timeout. It's probably faster to ping🤣

erossignon, 2024-08-11:
timeouts are provided in miliseconds.

1000 miliseconds is actually a very small value,
I suggest that you do not change requestedSessionTimeout from default value, unless you know what you are doing. 

Simply remove the line `requestedSessionTimeout: 1000` to revert back to the default

Rednas83, 2024-08-11:
Why should a network/session timeout take 40 sec or longer?

Usually it can be done in something like 50ms.

Waiting for 40sec to get a network timeout is bit annoying. Is there really now way to make it faster?

AndreasHeine, 2024-08-12:
@Rednas83 opc ua distinguishes between transport/connection-timeout and session-timeout... (different layers in the OSI-Model) Session contain information about useridentity and should not be shortlived by design! Sessions are designed to be reused / reactivated if a connection broke! 

with your approach of subsecond sessiontimout you would not be able to restore the session at all... which is a common missconseption!

> RequestedSessionTimeout:
Requested maximum number of milliseconds that a Session should remain open without activity. If the Client fails to issue a Service request within this interval, then the Server shall automatically terminate the Client Session. 

Reference:
https://reference.opcfoundation.org/Core/Part4/v105/docs/6.7

Rednas83, 2024-08-13:
I currently get a `ETIMEOUT` after 40sec after a **connect while not connected** and after a **disconnect** I immediatly get `ECONNREFUSED`.

Is it not better to also have something like `EDISCONNECT` when there is no communication? 

Advantages
* Client is notified directly when the server is not online when trying to connect (instead of getting an `ETIMEOUT` after 40sec)
* Client is directly informed when the server is not online (instead of getting `ECONNREFUSED`)

I guess it should be a setting wheter the session should be closed or waiting to be recovered within the `requestedSessionTimeout` depending on the use case.

```
Something like
const options = {
  applicationName: "opcua",
  connectionStrategy: connectionStrategy,
  securityMode: MessageSecurityMode.None,
  securityPolicy: SecurityPolicy.None,
  endpointMustExist: true,
  keepSessionAlive: true, // https://github.com/node-opcua/node-opcua/issues/341
  requestedSessionTimeout: 1000, // https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.opcuaclientoptions.html
  closeSessionOnDisconnect:=true // 0=Display EDISCONNECT warning | 1=Close session on disconnect
}
```

AndreasHeine, 2024-08-13:
after some research:
i assume your problem is caused by the tcp socket keepalive, this library uses nodejs's net.socket (buildin socket implementation) at this point i am not sure what the default values in nodejs are but it sounds like there is the actual issue... the bad news it is dependant on the OS -_-' which makes things complicated...

so how should node-opcua know that the socket has lost connection... if the socket does not know it... at this point...

https://nodejs.org/api/net.html#netcreateconnection
https://github.com/node-opcua/node-opcua/blob/af91638471767c90059a973cdc969b903eff0953/packages/node-opcua-transport/source/client_tcp_transport.ts#L37C22-L37C38

not sure if keepalive is active by default... @erossignon

erossignon, 2024-08-15:
To avoid confusion with the original thread, Please reopen a new ticket and fill the bug report form precisely, also provide exact steps to follow to reproduce the issues, also provide source code of a small programm that exhibit the error.

closed 2024-08-15 by erossignon