node-opcua/node-opcua #1197: Compliance: OPC UA Method Attribute behaviour

AndreasHeine, 2022-09-16:
node: v16
node-opcua version: 2.75.0

## Logged in with Call-Permissions:
[image]
[image]

## Logged in without Call-Permissions:
[image]
[image]

if i call it i get "UserAccessDenied" which is correct! but the "UserExecutable" Attribute shall be "false" according to the Spec.

https://reference.opcfoundation.org/v104/Core/docs/Part3/5.7/
[image]

erossignon, 2022-09-19:
Thank you for reporting.

erossignon, 2022-10-02:
fixed in node-opcua@2.79.0

closed 2022-10-02 by erossignon