node-opcua/node-opcua #1199: Node opcua extension object : data_type_manager_104

cdac2019, 2022-09-26:
I am using node-opcua version :"^2.53.0" if I install latest version below error showing on service at first time only(time of node service start)

 10:54:18.883Z :populate_data_type_manager_104:67    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
10:54:19.265Z :populate_data_type_manager_104:67    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
10:54:19.673Z :populate_data_type_manager_104:67    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
10:54:20.085Z :populate_data_type_manager_104:67    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3500
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
10:54:21.312Z :populate_data_type_manager_104:67    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15889
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
=15888
 ...                                                    at C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\source\private\populate_data_type_manager_104.ts:46:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled (C:\Users\web\web-service-nodejs\node_modules\node-opcua-client-dynamic-extension-object\dist\private\populate_data_type_manager_104.js:5:58)
 ...                                                    at processTicksAndRejections (internal/process/task_queues.js:95:5)
 

- **node-opcua version:** :"2.53.0"  working but if I use "^2.53.0" then it showing this isssue and also if  I install latest version that time also showing this issues

  
  

- **Node:**

  `node --version` =v14.17.3

appreciate your help .
Thanks

cdac2019, 2022-09-28:
any one face same issue related to Cannot find dataType Definition !

teropes, 2022-09-29:
Yes I have had this issue many times when Siemens S7 PLC is the OPC UA server. In one case (there could be others as well) it was about some specific "unusual" Siemens data type used in DB that was published to OPC UA interface. If you have easy acccess to the PLC and can do some testing, try to remove DBs from the OPC UA interface one by one to pinpoint the issue to some specific DB. Once you find it, check what data types are used and try to figure out which one is causing this.

IbfUwe, 2022-11-10:
Hi in my Datatype I only use Real, USInt and S7-String

late-e, 2023-03-26:
Having the same issue with S7-1500

Bossorcain, 2023-04-03:
Same issue here with a WAGO 750-8100 PFC100 plc. However we have the special case where we define our own structures and it usually works until the restart of our NodeJS server. Right after the restart the first read produces the error, however if we call the same endpoint from some other source for example from Postman from another host, then it magically starts working, and after that even the first source starts working too.

I am literally confused about it, because the error seems to be inconsistent. Moreover if I try to reproduce it in a local environment it just works and the error never happens to appear.

erossignon, 2023-04-24:
@Bossorcain @late-e , I would need your help to remotely access one of your device and investigate . please send an email at contact@sterfive.com to get in touch with me.

mikakaraila, 2023-04-25:
This seems to be the case for multiple issues in node-red-contrib-opcua.

maxi613, 2023-05-08:
I had today the same problem. To get the Datatype-NodeId I used the following code:
`
          const nodeId = _the node-id of the structure where i want to write;_
            let dataTypeNodeId = await session.read({
                nodeId, 
                attributeId: AttributeIds.DataType
            }); 
        
          let node = dataTypeNodeId.value.value as NodeId; 

          const ext_obj = await session.constructExtensionObject(node,formier_protocol); 
`

erossignon, 2023-05-09:
[cdac2019](https://github.com/cdac2019), I suspect the issue is a duplicate of
* https://github.com/node-opcua/node-opcua/commit/7ecd95b89ae7b2926dbd29ff3ed494c104ed4ce1 Re-entrancy of getExtraDataTypeManager: Made getExtraDataTypeManager re-entrant to avoid potential issues. (Issue https://github.com/node-opcua/node-opcua/issues/1232 )

which is not fixed in [node-opcua@v2.102.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.102.0)

closed 2023-08-17 by erossignon