node-opcua/node-opcua #1200: Internal Error : postponedExtensionObject not resolved issue while loading the nodeset file.

Graos, 2022-09-26:
NODE-OPCUA stack version : 2.77.0
NODEJS version : v16.16.0

**Loading the attached nodeset resulted in the following internal error from the node-opcua server stack**

performPostLoadingTasks Err  =>  Internal Error : postponedExtensionObject not resolved
 Error: Internal Error : postponedExtensionObject not resolved
    at ReaderState.<anonymous> (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:950:39)
    at Generator.next (<anonymous>)
    at \\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:4:12)
    at task (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:947:57)
    at \\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:1381:31
    at Generator.next (<anonymous>)
    at fulfilled (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:5:58)
Error  Error: Internal Error : postponedExtensionObject not resolved
    at ReaderState.<anonymous> (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:950:39)
    at Generator.next (<anonymous>)
    at \\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:4:12)
    at task (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:947:57)
    at \\nodeopcuaServer\node_modules\node-opcua-a
[10ao_out.zip](https://github.com/node-opcua/node-opcua/files/9647651/10ao_out.zip)
ddress-space\dist\source\loader\load_nodeset2.js:1389:31
    at Generator.throw (<anonymous>)
    at rejected (\\nodeopcuaServer\node_modules\node-opcua-address-space\dist\source\loader\load_nodeset2.js:6:65)

erossignon, 2022-09-27:
The file provided is illformed.

Graos, 2022-09-28:
That particular xml is manually created and hence that error. I agree that it should be encoding xml type and will correct it . Thanks for the find.

closed 2022-09-28 by erossignon