node-opcua/node-opcua #1203: Unsupported Service : CreateSessionRequest when running node-opcua-local-discovery-server

stevenhansel, 2022-10-03:
**Current behavior**

**Describe the bug**
Unable to run `node-opcua-local-discovery-server` due to an unsupported service

Here is the log output:
```
$ npx node-opcua-local-discovery-server
port                                     4840
automatically accept unknown certificate true
applicationName                          NodeOPCUA-DiscoveryServer
fqdn                                 webteam1
{ /*ApplicationDescription*/
 applicationUri                /*  String                                    */: urn:webteam1:NodeOPCUA-DiscoveryServer
 productUri                    /*  String                                    */: NodeOPCUA-LocalDiscoveryServer
 applicationName               /*  LocalizedText                             */: locale=en text=NodeOPCUA-LocalDiscoveryServer
 applicationType               /*  ApplicationType                           */: 3 /*(DiscoveryServer)*/
 gatewayServerUri              /*  String                                    */: 
 discoveryProfileUri           /*  String                                    */: 
 discoveryUrls                 /*  String                                []  */: [ /* empty*/ ]
};
discovery server started on port  4840
CTRL+C to stop
rejected Folder  /home/movel/.config/node-opcua-local-discovery-server-nodejs/PKI/rejected
trusted  Folder  /home/movel/.config/node-opcua-local-discovery-server-nodejs/PKI/trusted/certs
local-discovery-server$ 07:15:45.539Z :base_server                   :281   [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:45.540Z :base_server                   :454    messages  [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:49.623Z :base_server                   :281   [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:49.624Z :base_server                   :454    messages  [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:50.654Z :base_server                   :281   [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:50.654Z :base_server                   :454    messages  [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
07:15:54.742Z :base_server                   :281   [NODE-OPCUA-W07] Unsupported Service : CreateSessionRequest
```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Run `npx node-opcua-local-discovery-server`

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- OS version:

    - ( ) Windows : version : ______
    - ( ) Linux : version : Ubuntu 18.04
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

  
  

- **Node:**

  `node --version` = v14.20.1

AndreasHeine, 2022-10-03:
hey, the registration does not use a "session", just a "secure channel"!
the servers which are trying to register at the LDS try to create a session which is not correct...
what servers are they?

stevenhansel, 2022-10-03:
> hey, the registration does not use a "session", just a "secure channel"! the servers which are trying to register at the LDS try to create a session which is not correct... what servers are they?

I believe that it happens when I try connecting from UAExpert to the LDS. 

So, my goal right now is to expose my OPCUA Server to the local area network so it can be connected to devices within the same network. I researched for a bit and figured out that I need to run an OPCUA Local Discovery Server so it can be discovered within the network.

First i run the `npx node-opcua-local-discovery-network` and then after it finishes booting up, then I run my code that calls `node-opcua` API for creating a server like this:
```
   const server = new OPCUAServer({ 
	      port: 4830,
	      registerServerMethod: RegisterServerMethod.LDS,
	      securityModes: [MessageSecurityMode.None],
	      securityPolicies: [SecurityPolicy.None]
      });
```

For testing purposes, I want all the security to be disabled first to make things easier (I don't know if disabling it is what causing the problem).

After investigating a bit, the error shows up only when I try to connect from UAExpert. Is there anything wrong with the setup?

AndreasHeine, 2022-10-03:
no!
you could hookup on some events to see what happens on your server (port 4830)

[image]

and the LDS itself is not an normal opc ua server! you typically ask a LDS which servers he knows (FindServers : https://reference.opcfoundation.org/v104/Core/docs/Part4/5.4.2/)

and with that list you can choose to connect to one
[image]

in the example above i have a LDS on 4840 and a SampleServer on 5000
UaExpert sends a FindServers-Request to the LDS which returns my Port 5000 Server!

erossignon, 2022-10-04:
Thisi is intentional , the discovery server does'nt implement Session Services.
All interactions when the Discovery server are done outside a session.

closed 2022-10-04 by erossignon