node-opcua/node-opcua #1208: OPC UA Non Compliance ConformanceUnits\MonitorClientValueChange Err-006

Graos, 2022-10-17:
node-opcua server version: 2.81.0 

Description:
Subscribe OPC UA Diagnostic signals(monitored items) from OPC UA client with sampling interval of 50ms.
node-opcua Server returns revised sampling interval of 100ms for all the OPC UA diagnostic signal except service level(ns=0;i=2267) which has revised sampling interval of 1000ms (which results in the condition mentioned in the error test case revisedSamplingInterval is decades larger than therequested samplingInterval). Hence the client reports error for that signal.

Snippet from the OPC UA Compliance Test case MonitorClientValueChange Err-006

Inject the following error codes:

2. serviceResult is Good, but the revisedSamplingInterval is decades larger than therequested samplingInterval.

Clientmay log and/or display an information message to the end-user. c. Client logs and/ordisplays a warning message to the end-user. 
d. Client logs and/or displays an error message to the end-user. e. Client logs and/ordisplays an error message to the end-user.

erossignon, 2022-10-17:
An OPCUA Server will not allow the requested Sampling interval to go below the minimumSamplingInternal  of the node.

What is the minimumSamplingInternal attribute value on the node you are trying to monitor?

Graos, 2022-10-17:
minimumSampling interval for the OPC UA server diagnostic signals are not part of the nodeset file been loaded to the server. I assume it is internally configured by the OPC UA server which should be 50ms. The requested sampling interval from client for all the subscribed signals  include OPC UA server diagnostics is 50ms but the revised sampling interval of 1000ms is returned only for that Service Level signal. Please see the Wireshark capture attached. (Packet no.1545 and 1547)

[Wireshark capture.zip](https://github.com/node-opcua/node-opcua/files/9798079/Wireshark.capture.zip)

Graos, 2022-10-18:
closing the case as the minimum sampling rate confiured in the nodeset file for ServiceLevel signal is 1000ms.
[image]

closed 2022-10-18 by Graos