node-opcua/node-opcua #1211: Mutiple Monitored Items on a subscription

fatihkutuk, 2022-11-02:
Hi everybody, i need subscribe to mutliple monitored item on a subcripiton object. 

 it works:

        const async = require("async");
        const { OPCUAClient, makeBrowsePath, AttributeIds, resolveNodeId, TimestampsToReturn } = require("node-opcua");
        
        const endpointUrl = "opc.tcp://127.0.0.1:49320";
        const client = OPCUAClient.create({
            endpointMustExist: false
        });
        let the_session, the_subscription;
        
        async.series([
            function (callback) {
                client.connect(endpointUrl, function (err) {
                    if (err) {
                        console.log(" cannot connect to endpoint :", endpointUrl);
                    } else {
                        console.log("connected !");
                    }
                    callback(err);
                });
            },
            function (callback) {
                client.createSession(function (err, session) {
                    if (err) {
                        console.log(err)
                    }
                    the_session = session;
                    console.log("session created")
                    callback();
                });
            },
            function (callback) {
                const subscriptionOptions = {
                    maxNotificationsPerPublish: 1000,
                    publishingEnabled: true,
        
                };
                the_session.createSubscription2(subscriptionOptions, (err, subscription) => {
                    if (err) {
                        return callback(err);
                    }
        
                    the_subscription = subscription;
        
                    the_subscription
                        .on("started", () => {
                            console.log("subscription started for 2 seconds - subscriptionId=", the_subscription.subscriptionId);
                        })
                        .on("keepalive", function () {
                            console.log("subscription keepalive");
                        })
                        .on("terminated", function () {
                            console.log("terminated");
                        });
                    callback();
                });
            },
            function (callback) {
                const itemToMonitor = {
                    nodeId: resolveNodeId("ns=2;s=Channel1.Device1.temprature"),
                    attributeId: AttributeIds.Value
                }; const itemToMonitor2 = {
                    nodeId: resolveNodeId("ns=2;s=Channel1.Device1.altitude"),
                    attributeId: AttributeIds.Value
                };
                const monitoringParamaters = {
                    samplingInterval: 1000,
                    discardOldest: true
                };
                the_subscription.mon
                the_subscription.monitor(itemToMonitor, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
                    monitoredItem.on("changed", function (dataValue) {
                        console.log("monitored item changed:  temprature = ", dataValue.value.value);
                    });
                });
                the_subscription.monitor(itemToMonitor2, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
                    monitoredItem.on("changed", function (dataValue) {
                        console.log("monitored item changed:  altitude = ", dataValue.value.value);
                    });
                });
                console.log("-------------------------------------");
            },
            function (callback) {
                setInterval(() => "", 10 * 1000);
            },
            function (err) {
                if (err) {
                    console.log(" failure ", err);
                } else {
                    console.log("done!");
                }
                client.disconnect(function () { });
            }
        ])

but i need use just a  "the_subscription.monitor" method and each in all subscription somthing  like that ;

          the_subscription.monitor(monitoredItemsList, monitoringParamaters, TimestampsToReturn.Both, (err,monitoredItem) => {
                      monitoredItem.on("changed", function (datavalues) {
                          datavalues.each(){
                              //
                          }
                });
            });

is it possible ? Thanks in advance

AndreasHeine, 2022-11-02:
https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua.clientsubscription.html#monitoritems

fatihkutuk, 2022-11-02:
Hi Mr. Andreas @AndreasHeine . I checked, but I couldn't see multiple subscriptions on the datachange event.

for example i build this one on **.Net Core**

            _subscriptionList = _databaseManager.GetNodes(_clientId);
            _subscriptionList.CreateMonitoredItemList(out monitoredNodes)
            _nodes = monitoredNodes;
            _subscription.AddItems(_nodes);
            _subscription.FastDataChangeCallback = new FastDataChangeNotificationEventHandler(DataChanged);
            _session.AddSubscription(_subscription);
            _subscription.Create();
        
             private void DataChanged(Subscription subscription, DataChangeNotification notification, IList<string> stringTable)
             {
                  //
             }

in this code object of "DataChangeNotification " has all changes at _nodes object. and nodes object is a monitored items list.
So, i need somthing like that

AndreasHeine, 2022-11-03:
not sure what you want...

what i think you try to do is to have a data structure like a "Map" in which the current state of the monitored items is accessible!?

could be look like:

```js
const opcua = require("node-opcua");

async function main() {

    const opts = {
        clientName: `uaClient-1`,
        endpointMustExist: false,
        connectionStrategy: {
            initialDelay: 2000,
            maxDelay: 10 * 1000,
            maxRetry: 1000000
        },
        requestedSessionTimeout: 30000,
        keepSessionAlive: true
    }

    const data = new Map();
    const endpointUrl = "opc.tcp://127.0.0.1:4840";
    let client = opcua.OPCUAClient.create(opts);
    client.on("backoff", () => {
        console.log("backoff")
        data.clear()
    });
    client.on("connected", () => console.log("connected"));
    client.on("close", () => {
        console.log("connection_close")
        data.clear()
    });
    client.on("connection_reestablished", () => console.log("connection_reestablished"));
    await client.connect(endpointUrl);
    await client.createSession().then(async (session) => {

        session.on("session_closed", () => {
            console.log("session_closed")
            data.clear()
        })

        let nodeList = [
            {nodeId: "ns=3;s=FloatVariable1", attributeId: opcua.AttributeIds.Value},
            {nodeId: "ns=3;s=FloatVariable2", attributeId: opcua.AttributeIds.Value},
            {nodeId: "ns=3;s=StringVariable1", attributeId: opcua.AttributeIds.Value},
            {nodeId: "ns=3;s=Int16Variable1", attributeId: opcua.AttributeIds.Value},
            {nodeId: "ns=3;s=Int16Variable2", attributeId: opcua.AttributeIds.Value},
        ]

        const subscription = await session.createSubscription2({
            maxNotificationsPerPublish: 50,
            priority: 10,
            publishingEnabled: true,
            requestedLifetimeCount: 1000,
            requestedMaxKeepAliveCount: 5,
            requestedPublishingInterval: 2000
        })
        subscription.on("terminated", () => {
            console.log("subscription_terminated")
            data.clear()
        })
        
        nodeList.forEach(async (node) => {
            const nid = node.nodeId
            const mid = await subscription.monitor(
                node,
                {
                    samplingInterval: 500, 
                    queueSize: 100, 
                    discardOldest: true
                },
                opcua.TimestampsToReturn.Both
            )

            mid.on("changed", (dv) => {
                data.set(nid, dv)
            })

            mid.on("terminated", () => {
                data.delete(nid)
            })

            mid.on("err", (msg) => {
                console.log("Monitored Item Error: ", msg)
            })
        })

        const interval = setInterval(() => {
            console.log("##########################################################")
            console.log("MonitoredItems current values: ")
            console.log(data.entries())
            console.log("##########################################################")
        }, 5000)

        process.on('SIGINT', async () => {
            clearInterval(interval)
            await subscription.terminate()
            await session.close()
            await client.disconnect()
            await Promise.resolve().then(
                process.exit(0)
            )
        })
    })

}

(async () => {
    try {
        main()
    } catch (error) {
        console.log(error)
        process.exit(-1)
    }
})()
```

fatihkutuk, 2022-11-03:
Hi sir again @AndreasHeine 

Exactly what i want, Let's say we have 2 monitored items. I want to subscribe to the changes of these 2 different objects through a single datachange object. But in the example code you need to make a subscription instance again for each monitored item. if there are 2 monitored items the code will look like this;

          const itemToMonitor = {
                nodeId: resolveNodeId("ns=2;s=Channel1.Device1.temprature"),
                attributeId: AttributeIds.Value
            }; 
            const itemToMonitor2 = {
                nodeId: resolveNodeId("ns=2;s=Channel1.Device1.altitude"),
                attributeId: AttributeIds.Value
            };
          the_subscription.monitor(itemToMonitor, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
              monitoredItem.on("changed", function (dataValue) {
                  console.log("monitored item changed:  temprature = ", dataValue.value.value);
              });
          });
          the_subscription.monitor(itemToMonitor2, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
            monitoredItem.on("changed", function (dataValue) {
                console.log("monitored item changed:  altitude = ", dataValue.value.value);
            });
        });

but i need like this;

        let monitoredItems = []
        const itemToMonitor1 = {
            nodeId: resolveNodeId("ns=2;s=Channel1.Device1.temprature"),
            attributeId: AttributeIds.Value
        };
        const itemToMonitor2 = {
            nodeId: resolveNodeId("ns=2;s=Channel1.Device1.altitude"),
            attributeId: AttributeIds.Value
        };
        monitoredItems.push(itemToMonitor1 )
        monitoredItems.push(itemToMonitor2 )

        the_subscription.monitor(monitoredItems, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
            monitoredItem.on("changed", function (dataValues) {
             dataValues.map(function(item){
                  console.log(item.value.value)
             })
                console.log("monitored item changed:  temprature = ", dataValue.value.value);
            });
        });

it doesn't make sense to use the the_subscription.monitor feature over and over every time,
I know we can't give a montiroed items list to the conqueror the_subscription.monitor like this, I was wondering if there is any other way or feature I can do about it

AndreasHeine, 2022-11-03:
that's exactly what subscription.[monitoritems](https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua.clientsubscription.html#monitoritems) does! so whats the point?

fatihkutuk, 2022-11-03:
> the_subscription.monitor(monitoredItems, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
>         monitoredItem.on("changed", function (dataValues) {
>          dataValues.map(function(item){
>               console.log(item.value.value)
>          })
>         });
>     });

        the_subscription.monitor(monitoredItems, monitoringParamaters, TimestampsToReturn.Both, (err, monitoredItem) => {
            monitoredItem.on("changed", function (dataValues) {
             dataValues.map(function(item){
                  console.log(item.value.value)
             })
            });
        });

its not working.

AndreasHeine, 2022-11-03:
sure "monitor"-method does not accept lists/arrays like "monitorItems"-method

"monitorItems"-method
[image]

"monitor"-method
[image]

fatihkutuk, 2022-11-03:
I'm so sorry sir it's my fault. I focused on one method for two cases, but that's exactly what I was looking for. thank you very much

closed 2022-11-03 by fatihkutuk