node-opcua/node-opcua #1213: callback issue for invalid channel

raptorsj, 2022-11-09:
i'm trying to ping the server which is running from my opcua client and i'm facing the below error and why it is entering into invalid secure channel as i'm giving **keepSessionAlive** true

### code :

```
const opcua = require('node-opcua');

async function test() {
const options=  {
    requestedSessionTimeout: 30000,
    applicationName: 'PlantView',
    connectionStrategy: {
      initialDelay: 100,
      maxRetry: 1
    },
    securityPolicy: opcua.SecurityPolicy.None,
    keepSessionAlive: true
};
  const client =  opcua.OPCUAClient.create(options);
  const url = 'opc.tcp://192.168.2.10:4840';
  const endpointObj = {endpointUrl:url};
  client.getEndpoints(endpointObj,(err,response )=> {
    if(err) {
      console.log(err);
    }
    else {
      console.log(response);
    }
  });
}

test();

```

### Error : 

C:\Users\Z004DRKP\Plant_Objects\explore\node_modules\node-opcua-client\dist\private\client_base_impl.js:802
                callback(new Error("Invalid Secure Channel"));
                ^

ReferenceError: Cannot access 'callback' before initialization
    at Immediate.<anonymous> (C:\Users\Z004DRKP\Plant_Objects\explore\node_modules\node-opcua-client\dist\private\client_base_impl.js:802:17)
    at processImmediate (node:internal/timers:466:21)

AndreasHeine, 2022-11-09:
i don't see a client.connect() in you code!?

from: https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-samples/bin/simple_client.js
```js
async function main() {

    const optionsInitial = {

        securityMode,
        securityPolicy,

        endpointMustExist: false,
        keepSessionAlive: true,

        connectionStrategy: {
            initialDelay: 2000,
            maxDelay: 10 * 1000,
            maxRetry: 10
        },

        discoveryUrl
    };

    client = OPCUAClient.create(optionsInitial);

    client.on("backoff", (retry, delay) => {
        console.log(chalk.bgWhite.yellow("backoff  attempt #"), retry, " retrying in ", delay / 1000.0, " seconds");
    });

    console.log(" connecting to ", chalk.cyan.bold(endpointUrl));
    console.log("    strategy", client.connectionStrategy);

    try {
        await client.connect(endpointUrl);
        console.log(" Connected ! exact endpoint url is ", client.endpointUrl);

    } catch (err) {
        console.log(chalk.red(" Cannot connect to ") + endpointUrl);
        console.log(" Error = ", err.message);
        return;
    }

    const endpoints = await client.getEndpoints();
}
```

raptorsj, 2022-11-10:
i have a scenario to ping the ip addresses and get only valid ip adderess from the list.Is there anyway to do using findServers and findServersOnNetwork

erossignon, 2022-11-14:
You can get the endpoint this way:

```javascript 
const { OPCUAClient } = require("node-opcua");

const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
(async () => {
  const client = OPCUAClient.create({
    connectionStrategy: {
      maxRetry: 1,
    }
  });

  await client.connect(endpointUrl);
  try {
    const response = await client.getEndpoints();
    console.log(response.toString());
  } finally {
    await client.disconnect();
  }
})();
```

erossignon, 2022-11-14:
> i have a scenario to ping the ip addresses and get only valid ip adderess from the list.Is there anyway to do using findServers and findServersOnNetwork
 
You'll need to use the findServers and findServersOnNetwork command by connecting to the LDS (Local Discovery Server) 

``` javascript
const { OPCUAClient } = require("node-opcua");

const endpointUrl = "opc.tcp://localhost:4841";

(async () => {
  const client = OPCUAClient.create({
    connectionStrategy: {
      initialDelay: 100,
      maxRetry: 1,
    }
  });

  await client.connect(endpointUrl);
  console.log("connected to LDS");
  try {
    const response = await client.findServersOnNetwork({
        maxRecordsToReturn: 1000,
        serverCapabilityFilter: null,
        startingRecordId: 0
    });
    console.log(response.toString());
  } finally {
    await client.disconnect();
  }
})();
```` 

```` 
{ /*ServerOnNetwork*/
 recordId                      /*  UInt32                                    */: 1               0x1
 serverName                    /*  String                                    */: urn:XXXXX:MyLDS
 discoveryUrl                  /*  String                                    */: opc.tcp://XXXXX:4841/DiscoveryServer
 serverCapabilities            /*  String                                []  */: [ /* length =1*/
   LDS
 ]
};
```

the discoveryUrl properties contain the endpointUrls of the addressable servers.

erossignon, 2023-05-01:
@raptorsj,  is this problem solve now on your end ?

raptorsj, 2023-05-04:
resolved

closed 2023-05-04 by erossignon