node-opcua/node-opcua #1214: Properties with HasModellingRule = Mandatory are not instantiated when parents are created

jives, 2022-11-09:
**Current behavior**
I am using node-opcua to build a very simple OPC UA server that loads NodeSet2 XML files on startup.
My server code looks like this
```javascript
const opcua = require('node-opcua')
const path = require('path')
const fs = require('fs')

const xmlDir = process.env.XMLDIR || '.\\xml\\test';
var files = fs.readdirSync(xmlDir);
var xmlFiles = files.filter(file => path.extname(file).toLowerCase() === '.xml').map(f => path.join(xmlDir, f));
xmlFiles.unshift(opcua.nodesets.standard);
console.log(xmlFiles);

const server_options = {
  nodeset_filename: xmlFiles,
  port: 4840,
};

const server = new opcua.OPCUAServer(server_options);

function post_initialize() {
  console.log('Server initialized.');

  server.start(function() {
    console.log('Server is now listening... (press CTRL+C to stop)');
    console.log('Server port: ', server.endpoints[0].port);
    const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
    console.log('The primary server endpoint url is: ', endpointUrl );
  });
}

server.initialize(post_initialize);
```
The `.\xml\test` directory contains just one XML file:
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<UANodeSet xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd">
  <NamespaceUris>
    <Uri>urn:testNS</Uri>
  </NamespaceUris>
 
  <UAObjectType NodeId="s=testType" BrowseName="0:testType">
    <DisplayName>testType</DisplayName>
    <Description>testType</Description>
	<References>
	  <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
    </References>
  </UAObjectType>
  <UAVariable NodeId="s=testType.testProperty" BrowseName="0:testProperty" DataType="String" ValueRank="-1">
    <DisplayName>testProperty</DisplayName>
    <Description>testProperty</Description>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">s=testType</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
    </References>
    <Value>
      <String>HelloWorld</String>
    </Value>
  </UAVariable>
  
  <UAObject NodeId="s=testInstance" BrowseName="0:testInstance">
    <DisplayName>testInstance</DisplayName>
    <Description>testInstance</Description>
    <References>
	  <Reference ReferenceType="HasSubtype" IsForward="false">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">s=testType</Reference>
    </References>
  </UAObject>
</UANodeSet>
```
Starting the server with `node server.js` works just fine, the server finds and loads the XML file correctly and I can see the `testType` object type defined under Types/ObjectTypes/BaseObjectType as expected, with the `testProperty` variable.

**Describe the bug**

However, the `testInstance` object under Root/Objects does _not_ contain the `testProperty`, although the property has set the `HasModellingRule` reference to `i=78` (Mandatory), which is also shown correctly in my client. This issue exists with opcua-commander, UaExpert and FreeOpcUA Client.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy the XML above into `.\xml\test`
2. Start the above server script
3. Using a client, observe that the expected property is missing from the `testInstance` object

**Expected behavior**

The `testInstance` object should have a variable called `testProperty`.

**Screenshots**

**Gist**

 
gist:

See code above.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Windows PC
- OS version: Windows 10

    - (x) Windows : version : 10 (21H1)
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 
  2.82.0
  
  

- **Node:**

  `node --version` = v19.0.0

erossignon, 2022-11-14:
All nodes must be explicitly describes in the XML file. The testProperty node which is a child of the testinstance is missing in the XML file.

erossignon, 2023-05-01:
Closing due to inactiivty

closed 2023-05-01 by erossignon